tbnobody/OpenDTU #2: compiled bin

hismastersvoice, 2022-06-29:
Hi,

nice Project...
Can you upload a compiled bin version that it is easier to install?

Thanks in advance.

Dieter

itn3rd77, 2022-06-30:
That would be great!

Just ordered all components, but compiling everything is definitely a challange.

tbnobody, 2022-06-30:
I like to understand whats the complicated thing in compiling the firmware. 
You have to run a setup. Make two clicks to install an plugin, make two clicks to open a folder and press 1 button to start the compilation.
What is the complicated step? Is there anything that has to be more documented?

itn3rd77, 2022-06-30:
Thanks for your fast reply!
To be fair I should tried before simply asking for the low hanging fruit. I never used Visual Studio Code or need to compile something for my ESP devices. I'll give it a try and if I find something to improve I'll open a pull request.

Thanks for your time and the tremendous work you spend on this project!

ahinrichs, 2022-07-01:
> Following the instructions i cannot compile the code, tried on two different PCs with the same result.
> 
> src/Configuration.cpp: In member function 'bool ConfigurationClass::write()': src/Configuration.cpp:52:14: error: 'LittleFS' was not declared in this scope File f = LittleFS.open(CONFIG_FILENAME, "w");
> 
> How should i proceed, any hints?

Maybe a platform update is necessary. Try `pio pkg list -g --only-platforms` and check the available espressiv32 versions. You need espressif32 version 4. I created a [pull request](#26) to add this to the ini.

tbnobody, 2022-07-01:
Das ist korrekt. Es wird die aktuellste version benötigt. (Werde den PR spät. heute Abend mergen)

ahinrichs, 2022-07-01:
> Das ist korrekt. Es wird die aktuellste version benötigt. (Werde den PR spät. heute Abend mergen)

"Aktuellste" ist relativ, seit gestern gibt es 5.0.0. Ich war mir nicht ganz sicher, was wirklich Minimum ist. Platformio empfiehlt ja eine Semantische Version anzugeben, also z.B. `^4.4.0`, soll heißen irgendwas kompatibel mit Version 4.4.0. Ich habe erstmal `@>4` im PR, da LittleFS seit der 4er drin ist. Aber dann würde ich das nochmal in `espressif32@^4.4.0` ändern. Einverstanden?

tbnobody, 2022-07-01:
Hm lass mich ggf. mal die Changelogs noch anschauen. Ggf. kann man auch gleich auf die Version 5 gehen. (Dann würde aber vmtl. >4 auch wieder passen)

ahinrichs, 2022-07-01:
> Hm lass mich ggf. mal die Changelogs noch anschauen. Ggf. kann man auch gleich auf die Version 5 gehen. (Dann würde aber vmtl. >4 auch wieder passen)

5 nutzt die gleiche Toolchain wie 4.4.0. 

Wenn jemand nur 3 hat und in der ini steht `@^4.4.0`, installiert er 4.4.0. Wenn `@>4` drin steht, installiert er jetzt 5.

rosch99, 2022-07-01:
"pio pkg list -g --only-platforms" hat auch nicht funktioniert, musste erst noch PATH anpassen.
Anschliessend Update auf 5.0.0 und der build hat funktioniert :-) 
OpenDTU läuft deutlich stabiler als die ahoy Version auf einem WemosD1   
Vieeelen Dank für die Version und eure Hilfe!

hubsi5, 2022-07-02:
Ich schaffe es nicht einmal alles richrig runter zuladen.
Schade, dass es nicht einfach eine -bin Datei gibt

DaAlchemist, 2022-07-09:
Ich versuche jetzt seit ein paar Stunden die Firmware zu kompilieren, visual Studio code und PlatformIO Extension habe ich installiert, den Comport angepasst. Klicke ich auf "upload" kommt als Fehlermeldung "command 'platformio-ide.upload' not found" zusätzlich kammennoch die Meldungen "PlatformIO: Rebuilding IntelliSense Index" und "PlatformIO: Loading tasks..." weiter passiert dann nichts...
Ich weiss so langsam nicht mehr was ich noch machen kann.
Wäre für jede Hilfe dankbar!

DaAlchemist, 2022-07-09:
Hat sich erledigt auf einem alten i3 Laptop hats geklappt..