tbnobody/OpenDTU #5: TSUN M800 Suppport

viktorhotel, 2022-07-01:
Muss leider nochmal was fragen: Nachdem der Build, das Flashen und die Konfiguration des ESP32 super funktioniert haben:
 
Ich habe einen TSUN TSOL M800 Inverter Ser Nr: 1141647055xx  Leider wird von diesem nichts empfangen. Das Log der seriellen Console habe ich angehängt. Ein ESP8266 mit der Ahoy Software empfängt die Daten des Inverters. Verkabelung habe ich geprüft. NRF+ Module habe ich untereinander getauscht. Beide funktionieren am ESP8266. Ein Elko an den 3.3V bringt keine Verbesserung. Ein anderes ESP32 Modul habe ich gerade leider nicht verfügbar.
Verkabelung ist wie folgt lt. Doku:
CE GPIO4
CS GPIO5
SCK GPIO18
MOSI GPIO23
MISO GPIO19
IRQ GPIO16

serielles Log: 
Using 'COM6' as serial port.
Showing logs:
[08:11:46]ets Jul 29 2019 12:21:46
[08:11:46]
[08:11:46]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[08:11:46]configsip: 0, SPIWP:0xee
[08:11:46]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:11:46]mode:DIO, clock div:2
[08:11:46]load:0x3fff0018,len:4
[08:11:46]load:0x3fff001c,len:1044
[08:11:46]load:0x40078000,len:8896
[08:11:46]load:0x40080400,len:5828
[08:11:46]entry 0x400806ac
[08:11:46]
[08:11:46]Starting OpenDTU
[08:11:46]Initialize FS... done
[08:11:46]Reading configuration... done
[08:11:46]Initialize WiFi... done
[08:11:46]Setting Hostname... done
[08:11:46]Configuring WiFi STA using new credentials... done
[08:11:46]Initialize NTP... done
[08:11:46]Initialize MqTT... done
[08:11:46]Initialize WebApi... done
[08:11:46]Initialize Hoymiles interface... Connection successfull
[08:11:46]done
[08:11:46]WiFi connected
[08:12:01]Fetch inverter: 18972555629891
[08:12:01]TX 15 64 70 55 43 78 56 34 12 80 B 0 62 BE 90 33 0 0 0 5 0 0 0 0 2B 30 F5 
[08:12:01]RX Period End
[08:12:01]All missing
[08:12:01]Nothing received, resend whole request
[08:12:01]TX 15 64 70 55 43 78 56 34 12 80 B 0 62 BE 90 33 0 0 0 5 0 0 0 0 2B 30 F5 
[08:12:02]RX Period End
[08:12:02]All missing
[08:12:02]Nothing received, resend whole request
[08:12:02]TX 15 64 70 55 43 78 56 34 12 80 B 0 62 BE 90 33 0 0 0 5 0 0 0 0 2B 30 F5 
[08:12:02]RX Period End
[08:12:02]All missing
[08:12:02]Nothing received, resend whole request
[08:12:02]TX 15 64 70 55 43 78 56 34 12 80 B 0 62 BE 90 33 0 0 0 5 0 0 0 0 2B 30 F5 
[08:12:02]RX Period End
[08:12:02]All missing
[08:12:02]Nothing received, resend count exeeded
[08:12:16]Fetch inverter: 18972555629891
[08:12:16]TX 15 64 70 55 43 78 56 34 12 80 B 0 62 BE 90 42 0 0 0 5 0 0 0 0 79 58 BE 
[08:12:16]RX Period End
[08:12:16]All missing
[08:12:16]Nothing received, resend whole request
[08:12:16]TX 15 64 70 55 43 78 56 34 12 80 B 0 62 BE 90 42 0 0 0 5 0 0 0 0 79 58 BE 
[08:12:17]RX Period End
[08:12:17]All missing
[08:12:17]Nothing received, resend whole request
[08:12:17]TX 15 64 70 55 43 78 56 34 12 80 B 0 62 BE 90 42 0 0 0 5 0 0 0 0 79 58 BE 
[08:12:17]RX Period End
[08:12:17]All missing
[08:12:17]Nothing received, resend whole request
[08:12:17]TX 15 64 70 55 43 78 56 34 12 80 B 0 62 BE 90 42 0 0 0 5 0 0 0 0 79 58 BE 
[08:12:17]RX Period End
[08:12:17]All missing
[08:12:17]Nothing received, resend count exeeded

tbnobody, 2022-07-01:
Hallo,

nachdem die ESP8266 ahoy Firmware (meines wissens) aktuell nur die Hoymiles HM- unterstützt und es dort funktioniert sollte es generell mit der ESP32 Firmware auch laufen. Bezogen auf die Seriennummer (1141) wird der TSUN M800 als HM-800 erkannt (bzw. allgemein als 2-Kanal Gerät).

Das Log "All missing" bedeutet das kein einziges Datenpaket empfangen wurde. Jetzt gibt es folgende Möglichkeiten:
* Der Inverter empfängt nichts vom NRF
* Der Inverter versteht nicht was der NRF möchte
* Der NRF empfängt nichts vom Inverter

Hattest du in der ESP8266 Firmware die Sendeleistung umgestellt? Standardmäßig scheint die auf LOW zu stehen: https://github.com/grindylow/ahoy/blob/b1f09d482f107a8808fc26a229146dc7b25c4c7f/tools/esp8266/hmRadio.h#L75

Mein Default Setting ist MIN. Wobei folgende Reihenfolge gilt:
MIN < LOW < HIGH < MAX

Du könntest versuchen die Sendeleistung in der ESP Firmware ebenfalls auf Low zu stellen (Settings --> DTU Settings)

(Ein Problem in der Verkabelung kann man ziemlich sicher ausschließen da die Firmware mit dem NRF kommunizieren kann "Initialize Hoymiles interface... Connection successfull done")

Verwendest du in der ESP8266 Firmware die Default DTU_RADIO_ID?

viktorhotel, 2022-07-01:
Im AhoyESP8266 steht die Sendeleistung auf MIN. 
Ich verwende dort die Default DTU_RADIO_ID. 
Der AhoyESP8266 ist aktuell ausgeschaltet. Um sicher zu sein, dass nicht zwei auf einen Inverter einreden :-)
Eine Veränderung der Sendeleistung auf LOW, High, Max bringt keinen Erfolg.
AUS/EIN des TSOL ohne Erfolg

stefan123t, 2022-07-01:
Hallo @viktorhotel kannst Du mal Deine every.h vergleichen, bei mir lauten die letzten Zeilen 136-142:
```
#define EVERY_N_MINUTES_I(NAME, N) \
    static CEveryNMinutes NAME(N); \
    if (NAME)
#define EVERY_N_HOURS(N) EVERY_N_HOURS_I(CONCAT_MACRO(PER, __COUNTER__), N)
#define EVERY_N_HOURS_I(NAME, N) \
    static CEveryNHours NAME(N); \
    if (NAME)
```
Ich vermute Dir fehlen zwei Zeilen!

viktorhotel, 2022-07-01:
#define EVERY_N_MINUTES_I(NAME, N) \
    static CEveryNMinutes NAME(N); \
    if (NAME)
#define EVERY_N_HOURS(N) EVERY_N_HOURS_I(CONCAT_MACRO(PER, __COUNTER__), N)
#define EVERY_N_HOURS_I(NAME, N) \
    static CEveryNHours NAME(N); \
    if (NAME) 

Sieht erst mal gleich aus. Habe in Zeile 143 eine leere zeile eingefügt, dann kommt kein Fehler mehr ?

viktorhotel, 2022-07-02:
Es läuft jetzt. !!
Danke für die Unterstüzung an alle Beteilgten.
Ich habe ein anderes ESP32 Board genommen, neu geflasht und läuft sofort.
Das aktuell funktionierende ist ein LOLIN32 Board. Hier ist die Pinbelegung jedoch anders. 
Das nicht funktionierende war ein ESP32-DevKitCVE Board. Vielleicht ist hier der 3V3 Regler doch zu schwach für das NRF+. Das ist aber Spekulation.

stefan123t, 2022-07-02:
@viktorhotel das mit der Pinbelegung scheint mir am wahrscheinlichsten.
Versuche es doch nochmal mit einer anderen Einstellung für die Anschlüsse bzw. deren Verkabelung auf dem ESP32-DevKitCVE Board ?

viktorhotel, 2022-07-03:
@stefan123t Da es jetzt mit dem Lolin32 super läuft, sonst keiner Probleme mit dem ESP32 Board gemeldet hat, werdeich es nicht wieter damit probieren 
Das ESP32-DevKitCVE macht jetzt Präsenzerkennung via BLE, das läuft :-
[image]
)

closed 2022-07-03 by viktorhotel