tbnobody/OpenDTU #11: Statistik von yieldtoday für einen längeren Zeitraum !

eeprom23, 2022-07-04:
Ist es möglich den YieldToday in einer Statistik zu speichern und abrufbar zu machen?
[Datum][ertrag] eventuell die letzte Woche zurück ?!

tbnobody, 2022-07-04:
Das hatte ich mir auch schon überlegt. Jedoch wird man sich dadurch schneller das Eeprom kaputt schreiben als einem lieb ist. Man könnte sich vorstellen diese Werte im RAM zu speichern. Hier gehen sie bei einem Power Cycle verloren. Aber bei einer retention Time von 7 Tagen ist hier sowieso keine Aussagekräftige Statistik möglich. Für alles Andere ist dann sowieso ein MQTT Broker mit nachgelagerter InfluxDB oder TimescaleDB empfehlenswert.

eeprom23, 2022-07-04:
> Das hatte ich mir auch schon überlegt. Jedoch wird man sich dadurch schneller das Eeprom kaputt schreiben als einem lieb ist. Man könnte sich vorstellen diese Werte im RAM zu speichern. Hier gehen [...]

Im RAM wäre auch ok, wobei ja einmal am Abend langen würde die Werte zu schreiben !
Es geht ja nur darum das man sich die letzten Tage mal anschauen kann !

ericlangel, 2022-08-02:
Der "EEPROM" im ESP32 ist nur emuliert.  Alle Daten werden im Flash Speicher abgelegt mit Wear-Leveling.

Eine Langzeitgrafik wäre wirklich super. Alternativ kann man die Daten in Filesystem als Datei speichern. Dann hat die Weboberfläche eventuell leichter Zugriff?

Interessant wäre:
- yieldtoday über mehrere Wochen (insgesamt 365 Tage?)
- eingespeiste Leistung über 24 Std (bzw über den Tag)

ottelo9, 2022-09-06:
Die Google Chart API könnte da helfen. Nutzt z.B. Tasmota. Tasmota speichert jeden Abend 1x die Daten in den Flash, somit sollte das Wearing überschaubar sein.

andy5macht, 2023-01-03:
Eine Darstellung der Leistung im Tagesverlauf und/oder Energieproduktion in den letzten Tagen würde mich auch intressieren.
Dies per MQTT / Homeassistant zu realisieren vermag leider nicht jeder.
Das wäre aus meiner Sicht ein sehr großer Mehrwert für das Projekt.

glucosum, 2023-04-21:
I know it's not exactly what you meant, but if you solve the problem by saving on the SD card connected to our ESP32? 
Personally, I'm not satisfied with the MQTT data.

stefan123t, 2024-07-10:
@eeprom23 did you check the project / fork [OpenDTU-Database](https://github.com/RaBa64/OpenDTU-Database) ?
It uses the Flash-Filesystem on the ESP32 to store some of the values and displays them in a graph.
We do not know for sure how long the ESP32 can sustain such writes to the flash.
It may be better to attach an SD Card module and use that for storage.