tbnobody/OpenDTU #16: RFC: Switch license to AGPL

ahinrichs, 2022-07-05:
Hi,

in my opinion the [AGPL](https://www.gnu.org/licenses/why-affero-gpl.en.html) would be the better choice for this project. With the GPL it is possible to create modified binary firmware on hardware devices without contributing back.

Axel

ahinrichs, 2022-07-05:
> Hmmmmmm good question...
> 
> Source: https://de.wikipedia.org/wiki/GNU_Affero_General_Public_License
> 
> > Bei der GPL besteht nur bei der Weitergabe der [ausführbaren Datei](https://de.wikipedia.org/wiki/Ausf%C3%BChrbare_Datei) ein Rechtsanspruch auf den Quelltext.
> 
> Does that mean, if someone is creating a device, pre-flashes it with a modified firmware, he does not have to to contribute this changes back? 

I think I was wrong. I recall several lawsuits over router with Linux/GPL-derived work and in the end they had to publish the sources. And [this](https://opensource.stackexchange.com/questions/4718/hardware-with-gpl-firmware) reads the same. 

The question overall came to me when I started to overlook the AboutView. Either way it would make a lot of sense to make a proper impress with listing of libraries etc. and include the copyright notes in the files to be safe. In the end **we** are using/redistributing GPL licensend code and therefore required to include proper notes to the user ;-)

ahinrichs, 2022-07-11:
I did some research about this topic. Not to overdo but what to do for starters.

To me that means:
1. what part gets which license and
2. how and where to note that besides LICENSE.

For the second part [SPDX](https://spdx.dev/resources/use/) seems to be kind of standard and they have simple [examples](https://github.com/spdx/spdx-examples).

For the first part I read about the consequences of the different licenses that I found in the libraries used.

**OpenDTU**

Most important to me seems to check the license of OpenDTU. The GPL seems perfectly fine and fitting to me. But maybe there is an issue with the version 3. All exept one library use compatible licenses MIT or LGPL 3. But the RF24 lib is "GPLv2 only" licensed. This is considered [incompatible with GPLv3 projects](https://www.gnu.org/licenses/gpl-faq.html#v2v3Compatibility).

To all my understandig that means OpenDTU needs to be licensed under the GPLv2 terms too. It seems possible to me to use any of "GPL-2.0-or-later" or "GPL-2.0-only". I would prefere "-or-later".

**Library Hoymiles**

Appart from OpenDTU - src, includes, webapp and some more - you separated the Hoymiles code in lib/Hoymiles. This Library can have any license you like. I would suggest LGPL3 but that is up to you.

**Other libs**

We need to check the other libs.

ahinrichs, 2022-07-11:
Second part. It is recommended to have a license note in every source file. SPDX conventions help so that it is just a one line comment - and the full license as LICENSE or COPYING. 

It is suggested but not required, that every file also contains a copyright claim. That would give the following header for every source file:

```
/* SPDX-License-Identifier: GPL-2.0-or-later */
/* Copyright 2022 Thomas Basler and others */
```

Of course containing the respective final SPDX ids.

Served HTML and JS does not need a copyright/license note. But I would include the above information in a human readable form, a link to the license and maybe a list of or link to a list of the used libraries.

tbnobody, 2022-07-14:
You are right. Based on: https://www.gnu.org/licenses/gpl-faq.en.html#AllCompatibility
and due to the fact that the NRF24 library is GPLv2 only, the OpenDTU license in GPLv3 is not allowed. Therefor you are right, it would make sense to use GPLv2 or later

As far as I understood it, the LICENSE file itself just contains the license (in this case https://www.gnu.org/licenses/old-licenses/gpl-2.0.txt)

Am I correct that there is no specific license for the "-or-later" term? This is just mentioned in the SPDX header?

tbnobody, 2022-07-14:
> Appart from OpenDTU - src, includes, webapp and some more - you separated the Hoymiles code in lib/Hoymiles. This Library can have any license you like. I would suggest LGPL3 but that is up to you.

I think thats not true.  

> 2: While you may release your project (either your original work and/or work that you received and modified) under GPLv2-or-later in this case, note that the other code you're using must remain under GPLv2 only. As long as your project depends on that code, you won't be able to upgrade the license of your project to GPLv3-or-later, and the work as a whole (any combination of both your project and the other code) can only be conveyed under the terms of GPLv2.

Due to the fact that especially the Hoymiles lib uses the RF24 Lib, it cannot be licensed under GPLv3.

tbnobody, 2022-07-14:
> It seems possible to me to use any of "GPL-2.0-or-later" or "GPL-2.0-only". I would prefer "-or-later".

On the other hand, a pure GPL-2.0 should also work as the "-or-later" will not apply because v3 is already excluded?

ahinrichs, 2022-07-15:
> Due to the fact that especially the Hoymiles lib uses the RF24 Lib, it cannot be licensed under GPLv3.

Of course, you are right. I did not had enough insight in the dependencies for this.

ahinrichs, 2022-07-15:
> You are right. Based on: https://www.gnu.org/licenses/gpl-faq.en.html#AllCompatibility and due to the fact that the NRF24 library is GPLv2 only, the OpenDTU license in GPLv3 is not allowed. Therefor [...]

Convention seems to be to mention this in a file COPYING (example like the linux kernel does it):
```
OpenDTU - ESP32 firmware to control HoyMiles Inverter

Copyright (C) 2022 Thomas Basler and others

OpenDTU is provided under:

SPDX-License-Identifier: GPL-2.0-only

Being under the terms of the GNU General Public License version 2 only,
according with:

LICENSE
```

Or of course "-and-later" and then add a `or (at your option) any later version.`

If it helps I could create a pull request for this and the file headers.

ahinrichs, 2022-07-15:
> > It seems possible to me to use any of "GPL-2.0-or-later" or "GPL-2.0-only". I would prefer "-or-later".
> 
> On the other hand, a pure GPL-2.0-only should also work as the "-or-later" will not apply because v3 is already excluded?

I can think of several cases where it does mak a difference.

First, if someone wants to reuse parts of this project, let's say just the webapp. GPL2-or-later would make this possible for other GPL3 projects.

Second, if you later want to replace RF24 with another lib. But that could of course also be done with a license change at that time.

In the end, it is up to you.

ahinrichs, 2022-07-15:
As said, I could make a PR, if it helps. Just let me know.

tbnobody, 2022-07-15:
I've already modified several stuff in my current source tree. (Update LICENSE, Added SPDX Header to most of the c/h files etc., the libs are currently a different part). As far as I see it, I just need an additional COPYING file.... And SPDX Headers...

ahinrichs, 2022-07-15:
Yes. So according to the title "check license" I would close this issue, right?

tbnobody, 2022-07-15:
First of all, thank you for your help regarding this topic!
Please leave it open until I added all the stuff to the relevant files.

ahinrichs, 2022-07-15:
Welcome. Sure, you close it then.

ahinrichs, 2022-07-15:
I see you checked in the COPYING. There is a mismatch of the SPDX and the phrase later. Probably copy&paste.

For GPL-2.0-or-later the phrase should read as this:
```
Being under the terms of the GNU General Public License version 2
or any later version, according with:
```

ahinrichs, 2022-07-15:
Maybe my Word wrapping?

stefan123t, 2022-07-25:
Hi Thomas and Alex,

> The GPL seems perfectly fine and fitting to me. But maybe there is an issue with the version 3. All exept one library use compatible licenses MIT or LGPL 3. But the RF24 lib is "GPLv2 only" licensed. This is considered [incompatible with GPLv3 projects](https://www.gnu.org/licenses/gpl-faq.html#v2v3Compatibility).
> 
> To all my understandig that means OpenDTU needs to be licensed under the GPLv2 terms too (or replace the RF24 lib). It seems possible to me to use any of "GPL-2.0-or-later" or "GPL-2.0-only". I would prefer "-or-later".

I understand that it would be easier / preferred from @tbnobody to use GPL v3 or later according to the discussion so far for OpenDTU which relies on NRF24 to make the communication between the ESP32 and the Hoymiles Solar inverters possible.

@2bndy5 is the author of the RF24 library and usually responding very kind and quickly. 
Maybe he can give you a hint on his choice for not including the -or-later wording into the GPL-2.0 license for the NRF24 library ?

Thanks for all your work !

2bndy5, 2022-07-25:
I don't entirely understand the thread here, but from what I can tell you...

### As for SPDX identifier

I don't see anything in RF24 license that would require redistribution to use "only". I'm fine with "or-later".

### As for the project's license
Is there a problem with using multiple licenses? In that case, you can choose whatever you want. Just make sure to make note of any code that uses a different license and disclosure a copy of the shortened license. This is what we're doing for the RF24Ethernet library.

---
Because this issue is primarily symantic,
@stefan123t I'm not the "author" of RF24, just a passionate maintainer. The author (maniacbug) hasn't been responsive since around 2016.

tbnobody, 2022-07-25:
This file / line https://github.com/nRF24/RF24/blob/89358b2df7de37d64f2bba180d61b6d200c54070/library.json#L5 says GPL-2.0-only

Therefor any application or library which uses the NRF24 library has to be GPL-2.0 because they are derivative work as I understand it.

2bndy5, 2022-07-25:
That's the PlatformIO description. And I think you're getting caught up on the "only" phrase. You can redistribute the RF24 lib under the GPLv2 and still license your code (that uses the RF24 code) using a different license.

2bndy5, 2022-07-25:
The SPDIX isn't legally binding. If the RF24 lib used external sourced code that was licensed under GPLv3, then that SPDIX would say "or-later". The SPDIX is not part of the license, is only meant as a helpful description.

In any case, reading the license file itself is the best way to understand your permissions and prohibitions.

2bndy5, 2022-07-25:
Fun fact: RF24Ethernet is the reason I urged PlatformIO to support multiple SPDIX identifiers.

stefan123t, 2022-09-19:
> 3. All exept one library use compatible licenses MIT or LGPL 3. But the RF24 lib is "GPLv2 only" licensed. This is considered incompatible with GPLv3 projects.

> To all my understandig that means OpenDTU needs to be licensed under the GPLv2 terms too (or replace the RF24 lib). It seems possible to me to use any of "GPL-2.0-or-later" or "GPL-2.0-only". I would prefer "-or-later".

So if they (RF24 maintainers) us the GPLv2-or-later AND other licenses for their projects (RF24Ethernet here authors), we can do the same with confidence too.

We just have to include the permissions granted to RF24 users under the GPLv2-only license that is we must regrant access under any GPLv2-only OR GPLv2-or-later. We cannot license it only under the GPLv3 and not GPLv2, thats it.
But we can license code that comes on top of RF24 library as both GPLv2 AND GPLv3 or any other license you like to grant access to.

tbnobody, 2023-04-11:
Most of the source contains the license headers. Going to close this issue. Thank you for the support!

closed 2023-04-11 by tbnobody