tbnobody/OpenDTU #18: WLAN password limited to 32 characters?

itn3rd77, 2022-07-06:
Hi,

I just manged to install my EPS32 with this fantastic software. Great work - thanks a lot!

But I am unable to get this connected with my WLAN and suppose this is due to a limitation in the WLAN password length.
All my WLAN passwords are 63 chars long with really all kind of characters. I tried with short PSK and that work.

Would you be able to allow the max allowed length of 63 chars for WPA2 PSK ?

Thanks and best regards
Ingo

tbnobody, 2022-07-06:
Hi Ingo,

du hast recht. Das ist in der WebGUI noch auf 32 Zeichen limitiert (siehe https://github.com/tbnobody/OpenDTU/blob/44ac770c53dac24692e7a99b88e84cfea6bfbc76/webapp/src/components/NetworkAdminView.vue#L24)

Im Backend sind schon 64 Zeichen erlaubt. Werde versuchen heute ein paar Minuten Zeit zu finden und das zusammen mit anderen Änderungen zu committen...

itn3rd77, 2022-07-06:
Super, Danke Dir!

closed 2022-07-06 by tbnobody