tbnobody/OpenDTU #20: Live-Data Ansicht - kleiner Schönheitsfehler

hismastersvoice, 2022-07-07:
Nachdem die Daten in der mobilen Ansicht jetzt perfekt passen, gibt es einen kleinen Schönheitsfehler bei der Einteilung der <div>
In meinem Fall wird das letzte Feld auf die volle Breite gezogen, und nicht nur so breit wie die bestehenden.
Zwar nur ein winziger Fehler aber.... ;)

[image]

ahinrichs, 2022-07-07:
Ich habe es länger probiert, kriege aber keine Kontrolle, wo der Umbruch gemacht wird. Schwierig ist es, da die Anzahl der Kästchen ja variabel ist. Mal zwei, mal drei, mal fünf. Er probiert es stur von rechts nach links und was nicht passt kommt in die nächste Zeile. Jeder Versuch da einzugreifen, hatte größere Nebenwirkungen als Verbesserungen.

Zwei Alternativen die ich mir vorstellen kann (ohne Garantie, dass das umsetzbar ist): 

Man verhindert, dass die Kästchen in der Breite wachsen. Dann wäre die Ausrichtung halt rechts nicht bündig, aber kein Kästchen so extrem verbreitert.

Oder man findet einen Weg, dass das Phase-Kästchen einzeln steht. Da ist mir aber nichts gelungen, ohne dass ich das als Hack empfinden würde (z.B. die Reihenfolge im DOM vertauschen und col/row-reverse nutzen). Außerdem hilft das ja nur, wenn es bei allen Modellen wie hier als ersten Kanal die AC Infos gibt, und danach eine gerade Anzahl von DCs. Kurz, auch das kann eigentlich nicht klappen, da die Ansicht ja für verschiedene Konfigurationen passen muss. Oder man macht es als Fleißarbeit "handoptimiert" für jede mögliche Anzahl. Aber das war mir in Anbetracht der knappen Resourcen zu unsympathisch, bzw. ich bin nicht gut genug in Vue.

hismastersvoice, 2022-07-07:
Danke schon mal fürs anschauen.
Ich würde die Prio nicht zu hoch setzten, ist ja nur ein Schönheitfehler ;)

ahinrichs, 2022-07-07:
Ich habe eigentlich nur noch Schönheitsfehler auf meiner Liste :-)

Aber im Ernst, von allen Lösungen, die ich bisher probiert habe, ist die Aktuelle die mit dem kleinsten Schönheitsabstrich. Weil obige Variante, also rechts unausgerichtet, gefällt mir gar nicht. Wenn, würde ich nochmal versuchen, dass er die Phase als erstes einzeln setzt. Aber wann ich das nochmal probiere weiß ich nicht.

tbnobody, 2022-07-07:
Werde ggf .auch noch nen blick drauf werfen. Weil Spoiler: Ich muss folgende JSON Response noch schön integrieren
`{"116171603546":{"count":2,"0":{"message_id":1,"message":"Inverter start","start_time":15858,"end_time":18930},"1":{"message_id":212,"message":"Port 4 no input","start_time":15866,"end_time":0}},"116171603547":{"count":0}}`

Bin nur noch unschlüssig wo

tbnobody, 2022-07-07:
Was denkt ihr über so ein Layout?
[image]

Die Phase ist immer als Einzelkarte oben und die Strings darunter. (Der Abstand zwischen Phase und Strings stört mich noch etwas)

hismastersvoice, 2022-07-08:
Wenn das dann auch in der mobilen Ansicht passt würde ich das so gut finden... Danke.

ahinrichs, 2022-07-08:
In die Richtung habe ich es auch probiert und bisher nicht geschafft. Also nicht "responsive". 

Problem dabei ist ja, dass wir nicht wissen, wieviele Kanäle es gibt und wieviel auf dem aktuellen Gerät nebeneinander passen. Beispiel bei dem genannten Screenshot: Wenn Du den ein bisschen enger ziehst, das nur noch drei Kanäle nebeneinander passen. Dann hast Du oben Phase breit, dann drei Kanäle nebeneinander und den vierten wieder in einer Reihe breit...

Oder jemand hat nur einen 1-Panel Aufbau. Dann würde das Layout trotzdem auf dem PC die Phase in voller Breite anzeigen und das eine Panel darunter in voller Breite. Auch nicht optimal.

Mit Bootstrap Grid beißt sich einfach, dass wir nur die letzte Reihe nicht rechtsbündig haben möchten. Und das klappt bei mir irgendwie nicht. Ich wollte es (irgendwann die Woche) nochmal mit ohne Grid und nur Flex probieren.

tbnobody, 2022-07-08:
Das ist wohl richtig. Ich habe aktuell ein grid mit 2 rows gebaut. CH0 ist immer die Phase. eine Multiphasen Unterstüzung ist aktuell noch in weiter ferne, falls überhaupt möglich. Das wären die HMS Inverter die jedoch kein NRF24 Modul verwenden. 
Habe daher Channel 0 fest in eine Row gelegt und darunter die Channels 1 - 4 als columns. Du hast schon recht, bei großer Breite sind alle 4 (bei 4 Kanal) nebeneinander. Dann 3-1, 2-2 und 1-4.... Hier gefällt mir eigentlich nur das 3-1 nicht. Einen 1-Kanal Inverter mit 2 Rows könnte ich gerade noch so vertragen :-)

ahinrichs, 2022-07-08:
Wie gesagt, ich wollte Flex nochmal probieren. Alternativ wollte ich gucken, ob es nicht doch am einfachsten ist, man denkt sich ein passendes responsive Layout für 5, 4, 3, 2 Kanäle aus und macht das dann Konditional. Es ist ja nur Fleißarbeit, mal 5 Kanäle responsive anzuordnen, dann vier, usw. Und dann lässt man Vue das halt so erzeugen.

ahinrichs, 2022-07-08:
Mal sehen, ob ich ein Video posten kann, yup geht. Das hier ist noch am nächsten an meiner Idealvorstellung. Hat aber einen technischen Schönheitsfehler, man muss im HTML die Reihenfolge umdrehen. Also im DOM steht dann die Phase als letzter DIV...

https://user-images.githubusercontent.com/1071469/177984626-88c1d50f-cb7f-4dda-a389-273b2243b2b6.mp4

ahinrichs, 2022-07-08:
Ach, nochwas: das funktioniert am besten, wenn man den HomeView auf container-fluid setzt, also so breit wie möglich. Wenn man es auf container-xxl lässt, dann passen bei mir nur 3 Kanäle nebeneinaner. Hier mal Screenshots mit verschiedenen Kanalanzahlen auf maximaler Breite container-xxl.

6 Kanäle (fiktiv):
[image]

4 Kanäle:
[image]

2 Kanäle:
[image]

1 Kanal:
[image]

eeprom23, 2022-07-09:
Die Live Ansicht finde ich sehr schön, ist es möglich die Power AC werte an erster Stelle anzuzeigen und eventuell die Power DC werte für die einzelnen Strings noch einzupflegen ? Ich Danke euch für eure Tolle Arbeit !

tbnobody, 2022-07-09:
> Power DC werte für die einzelnen Strings noch einzupflegen
Es gibt für jeden String einen Power Wert. Das ist der DC Wert. AC Werte währen auf String Basis auch gar nicht verfügbar.

eeprom23, 2022-07-09:
> Power DC werte für die einzelnen Strings noch einzupflegen
> Es gibt für jeden String einen Power Wert. Das ist der DC Wert. AC Werte währen auf String Basis auch gar nicht verfügbar.

Stimmt 🙈 war mein Fehler in der Früh ! Schön wäre wie gesagt die Position der ac Leistung an erster Stelle darzustellen !

tbnobody, 2022-07-11:
Sollte sich mit 96e66dde4743a5d69de0878269f0d00ebdd0ced9 erledigt haben...

closed 2022-07-11 by tbnobody