tbnobody/OpenDTU #23: Initialize Hoymiles interface... Connection error!!

hubsi5, 2022-07-08:
Ich habe hier einen neuen ESP32 von AZ-delivery und einen nRF24L01+
Habe die Verkabelung so gemacht wie hier angegeben.
Also:
ESP             nRF
3.3V  ----  3.3V
GND ----  GND
23 ------   MOSI
19 ------   MISO
18 ------  SCK
5   ------  CSN
16  -----   IRQ
4   ------  CE

Habe schon einige nRF Module ausprobiert. An diesem kann es nicht liegen.
ESP32 habe ich leider nur diesen einen.

Verkabelung sollte richtig sein. Leider sehe ich im Serial Log, dass keine Verbindung zum Funkmodul zustande kommt.
Ist da der ESP32 defekt (obwohl nagelneu) oder beim debuggen etwas schief gelaufen (obwohl keine Fehler angezeigt wurden).

Wo könnte ich die Pinbelegung im Programm tauschen?

tbnobody, 2022-07-08:
Ich habe gerade etwas gebaut um das pin assignment für CE/CS und IRQ zu ändern. Siehe 4f2d705314c64f24d21777c4686f760ca1730444

hubsi5, 2022-07-08:
Danke für deine schnelle Hilfe!
Habe zuerst versucht die Pins zu ändern. Hat leider nichts geholfen.
Dann habe ich einmal nachgemessen ob es die 3,3V gibt.
Die gibt es, aber mein GND ist nicht auf Ground.
habe nun den anderen GND Pin genommen, jetzt funzt es.
Sieht Top aus! 

Danke!

closed 2022-07-08 by tbnobody