tbnobody/OpenDTU #25: Lan Support | ESP32-POE & MOD-NRF24LR

klahus1, 2022-07-09:
Hallo tbnobody,
vielen Dank für das Erstellen des openDTU Projektes.

Gerne würde ich folgende Erweiterungen in dein Projekt einfließen lassen:
ESP32 Hardware:
- https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware
bzw.
- https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware
o.ä.

NRF Hardware:
- https://www.olimex.com/Products/Modules/RF/MOD-NRF24LR/

Entgegen der technischen Beschreibung und des Schaltplan von Olimex, 
ist auf meinem MOD-NRF-24LR Rev.B von 2021 ein "24L01+" verbaut.
Das Initialisieren mit 250 kbps funktioniert.

[image]

Durch die Nutzung des UEXT Moduls, fallen auch Lötarbeiten etc weg.
Zusätzlich bieten die Boards auch noch einen 
"LiPo battery charger with LiPo battery connector".

Den NRF Teil (HSPI + Pins) hab ich schon mal so weit an den UEXT
Connector angepasst und mit realen HM-300 getestet.

- [x] Inverter over UEXT (HSPI)

Lan-Support anstelle von WLan sollte auch machbar sein.

- [x] MQTT over Lan
- [ ] WebApi over Lan

Nur beim WebApi über Lan hab ich aktuell noch Probleme mit einigen Browsern. 
Auf meinem IPad funktioniert alles, auf dem Linux-Notebook leider nicht.
Ich bekomme dort beim Aufruf einen wdt-timeout:
```
12:35:53.961 > E (754746) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
12:35:53.969 > E (754746) task_wdt:  - async_tcp (CPU 0/1)
12:35:53.972 > E (754746) task_wdt: Tasks currently running:
12:35:53.977 > E (754746) task_wdt: CPU 0: IDLE
12:35:53.980 > E (754746) task_wdt: CPU 1: IDLE
12:35:53.983 > E (754746) task_wdt: Aborting.
12:35:53.986 > 
12:35:53.986 > abort() was called at PC 0x40118380 on core 0
12:35:53.989 > 
12:35:53.989 > 
12:35:53.991 > Backtrace:0x400837a1:0x3ffbe9ec |<-CORRUPTED
```
Kennt jemand diesen Fehler bzw. das Verhalten?

Viele Grüße
Klaus

klahus1, 2022-07-12:
Ja, ist schon komisch. 
Ich werde in den nächsten Tagen mal mit "Wireshark" schauen, welche Unterschiede es in der Kommunikation gibt...

klahus1, 2022-07-16:
Hallo zusammen,
einen Fehler konnte ich leider nicht finden jedoch ein wenig eingrenzen.
In den Wireshark Logs ist nichts Auffälliges zu finden, die Kommunikation hört plötzlich auf.

[image]

Verbunden ist das Notebook über das interne Ethernet-Interface.

Als nächstes habe ich das eingebaute WiFi-Interface und einen externen USB-Ethernet-Adapter getestet.
Da läuft alles einwandfrei und ohne Probleme.

Selbst in der VirtualBox mit einem neueren Linux, hatte ich per eingebautem Ethernet-Interface das Problem.
Es scheint also ein Problem HW- oder Treiber-Problem zu sein, dass hoffentlich nur auf meinem alten Notebook besteht.

VG
Klaus

tbnobody, 2022-07-16:
Hallo Klaus,

ich habe auf Basis deines PR selbst schon einen Branch mit Ethernet Unterstützung lokal am laufen. Bisher konnte ich keine Probleme feststellen. Habe ein paar Dinge etwas anders Implementiert. Für den MQTT Support fehlen mir noch ein paar Anpassungen aber dann kann ich das mal freigeben.

tbnobody, 2022-07-19:
Ich habe gerade eben ein paar commits gepushed. Man kann das Environment (In der Statusleiste von VSCode) auf olimex_esp32_poe umstellen. (Achtung, es gibt für dieses Environment andere COM Port Einstellungen)

Connection handling usw. klappt alles schon wunderbar. Ich bin nur mit der Darstellung in der WebGUI noch nicht 100% fertig. Aber testen kann man schon :)

tbnobody, 2022-07-19:
Hallo Klaus,
die letzten Commits von heute haben ein neues Environment hinzugefügt "olimex_esp32_poe".
(Dies kann in VSCode in der Statusleiste umgeschaltet werden. Achtung, es gibt dafür separate Port Einstellungen)

Das Umschalten zwischen WiFi und Ethernet klappt auch schon Problemlos (hoffe ich). 
Was ich heute noch nicht ganz geschafft habe ist die Darstellung in der WebGUI. IP usw. werden vom korrekten, aktiven Interface angezeigt.. Jedoch ist noch nicht ersichtlich welches Interface aktiv ist.

Es ist zur Zeit so implementiert, dass bei aktivem Ethernet das WiFi immer deaktiviert wird bzw. anders herum.

klahus1, 2022-07-19:
Hallo Thomas,
vielen Dank für das Integrieren in dein Projekt.
Es mach sehr viel Arbeit gewissenhaft die Änderungswünschen einzubauen.
Das ist nicht selbstverständlich...
Ich hoffe, dass ich zukünftig noch etwas mehr zum openDTU beitragen kann.

Am Wochenende werde ich mal den Code zusammenführen und testen.

Viele Grüße
Klaus

stefan123t, 2022-07-25:
@klahus1 habe mir das Issue mal aufmerksam durchgelesen. 

Hier die Punkte die mir spontan dazu einfallen:
* einige der Seiten in der OpenDTU nutzen m.W. WebSocket für aktuelle Daten, hier haben evtl. ältere Browser Patchbedarf und lassen eine Verbindung offen ? 
Oder AsynTCP und ESPAsyncWebServer kommen mit ein paar TCP Signalen SYN, ACK, etc. nicht klar die hier vom Browser nach einer evtl. veralteten WebSocket Spec verwendet wurden/werden.

* Dein TCPDump mit Wireshark deutet für mich darauf hin, dass der ESP (openDTU-eth.fritz.box 192.168.1.39 ?) seinen ARP Cache aktualisieren möchte um die IP 192.168.1.211 bzw. 192.168.1.111 (Notebook ?) zu erfahren. 
Das Notebook antwortet auch mit der MAC Adresse 08:62:66:b6:20:b7 für 192.168.1.111.
Die anderen Source / Destinations sind nur als Namen angegeben das macht es etwas schwierig das wirklich nachzuvollziehen.
Warum gibt es zwei IP Adressen ..211 und ..111 ? 
Was macht der TCP Stack auf dem Olimex ETH Device (Version?) bzw. dem ESP WiFi zu diesem Zeitpunkt -> Serial Logs ?
Eventuell hat der ESP auch eine Verbindung mit WiFi und ETH gleichzeitig hergestellt und er muss plötzlich als Bridge arbeiten ?

@tbnobody kannst Du das Problem bei Dir mit Deiner Hardware überhaupt nachstellen ?

tbnobody, 2022-07-25:
Habe hier das gleiche Olimex Board und habs mit mehreren Browsern (allerdings alles neue Versionen) getestet. Konnte keine Fehler feststellen bisher.

stefan123t, 2022-07-26:
@klahus1 kannst Du mal Deine Browser Version posten oder willst Du das Issue als Browser abhängig schließen?
Wie gesagt ich könnte mir auch noch vorstellen dass Dein ESP Irgendwie in den Bridge Modus geraten ist und das Routing nicht mehr klappt ?

klahus1, 2022-07-26:
Hallo zusammen,
da ich das Problem nur auf einem alten Notebook (Linux) habe, kann ich gerne das issue schließen.

closed 2022-07-26 by klahus1

klahus1, 2022-07-26:
Noch ein kleiner Nachtrag, 

> * Warum gibt es zwei IP Adressen ..211 und ..111 ?

...111 ist mein Notebook, ...211 ist der MQTT-Broker.

Da ich nur den Traffic meines Notebook lokal mitgeschrieben habe und **nicht** remote auf den Switchen / Routern "sieht" mein Notebook zwar den ARP Request des ESP, jedoch keinen weiteren TCP/IP Traffic der anderen Teilnehmer.