tbnobody/OpenDTU #29: connection problems

braini75, 2022-07-13:
Hallo,

thank you very much for this awsome project and all the work you've out into it! I really enjoy this tool.

Unfortunately I have 2 connection problems: 
1. Sometimes the connection to the inverter is lost. 
Error pattern: 
 - Data age increases
 - no new data comes in
 - Settings -> Inverter Settings -> Inverter Type is set to unkown
Workaround:
 - open edit (pen) the inverter settings and just save settings
 -> everthing works

2. Web access not working every morning. No new values ​​arrive at the MQTT either. Only when I press the reset button on the esp32 everything works again.

tbnobody, 2022-07-13:
Hello,

could you please paste any serial output of the specific time? If the serial number is shown as unknown means that the serial of the inverter is invalid. this should not occour during runtime. if it's once detected it should be always the same.

braini75, 2022-07-14:
Hello,
thanks for your quick response. I just restartet the ESP32 and do logging of serial output. Let you know once I have those values ;)

tbnobody, 2022-07-14:
It would also be interesting which version you are using. Firmware Version + Git Hash. You can find both values in Info --> System dialog.

tbnobody, 2022-07-14:
@pangamut, is this the first version of this sofware that you are using? Or did you also run a previous version? 
It would be interesting if e.g. this problem also occours with git hash 96e66dde4743a5d69de0878269f0d00ebdd0ced9

(`git checkout 96e66dde4743a5d69de0878269f0d00ebdd0ced9`)

In the version after this commit I changed some stuff regarding the parser.

pangamut, 2022-07-14:
I did user previous versions, and experienced similar problems with them.  What puzzles me, is that it works fine for hours or even days, then i get this issue. Saving settings again, everythings fine. Like some bits flip over time.

tbnobody, 2022-07-14:
Based on your screenshot I see that your inverter does not have a name. Have you entered a name and the value is gone after the corruption or did you just leave the field empty?

pangamut, 2022-07-14:
Oh, i do have a name ("Balkon"), but you are right, its not displayed in the screenshot/error situation. I didn't notice that yet, actually.

pangamut, 2022-07-14:
I have one HM-600, serial begins with 1141. I switch power off on sunset, and on on sunrise, so uptime this afternoon was about 10hrs.

tbnobody, 2022-07-14:
Could you please also paste the information regarding hardware and memory usage? (I just want to see if there are differences regarding heap size etc.)

I will also apply some patches to the source code to double check several package length and also outpput error messages if this conditions occour. If it occurs you should find something like "FATAL: (" in the serial log. 

If you run the serial monitor via the Visual Studio PlatformIO extension the serial log should also be logged to a file.

pangamut, 2022-07-14:
i can run a serial monitor tomorrow.

[image]

braini75, 2022-07-14:
My system output looks the same, except my Git Hash is g608456b.
My Inverter is HM-1500 serial 1161xxxxxxxx.
Since this morning everything is running flawless. But I never got more than 24h running time so far.

tbnobody, 2022-07-14:
Are you using the MQTT support? What NTP Server settings are you using?
(Just in case that I cannot reproduce that problem within the next 24 or 48hr I want to know what's maybe the different in the setups)

pangamut, 2022-07-15:
Yes, i use mqtt with a local server, NTP server is de.pool.ntp.org. I try to have it on a serial console today. unfortunatly, the error happens quite sporadic.

braini75, 2022-07-15:
I'm using mqtt and default ntp server pool.ntp.org.
My monitoring last night didn't worked :( Somehow the Computer I used for monitoring had rebooted at night. Now I started monitoring again with a Raspberry Pi connected. ;)

tbnobody, 2022-07-15:
I added a check in c6499e09bd18672e5283106feb72cd899e6ad3c4
You should try the latest source. Based on the current experience this is maybe a problem in the inverters firmware as a zero packet should not happen. But on the other hand 2 CRC checks of this packet succeeded.

braini75, 2022-07-16:
Thanks! That was fast WOW!
I just updated my ESP and will test your implementation.

I also got an idea about Point 2 of my issue:
_> Web access not working every morning. No new values ​​arrive at the MQTT either. Only when I press the reset button on the esp32 everything works again._

I think the ESP will not reconnect WIFI connection, when it got lost some time in between! Because the device was logging to serial output this morning as normal, but WIFI connection was lost midnight:
`Sa 16. Jul 00:00:05 BST 2022 WiFi lost connectionSa 16. Jul 00:00:05 BST 2022 WiFi lost connection`

Every night I reset my Wifi for parental control reasons. So it is offline for few seconds. ;) But ESP will not reconnect...

tbnobody, 2022-07-16:
That's very strange. The Arduino Framework offers a setAutoReconnect method which is true by default. But it seems to do nothing in case of ESP32 (https://github.com/espressif/arduino-esp32/search?q=_autoReconnect)

I will doublecheck several things and maybe implement a reconnect by my own.

ahinrichs, 2022-07-16:
It does [something](https://github.com/espressif/arduino-esp32/blob/cf01523ded6d40e6feaa92b9aa647e5553148b60/libraries/WiFi/src/WiFiSTA.cpp line 973ff):

I did had similar problems with another project that are gone now. I will check that too.

tbnobody, 2022-07-16:
You are right. I was only looking for the internal variable.
But the reconnect feature depends on the following disconnect reasons:

`if((reason == WIFI_REASON_AUTH_EXPIRE) || (reason >= WIFI_REASON_BEACON_TIMEOUT && reason != WIFI_REASON_AUTH_FAIL))
`

Question is, what is a Fritzbox or Speedport doing when disconnecting it's clients. Because if I look at the reason codes >= BEACON_TIMEOUT:

```
    WIFI_REASON_BEACON_TIMEOUT           = 200,
    WIFI_REASON_NO_AP_FOUND              = 201,
    WIFI_REASON_AUTH_FAIL                = 202,
    WIFI_REASON_ASSOC_FAIL               = 203,
    WIFI_REASON_HANDSHAKE_TIMEOUT        = 204,
    WIFI_REASON_CONNECTION_FAIL          = 205,
    WIFI_REASON_AP_TSF_RESET             = 206,
    WIFI_REASON_ROAMING                  = 207,
```

it should reconnect e.g. when the AP was not found.

ahinrichs, 2022-07-16:
Were having kind of the same ideas. I just researched and the reason gets logged with log_w. @braini75 Could you add a `esp_log_level_set("wifi", ESP_LOG_WARN);` in `src/main.cpp`, maybe line 55 where WiFi gets initialized and check, whether the logs shows something more helpful tomorrow.

sivar2311, 2022-07-16:
Afaik you have to call ```WiFi.setAutoReconnect(true);``` after the connection has been established successfully. Calling it before doesn't work.

braini75, 2022-07-17:
I just did a restart of my Wifi. Following log was generated:

```
**13:49:08.259 > WiFi lost connection**
13:49:08.262 > Disconnected from MQTT.
13:49:08.264 > Websocket: [/livedata][2] disconnect
13:49:08.991 > Fetch inverter: 114181025555
13:49:08.994 > TX 15 81 2 55 55 78 56 34 12 80 B 0 62 D3 F7 34 0 0 0 5 0 0 0 0 D2 61 D1 
13:49:09.025 > Interrupt received
13:49:09.027 > RX 95 81 2 55 55 81 2 55 55 1 0 1 1 39 3 70 A C0 1 18 3 C7 A 8F 0 0 4C 
13:49:09.065 > Interrupt received
13:49:09.067 > RX 95 81 2 55 55 81 2 55 55 2 3 EA 0 0 4 78 3 9A 4 2B 9 42 13 89 14 5A 2B 
13:49:09.126 > Interrupt received
13:49:09.129 > RX 95 81 2 55 55 81 2 55 55 83 0 0 0 DC 3 E8 2 D 0 C 87 88 2D 
13:49:09.196 > RX Period End
13:49:09.197 > Success
13:49:13.991 > Fetch inverter: 116181100241
13:49:13.994 > TX 15 81 10 2 41 78 56 34 12 80 B 0 62 D3 F7 39 0 0 0 5 0 0 0 0 42 39 45 
13:49:14.060 > Interrupt received
13:49:14.062 > RX 95 81 10 2 41 81 10 2 41 1 0 1 1 43 2 73 2 75 7 E7 7 ED 0 0 2F D6 22 
13:49:14.080 > Interrupt received
13:49:14.081 > RX 95 81 10 2 41 81 10 2 41 2 0 0 45 13 6 52 6 C8 1 44 2 78 2 75 7 FC E8 
13:49:14.141 > Interrupt received
13:49:14.143 > RX 95 81 10 2 41 81 10 2 41 3 7 F1 0 0 1D CB 0 0 4C 14 6 60 6 D7 9 3E 6E 
13:49:14.181 > Interrupt received
13:49:14.183 > RX 95 81 10 2 41 81 10 2 41 84 13 89 1E 2B 0 E1 1 46 3 E8 2 96 0 6 C 21 4C 
13:49:14.212 > RX Period End
13:49:14.214 > Success
13:49:18.992 > Fetch inverter: 114181025555
13:49:18.995 > TX 15 81 2 55 55 78 56 34 12 80 B 0 62 D3 F7 3E 0 0 0 5 0 0 0 0 72 1F 5 
13:49:19.120 > Interrupt received
13:49:19.122 > RX 95 81 2 55 55 81 2 55 55 83 0 4 0 CE 3 E8 2 C 0 C 33 6D 6B 
13:49:19.207 > RX Period End
13:49:19.208 > Middle missing
13:49:19.210 > Request retransmit: 1
13:49:19.212 > TX 15 81 2 55 55 78 56 34 12 81 1F 
13:49:19.241 > Interrupt received
13:49:19.243 > RX 95 81 2 55 55 81 2 55 55 1 0 1 1 39 3 3A A 18 1 19 3 80 9 D4 0 0 C0 
13:49:19.273 > RX Period End
13:49:19.275 > Middle missing
13:49:19.277 > Request retransmit: 2
13:49:19.279 > TX 15 81 2 55 55 78 56 34 12 82 1C 
13:49:19.325 > Interrupt received
13:49:19.327 > RX 95 81 2 55 55 81 2 55 55 2 3 EA 0 0 4 78 3 9A 4 2B 9 3E 13 89 13 6 C 
13:49:19.342 > RX Period End
13:49:19.344 > Success
**13:49:23.508 > Disable search for AP... done**
13:49:23.994 > Fetch inverter: 116181100241
13:49:23.997 > TX 15 81 10 2 41 78 56 34 12 80 B 0 62 D3 F7 43 0 0 0 5 0 0 0 0 20 22 46 
13:49:24.035 > Interrupt received
13:49:24.038 > RX 95 81 10 2 41 81 10 2 41 1 0 1 1 42 1 C4 1 C4 5 AF 5 AE 0 0 2F D6 2E 
13:49:24.097 > Interrupt received
13:49:24.099 > RX 95 81 10 2 41 81 10 2 41 2 0 0 45 13 6 52 6 C8 1 42 1 C6 1 C1 5 B5 AF 
13:49:24.155 > Interrupt received
13:49:24.157 > RX 95 81 10 2 41 81 10 2 41 3 5 A7 0 0 1D CC 0 0 4C 15 6 61 6 D8 9 3A 36 
13:49:24.196 > Interrupt received
13:49:24.198 > RX 95 81 10 2 41 81 10 2 41 84 13 89 15 97 0 DE 0 EA 3 E7 2 96 0 6 F2 10 A9 
13:49:24.218 > RX Period End
13:49:24.219 > Success
13:49:28.995 > Fetch inverter: 114181025555
13:49:28.997 > TX 15 81 2 55 55 78 56 34 12 80 B 0 62 D3 F7 48 0 0 0 5 0 0 0 0 10 51 5F 
13:49:29.047 > Interrupt received
13:49:29.048 > RX 95 81 2 55 55 81 2 55 55 1 0 1 1 39 1 C5 5 86 1 1A 1 EF 5 71 0 0 6B 
13:49:29.105 > Interrupt received
13:49:29.107 > RX 95 81 2 55 55 81 2 55 55 2 3 EB 0 0 4 79 3 9B 4 2C 9 39 13 89 A 79 6B 
13:49:29.146 > Interrupt received
13:49:29.147 > RX 95 81 2 55 55 81 2 55 55 83 0 2 0 72 3 E8 2 C 0 C 3C 0 B3 
13:49:29.221 > RX Period End
13:49:29.222 > Success
13:49:33.996 > Fetch inverter: 116181100241
13:49:33.999 > TX 15 81 10 2 41 78 56 34 12 80 B 0 62 D3 F7 4D 0 0 0 5 0 0 0 0 40 6E 64 
13:49:34.041 > Interrupt received
13:49:34.043 > RX 95 81 10 2 41 81 10 2 41 1 0 1 1 40 1 73 1 77 4 A3 4 B0 0 0 2F D7 3B 
13:49:34.079 > Interrupt received
13:49:34.081 > RX 95 81 10 2 41 81 10 2 41 2 0 0 45 14 6 53 6 C9 1 40 1 78 1 78 4 B4 AD 
13:49:34.140 > Interrupt received
13:49:34.143 > RX 95 81 10 2 41 81 10 2 41 3 4 B5 0 0 1D CC 0 0 4C 15 6 61 6 D8 9 3F 20 
13:49:34.183 > Interrupt received
13:49:34.186 > RX 95 81 10 2 41 81 10 2 41 84 13 89 11 CC 0 DA 0 C1 3 E7 2 96 0 6 C 59 6E 
13:49:34.203 > RX Period End
13:49:34.205 > Success
```
Wifi will connect only. if I push the ESP32 reset button. You should be able to reconstruct the issue. Just restart you wifi. ;)

ahinrichs, 2022-07-17:
I'm sorry, but my recommendation was not effective. I did find an old esp here and could do some tests here. Here are my findings:
- autoReconnect is true by default (as @tbnobody said)
- when I restart wifi on my AP, my ESP shows disconnect with reason 2 WIFI_REASON_AUTH_EXPIRE, but reconnect is not triggerd
- when I - redundantly - set `WiFi.setAutoReconnect(true)` after connect and then restart wifi on my router it shows disconnect reason 3 WIFI_REASON_AUTH_LEAVE. (Tripple checked that, because that does not make any sense: i set autoReconnect to true that is already true changed the disconnect reason... no idea)

So there must be something different, that prevents the esp reaching the cited reconnect code.

@braini75 : Could you please revert the suggested line and add this 
```
WiFiEventId_t eventID = WiFi.onEvent([](WiFiEvent_t event, WiFiEventInfo_t info){
  Serial.print("WiFi lost connection. Reason: ");
  Serial.println(info.wifi_sta_disconnected.reason);
}, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_DISCONNECTED);
```
at the end of `WiFiSettingsClass::setupMode()`, that is after line 36 in `src/WiFiSettings.cpp`. That worked here and gave the numeric reason on serial log.

ahinrichs, 2022-07-17:
> Just restart you wifi.

Can you please check with my kids, that I need to "just restart the wifi" several times on sunday :-)

braini75, 2022-07-17:
Thanks! Don't touch the Wifi :)

I got Reason 3:

```
23:29:36.137 > Nothing received, resend count exeeded
23:29:37.041 > WiFi lost connection. Reason: 3
23:29:37.045 > WiFi lost connection
23:29:37.047 > Disconnected from MQTT.
```

ahinrichs, 2022-07-17:
More infos here. Firmware with CORE_DEBUG=5 gives (ignore HoyMiles error):
```
23:01:03.127 > Starting OpenDTU
23:01:03.127 > Initialize FS... done
23:01:03.127 > Reading configuration... done
23:01:03.127 > Initialize WiFi... [    94][W][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
23:01:03.127 > done
23:01:03.127 > Setting Hostname... done
23:01:03.127 > Configuring WiFi STA using [   183][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
23:01:03.127 > new credentials... [   184][W][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
23:01:03.127 > [   197][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
23:01:03.127 > done
23:01:03.127 > Initialize NTP... done
23:01:03.127 > Initialize MqTT... done
23:01:03.127 > Initialize WebApi... done
23:01:03.127 > Initialize Hoymiles interface... Connection error!!
23:01:03.127 > done
23:01:03.127 > [   251][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: IoT, BSSID: e2:94:f6:xx:xx:xx, Channel: 1, Auth: WPA2_PSK
23:01:03.127 > [   252][W][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
23:01:03.127 > [   286][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.1.33
23:01:03.127 > [   286][W][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
23:01:03.127 > [   289][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.1.33, MASK: 255.255.255.0, GW: 192.168.1.254
23:01:03.127 > WiFi connected
23:01:41.108 > [ 38785][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: IoT, BSSID: e2:94:f6:xx:xx:xx, Reason: 2
23:01:41.119 > [ 38786][W][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
23:01:41.127 > [ 38793][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
23:01:41.134 > [ 38800][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
23:01:41.140 > [ 38808][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
23:01:41.152 > WiFi lost connection. Reason: 2
23:01:41.155 > WiFi lost connection
23:03:41.107 > [158787][V][WiFiGeneric.cpp:309] _arduino_event_cb(): STA IP Lost
23:03:41.113 > [158788][W][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 9 - STA_LOST_IP
```

And that was it. Waited 10 minutes.

So, when the ESP is kicked out and happens to get reason 2, then the reconnect is triggered in line 889 independently from autoReconnect setting, because `first_connect` is true at least one time.
```
if(first_connect && ((reason == WIFI_REASON_AUTH_EXPIRE) ||
  (reason >= WIFI_REASON_BEACON_TIMEOUT)))
{
  log_d("WiFi Reconnect Running");
  WiFi.disconnect();
  WiFi.begin();
  first_connect = false;
}
else if(WiFi.getAutoReconnect()){
...
```

`Wifi.disconnect(); WiFi.begin();` is called at least once but does not connect. I'm not sure the Wifi gets fully started, because I would expect this to result in some WiFiEvents (timeout, ... something).

ahinrichs, 2022-07-17:
> I got Reason 3:

As I read the code then reason 3 will not trigger any attempt to reconnect. Unfortunately I haven't found any pattern when or why the reason 2 or 3 occur. Here simply sometimes it gives reason 2, sometimes 3.

tbnobody, 2022-07-19:
I am currently rewriting the WiFiSettingsClass to support multiple interfaces. 
My first try will be to just call WiFi.reconnect() in the ARDUINO_EVENT_WIFI_STA_DISCONNECTED event.

ahinrichs, 2022-07-19:
Yes, actually I would suggest the same. I've not found a predictive way how to improve the built in reconnect. 

Beside this I still wonder, why the first `WiFi.begin()` does not succeed. In most of my tests the wifi was already restarted and reachable, when it was called.

tbnobody, 2022-07-19:
I've just commited an updated version of the connection handling. I've tested it several times by kicking out the client of the WiFi AP. Reconnect worked perfectly. (I didn't power off my WiFi because of a redundant setup)

tbnobody, 2022-07-19:
@braini75 ist eigentlich Problem 1 nochmal aufgetreten?

pangamut, 2022-07-20:
> @braini75 ist eigentlich Problem 1 nochmal aufgetreten?

Bei mir läuft die c6499e0 seit Veröffentlichung ohne Fehler durch. Danke!

braini75, 2022-07-20:
Problem 1 scheint tatsächlich behoben zu sein! Vielen Dank @tbnobody !!!

Problem 2 habe ich bei mir Lokal einen fix in der Wifisettings.cpp gemacht, so dass es auch funktioniert. Aber ich habe gerade festgestellt, dass tbnobody scheinbar schon überarbeitet und eine neue Klasse bereitstellt. Bin gespannt!

tbnobody, 2022-08-10:
Closed from my side... If one of the errors still exist, be free to open this issue again.

commit c6499e09 (2022-07-15): Ignore fragment number zero. Should fix #29

It seems that some inverters send packages with fragment number zero and a valid CRC. Must ignore such packets!
closed 2022-08-10 by tbnobody