tbnobody/OpenDTU #30: No Values in Irradiation

Hans-JR, 2022-07-14:
Moin,
erstmal vielen Dank für die tolle Arbeit.
Ich habe heute mal die aktuelle Version 01.17 geflasht. Soweit hat alles geklappt. Aufgefallen ist mir, dass das Feld für die Irradiation den Wert 0.00% anzeigt. Ich habe vorher den Adapter von ahoy getestet. Da wurde der Wert angezeigt.
Mein WR ist ein HM-300.
[image]
[image]

braini75, 2022-07-14:
Moin,
hast du unter Settings -> Inverter Settings -> Edit Inverter... den Max power Wert eingetragen?

tbnobody, 2022-07-14:
Hallo,

es sieht so aus als hättest du bei den Inverter Eingängen keinen Maximalwert konfiguriert: (siehe: https://github.com/tbnobody/OpenDTU/blob/master/docs/screenshots/13_InverterSettings.png)

Hans-JR, 2022-07-14:
Das wars. Jetzt wird der Wert angezeigt. Ich hatte mich natürlich an das o. g. Bild gehalten. Da sind leider keine Werte eingetragen. So hab ich das dann auch gemacht.
Danke. Falls mir noch was auffällt, werde ich es berichten. 
Gruß
Hans-Jürgen

closed 2022-07-14 by Hans-JR