tbnobody/OpenDTU #33: Feature: POC Home Assistant MQTT Discovery

itn3rd77, 2022-07-16:
Hi @tbnobody,

wie schon mal angekündigt habe ich ein POC für [Home Assistant MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery/) erstellt.
Damit wird dann ein Inverter in Home Assistant automatisch als Gerät erkannt und mit allen verfügbaren Sensoren eingerichtet ohne das der Benutzer einer manuelle Konfiguration erstellen muss.

Es gibt auch noch weitere Heimautomatisierungen, welche MQTT Discovery unterstützen und die Informationen in einer definierten Struktur erwarten. Es wäre also sinnvoll einer modulare Lösung zu schaffen, so dass bei Bedarf einfach weitere Implemenationen hinzugefügt werden können.

So sieht das dann in Home Assistant aus:

[image]

Ich habe für den POC einfach mal die MqttPublishing.cpp misbraucht. Deshalb kommt auch kein PR und ich brauche natürlich Deine Unterstützung und Zustimmung openDTU diesbezüglich zu erweitern.

Folgende Punkte wären zu klären/erweitern:

- Die MQTT Konfiguration über die UI muss eine Option haben zu wählen welche Strategie angewand werden soll. D.h. Default, Home Assistant, etc.. Evtl. müssen auch neue andere Optionen eingeblendet werden, je nach Auswahl.
- Im Backend muss je nach gewählter MQTT Strategie eine passende "MqttPublishing" angesteuert werden.

Ich kopiere mal meine angepasste MqttPublishing.cpp hier rein, damit Du mal einen Eindruck bekommst was ich verbrochen habe. Sind auch ein paar Kommentare was noch gelöst werden muss.
```
#include "ArduinoJson.h"
#include "MqttPublishing.h"
#include "MqttSettings.h"

MqttPublishingClass MqttPublishing;

void MqttPublishingClass::init()
{
}

void MqttPublishingClass::loop()
{
    if (!MqttSettings.getConnected() && Hoymiles.getRadio()->isIdle()) {
        return;
    }

    CONFIG_T& config = Configuration.get();

    if (millis() - _lastPublish > (config.Mqtt_PublishInterval * 1000)) {
        //MqttSettings.publish("dtu/uptime", String(millis() / 1000));
        //MqttSettings.publish("dtu/ip", WiFi.localIP().toString());

        // Loop all inverters
        for (uint8_t i = 0; i < Hoymiles.getNumInverters(); i++) {
            auto inv = Hoymiles.getInverterByPos(i);

            char buffer[sizeof(uint64_t) * 8 + 1];
            sprintf(buffer, "%0lx%08lx",
                ((uint32_t)((inv->serial() >> 32) & 0xFFFFFFFF)),
                ((uint32_t)(inv->serial() & 0xFFFFFFFF)));
            String serial = String(buffer);

            // MqttSettings.publish(subtopic + "/name", inv->name());

            String subtopic = "sensor/dtu_" + serial;

            uint32_t lastUpdate = inv->getLastStatsUpdate();
            if (lastUpdate > 0 && lastUpdate != _lastPublishStats[i]) {
                _lastPublishStats[i] = lastUpdate;

                // Loop all channels
                for (uint8_t c = 0; c <= inv->Statistics()->getChannelCount(); c++) {
                    publishField(subtopic, inv, c, FLD_UDC);
                    publishField(subtopic, inv, c, FLD_IDC);
                    if (c == 0) {
                        publishField(subtopic, inv, c, FLD_PDC, "PowerDC");
                    } else {
                        publishField(subtopic, inv, c, FLD_PDC);
                    }
                    publishField(subtopic, inv, c, FLD_YD);
                    publishField(subtopic, inv, c, FLD_YT);
                    publishField(subtopic, inv, c, FLD_UAC);
                    publishField(subtopic, inv, c, FLD_IAC);
                    publishField(subtopic, inv, c, FLD_PAC);
                    publishField(subtopic, inv, c, FLD_F);
                    publishField(subtopic, inv, c, FLD_T);
                    publishField(subtopic, inv, c, FLD_PCT);
                    publishField(subtopic, inv, c, FLD_EFF);
                    publishField(subtopic, inv, c, FLD_IRR);
                }
            }

            yield();
        }

        _lastPublish = millis();
    }
}

void MqttPublishingClass::publishField(String subtopic, std::shared_ptr<InverterAbstract> inv, uint8_t channel, uint8_t fieldId, String topic)
{
    if (inv->Statistics()->hasChannelFieldValue(channel, fieldId)) {
        CONFIG_T& config = Configuration.get();

        String chanName;
        String chanNameLower;

        if (topic == "") {
            chanName = inv->Statistics()->getChannelFieldName(channel, fieldId);
        } else {
            chanName = topic;
        }

        chanNameLower = chanName;
        chanNameLower.toLowerCase();
        
        char buffer[sizeof(uint64_t) * 8 + 1];
        sprintf(buffer, "%0lx%08lx",
            ((uint32_t)((inv->serial() >> 32) & 0xFFFFFFFF)),
            ((uint32_t)(inv->serial() & 0xFFFFFFFF)));
        String serial = String(buffer);

        // Home Assistant configuation topic payload (JSON)
        StaticJsonDocument<1024> jsonConfigurationPayload;
        jsonConfigurationPayload["uniq_id"] = "dtu_" + serial + "_" + chanNameLower + "_" + String(channel);
        jsonConfigurationPayload["obj_id"] = "dtu_" + serial + "_" + chanNameLower + "_" + String(channel);
        jsonConfigurationPayload["name"] = (channel == 0) ? chanName : chanName + "_" + String(channel);
        jsonConfigurationPayload["unit_of_meas"] = inv->Statistics()->getChannelFieldUnit(channel, fieldId);
        jsonConfigurationPayload["val_tpl"] = "{{ value_json.value }}";
        jsonConfigurationPayload["stat_t"] = String(config.Mqtt_Topic) + "/" + subtopic + "/" + chanNameLower + "_" + String(channel) + "/state";
        
        // Create device object
        JsonObject jsonDeviceObject = jsonConfigurationPayload.createNestedObject("device");
        jsonDeviceObject["name"] = String(inv->name());
        jsonDeviceObject["mdl"] = String(inv->typeName());
        jsonDeviceObject["mf"] = "Hoymiles";
        jsonDeviceObject["cu"] = "http://" + WiFi.localIP().toString();
        JsonArray idsJsonArray = jsonDeviceObject.createNestedArray("ids");
        idsJsonArray.add(serial);       

        if (channel > 0) {
            jsonConfigurationPayload["ent_cat"] = "diagnostic";
            jsonConfigurationPayload["enabled_by_default"] = "false";
        }

        // TODO: How to solve this?
        String unitOfMeasurement = inv->Statistics()->getChannelFieldUnit(channel, fieldId);

        if (unitOfMeasurement == "A") {
            jsonConfigurationPayload["dev_cla"] = "current";
            jsonConfigurationPayload["stat_cla"] = "measurement";
        } else if (unitOfMeasurement == "V") {
            jsonConfigurationPayload["dev_cla"] = "voltage";
            jsonConfigurationPayload["stat_cla"] = "measurement";           
        } else if (unitOfMeasurement == "W") {
            jsonConfigurationPayload["dev_cla"] = "power";
            jsonConfigurationPayload["stat_cla"] = "measurement";   
        } else if (unitOfMeasurement== "Wh") {
            jsonConfigurationPayload["dev_cla"] = "energy";
            jsonConfigurationPayload["stat_cla"] = "total_increasing"; 
        } else if (unitOfMeasurement == "kWh") {
            jsonConfigurationPayload["dev_cla"] = "energy";
            jsonConfigurationPayload["stat_cla"] = "total_increasing"; 
        } else if (unitOfMeasurement == "Hz") {
            jsonConfigurationPayload["dev_cla"] = "frequency";
            jsonConfigurationPayload["stat_cla"] = "measurement"; 
        } else if (unitOfMeasurement == "°C") {
            jsonConfigurationPayload["dev_cla"] = "temperature";
            jsonConfigurationPayload["stat_cla"] = "measurement"; 
        }

        String configurationPayload = "";
        serializeJson(jsonConfigurationPayload, configurationPayload);

        // Publish configuration topic.
        // TODO: Only send once e.g. on device start
        MqttSettings.publish(subtopic + "/" + chanNameLower + "_" + String(channel) + "/config", configurationPayload);

        // Update Home Assistant sensor with new measurement
        MqttSettings.publish(subtopic + "/" + chanNameLower + "_" + String(channel) + "/state", "{ \"value\": \"" + String(inv->Statistics()->getChannelFieldValue(channel, fieldId)) + "\"}");
    }
}
```

Also wenn Du Zeit und Lust hast würde ich mich freuen, wenn wir das angehen könnten. Bin natürlich auch bereit den Home Assistant Teil dann später zu pflegen bei Problemen und zu supporten.

Freue mich auf Deine Rückemeldung!

Danke und Gruß
Ingo

tbnobody, 2022-07-16:
Hallo Ingo,

ich finde die Idee gut HASS Support einzubauen. 
Man müsste dafür nicht mal meine aktuelle Topic Struktur ändern. Es braucht nur zusätzlich Einträge in der Auto-Discovery Topic von HASS.

zigbee2mqtt oder espurna macht sowas:
[image]

`{"dev":{"ids":["espurna_f93b65"],"name":"SolarPanel","sw":"1.15.0-dev.gitc95ea71c","mf":"BLITZWOLF","mdl":"BWSHPX_V23"},"uniq_id":"espurna_f93b65_current_0","name":"SolarPanel current 0","stat_t":"Garden/Terrace/SolarPanel/current","unit_of_meas":"A"}`

Dadurch hat man keine JSON values in standard MQTT Topics (die gehören sich da nämlich nicht hin) und HASS bekommt trotzdem was es will.

tbnobody, 2022-07-16:
Unter espurna schauen die HASS config parameter einfach so aus:
[image]

das könnte ich mir für OpenDTU auch so vorstellen.

itn3rd77, 2022-07-17:
Hi, hört und sieht sich gut an!

Irgendwie hing ich gedanklich fest, das ich die eigentlichen Messwerte auch bei den Sensor packen muss. Aber klar macht Sinn die Werte so zu lassen und einfach nur die Auto-Discovery Topics anzulegen.

Hast Du Zeit das zu machen? Wie kann ich unterstützen?

Danke und Gruß
Ingo

tbnobody, 2022-07-18:
Ich hab da mal ein wenig was gebaut: 1124a9aaa13bdf31d215d3586b18d69dd662d52b

In HASS wird das erstmal korrekt angezeigt usw. Ob auch Werte kommen (wovon ich mal ausgehe) wird sich morgen zeigen wenn Licht vorhanden ist.

[image]

itn3rd77, 2022-07-19:
Hi und guten Morgen!

Habe mein Baby gleich mal mit dem Update versehen und ICH BIN BEGEISTERT 🤩
Wirklich super klasse in welcher Geschwindigkeit Du das hinzauberst - Chapeau!

Besten Dank und viele Grüße
Ingo

closed 2022-07-19 by itn3rd77

jayefjay, 2022-08-09:
Hallo, das auto discovery funktioniert in Home Assistant wunderbar, allerdings ist mit dem konkreten Setup die Nutzung im Energy Dashboard (https://www.home-assistant.io/blog/2021/08/04/home-energy-management/) nicht möglich da der Sensor hier nicht aufscheint. 

Folgende Message kommt bei Home Assistent für die aktuelle Power an:

{
    "name": "HM800 Power",
    "stat_t": "solar/114181014957/0/power",
    "unit_of_meas": "W",
    "uniq_id": "114181014957_ch0_Power",
    "dev": {
        "name": "HM800",
        "ids": "114181014957",
        "cu": "http://192.168.1.44",
        "mf": "OpenDTU",
        "mdl": "HM-600, HM-700, HM-800",
        "sw": "g725d482"
    },
    "exp_aft": 10,
    "dev_cla": "power",
    "stat_cla": "measurement"
}

Das Problem scheint die gewählte device_class "power" zu sein, um automatisch für die Energiestatistiken in Home Assistant zu funktionieren müsste diese wohl "energy" sein.

tbnobody, 2022-08-09:
Hm sicher das man da den aktuellen Verbrauch und nicht YieldTotal o.ä. haben möchte? Power gibt ja nur den aktuellen Ertrag an. YieldTotal die gibt erzeugten kWh zurück (die HASS auch auf Tage/Monate etc. aufteilen/kumulieren kann).
Habe es auch hier gerade getestet. YieldTotal wird angezeigt und lässt sich im EnergieDashboard verwenden.