tbnobody/OpenDTU #36: Irradiation is always zero

b-a-9-0, 2022-07-21:
Hi, 
I have OpenDTU running with 1x HM-1200 connected. Everythink works fine so far, except the irradiation value is always zero (see attachted picture). In my setup, there is only one PV module connected to the inverter. Irradiation is zero indepentend from current DC power.

[image]

Thanks for this amazing project!

b-a-9-0, 2022-07-21:
used git hash: 0abd05c7ce28ea0fbb216524432a6211412dd303

tbnobody, 2022-07-21:
Please go to the editor edit window and enter the max power of your PV module in the right channel as shown here: https://github.com/tbnobody/OpenDTU/blob/master/docs/screenshots/13_InverterSettings.png

b-a-9-0, 2022-07-21:
Thanks, never saw these fields before!

closed 2022-07-21 by b-a-9-0

tbnobody, 2022-07-21:
Due to the fact that you where the second user who didn't see the addtional parameters I added a hint:
[image]

b-a-9-0, 2022-07-21:
Perfect!