tbnobody/OpenDTU #38: Keine verbindung zum OpenDTU AP möglich

G1nac, 2022-07-23:
Moin, kriege einfach keine Verbindung mit dem openDTU AP hin, sagt immer nur keine Verbindung möglich. Sind meine ersten Erfahrungen mit einem ESP32.
Das Terminal sagt denke ich alles i.O ?

- 18:13:09.077 > Starting OpenDTU
- 18:13:09.081 > Initialize FS... done
- 18:13:09.081 > Reading configuration... done
- 18:13:09.097 > Initialize Network... done
- 18:13:09.219 > Setting Hostname... Initialize NTP... done
- 18:13:09.223 > Initialize MqTT... done
- 18:13:09.227 > Initialize WebApi... done
- 18:13:09.227 > Initialize Hoymiles interface... Connection successfull
- 18:13:09.244 > done
- 18:13:09.244 > Switch to WiFi mode
- 18:13:09.244 > Setting Hostname... done
- 18:13:25.048 > Disable search for AP... done

Könnte es an der Pin-Belegung liegen?
Habe ein ESP32 mit 30 Pins die Belegung habe ich mir aus Diversen Schemata zusammengereimt.
Das habe ich in die Platformio.ini eingetragen, nur die Zahlen nicht die Buchstaben und die NRF24L01+ auch so angeschlossen.
- CE - D4 
- CS - D5
- SCK - D18
- MOSI - D23
- MISO - D19
- IRQ - RX2 16?

Habe ich irgendwo einen Fehler drin?
Am Handy kommt immer nur prüfe das Passwort am Laptop keine Verbindung möglich.
Passwort = "openDTU42" ?

FSys77, 2022-07-23:
1) Kannst Du den genauen Link zu deinem Modul schicken? Oder ein Foto? Es schwirren Millionen verschiedener ESP32 Varianten herum, viele mit unterschiedlichen Pin Layouts.

2) Ich hatte ewig keine Verbindung, hab alles doppelt und dreifach geprüft - nur um dann herauszufinden, dass mein NRF24L01+ nicht richtig funktionierte (ich hatte mir damals 5 Stück gekauft; keines davon funktionierte). Hab dann eines bestellt welches im mikrocontroller.net Forum empfohlen wurde und mit dem klappte es dann auf Anhieb.

G1nac, 2022-07-23:
Das Problem muss ja schon beim ESP32 liegen https://www.ebay.de/itm/255283221996 , da ich für die ersten Einstellungen schon keine Verbindung aufbauen kann. Als NRF habe ich diesen https://www.ebay.de/itm/255283312809

FSys77, 2022-07-23:
> Das Problem müsste ja schon beim ESP32 liegen https://www.ebay.de/itm/255283221996 , da ich für die ersten Einstellungen schon keine Verbindung aufbauen kann. Als NRF habe ich diesen https://www.ebay.de/itm/255283312809

Was meinst Du mit "schon keine Verbindung"? Macht der bei Dir keinen Access Point auf? So wie hier beschrieben?
https://github.com/tbnobody/OpenDTU#first-configuration

> Kannst du mir evtl. einen Link zu einem funktionierenden ESP32 geben?

Bei mir hat sich über die Jahre eine Sammlung an ESP32s gebildet :-D ... ich hab einfach einen genommen der exakt 1:1 das selbe GPIO-Pinout hat wie der hier von tbnobody verwendete.

Eine schnelle ebay Suche hat den hier geliefert: https://www.ebay.de/itm/143958031194

G1nac, 2022-07-24:
> Was meinst Du mit "schon keine Verbindung"? Macht der bei Dir keinen Access Point auf? So wie hier beschrieben? https://github.com/tbnobody/OpenDTU#first-configuration

Jo stand schon im ersten Post keine Verbindung mit dem openDTU Access Point, Handy oder Laptop versuchen es kurz dann kommt nur fehlgeschlagen oder überprüfe Passwort.

Na dann versuche ich es mal mit der 38 Pin Variante aus dem Link und melde mich falls es dann funktioniert.

G1nac, 2022-07-27:
Dann sag ich erstmal danke. Super Projekt läuft mit der neuen ESP32 von AZ.
Habe es jetzt wieder wie in der Zeichnung angeschlossen und die Pins so festgelegt
-DHOYMILES_PIN_MISO=19
-DHOYMILES_PIN_MOSI=23
-DHOYMILES_PIN_SCLK=18
-DHOYMILES_PIN_IRQ=16
-DHOYMILES_PIN_CE=4
-DHOYMILES_PIN_CS=5
Schade das die Belegung der Zeichnung auf der Hauptseite nicht mit der Pin-belegung direkt darunter übereinstimmt das ist leicht verwirrend.

Nochmal danke.

closed 2022-07-27 by G1nac