tbnobody/OpenDTU #39: Feature Request: REST Interface

FSys77, 2022-07-23:
Es wäre toll wenn Ihr ein REST Interface implementieren könntet um die Daten im JSON Format zu bekommen. Und - später mal - die maximale Ausgangsleistung von extern setzen zu können.

tbnobody, 2022-07-24:
Meinst du sowas? https://github.com/tbnobody/OpenDTU/issues/8#issuecomment-1190663431

FSys77, 2022-07-24:
> Meinst du sowas? [#8 (comment)](https://github.com/tbnobody/OpenDTU/issues/8#issuecomment-1190663431)

Ja genau :-D

Superklasse!

tbnobody, 2022-07-24:
Bei der Rest API findet keinerlei Javascript Check oder ähnliches statt. Mit einem einfachen Get Request auf die genannte URL kommt nur JSON zurück.

FSys77, 2022-07-24:
> Bei der Rest API findet keinerlei Javascript Check oder ähnliches statt. Mit einem einfachen Get Request auf die genannte URL kommt nur JSON zurück.

*facepalm* ... ich hab Code kopiert und hatte da nen POST statt nen GET drin stehen. Habs erst jetzt gemerkt. Jetzt gehts.

Ich hätte noch eine Anmerkung: könnte man das API noch etwas vereinheitlichen und verschlanken? Um den Power-Wert abzufragen muss man z.B. das hier machen:
`return json_response[0]["0"]["Power"]["v"]`

Das erste [0] bräuchte es gar nicht, weil es außer [0] nix anderes (z.B. [1]) gibt. Es ist also eine Ebene zu viel drin

tbnobody, 2022-07-24:
Die erste Ebene ist der inverter. Für Anwendungen mit mehr als einem inverter wird hier hochgezählt. Diese Ebene bei nur einem inverter wegzulassen macht alles komplizierter weil es dann ein Sonderfall ist den man speziell abfragen muss.

FSys77, 2022-07-24:
> Die erste Ebene ist der inverter. Für Anwendungen mit mehr als einem inverter wird hier hochgezählt. Diese Ebene bei nur einem inverter wegzulassen macht alles komplizierter weil es dann ein Sonderfall ist den man speziell abfragen muss.

Ah OK, verstehe. Ja das macht Sinn.

Entschuldigung für meine blöden Fragen.

closed 2022-07-25 by tbnobody

stefan123t, 2022-07-25:
@FSys77 das issue ist m.E. gelöst und kann geschloßen werden oder willst Du es noch ansprechend dokumentieren, damit andere auch leicht den Einstieg in den REST API Endpoint: `/api/livedata/status` finden ?
> Der liefert ein json Objekt mit allen Daten die auch im Live View angezeigt werden.

stefan123t, 2022-07-25:
@tbnobody beat me to it