tbnobody/OpenDTU #42: TLS für MQTT

helgeerbe, 2022-07-27:
Hallo @tbnobody,
erst mal vielen Dank für dieses tolle Projekt. Aktuell warte ich noch auf meinen Sender. Da mein MQTT Broker TLS verwendet, wollte ich den Code so anpassen, dass man über die Config "Enable TLS" verwenden kann.

Leider musste ich feststellen, dass https://github.com/marvinroger/async-mqtt-client.git kein TLS für den esp32 unterstützt. Die nötigen Portierungen vom esp8266 sind wohl nie gemacht worden.

Gibt es irgendwelche Gründe warum du "async-mqtt-client" genutzt hast? Ansonsten würde ich die Lib z. B. mit "pubsubclient" austauschen.

Idee:
- ssl optional
- Root CA Certificate in die Config. Default "ISRG Root X1" für Letsencrypt Zertifikate

Was ich aktuell herausgefunden habe.
- Jason Docsize muss vergrössert werden
- Die Partition für Spiffs muss vergrößert werden

Ich habe keine Erfahrung mit vue. Aber ich habe herausgefunden, wie ich die WebConfig erweitern kann. Für das Zertifikat nutze ich eine ``textarea``. Das Lesen und Schreiben der Config funktioniert. Aber das Zertifikat ist so um die 2048 Zeichen lang. Es wird  wird nicht in die ``texarea``übernommen. Wenn ich weniger Zeichen nehme funktioniert alles. Gibt es da irgendwelche Einstellungen?

tbnobody, 2022-07-27:
> Gibt es irgendwelche Gründe warum du "async-mqtt-client" genutzt hast? Ansonsten würde ich die Lib z. B. mit "pubsubclient" austauschen.

Performance Gründe. Und ich finde auch allgemein das Handling mit den Async Bibliotheken einfacher. Auch mit dem Reconnection und Connection Status Handling habe ich bisher mit der async-mqtt-client Bibliothek bessere Erfahrungen gemacht.

helgeerbe, 2022-07-27:
Hmmm, dann müsste ich mir mal vertieft "AsyncTCP_SSL" ansehen....

helgeerbe, 2022-07-28:
So, habe mal ein wenig geforscht. Die von dir verwendete async-mqtt-client Bibliothek (https://github.com/marvinroger/async-mqtt-client.git) hat seit einem Jahr kein update mehr bekommen. Ich habe jetzt https://github.com/bertmelis/espMqttClient genommen. Damit geht auch SSL.

Habe jetzt noch ein letztes Problem mit vue. 

Habe enable TLS hinzugefügt:

[image]

Man kann jetzt auch das Root CA Certificate hinzufügen:
[image]

Allerdings wird die "Textarea" nicht mit dem Zertifikat initialisiert, wenn der Text zu lang ist:

[image]

Wenn ich das Zertifikat in `defaults.h` "kürze" geht alles.

````c
// ISRG_Root_X1.crt -- Root CA for Letsencrypt
#define MQTT_ROOT_CA_CERT "-----BEGIN CERTIFICATE-----\n" \
                            "MIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCiwAwDQYJKoZIhvcNAQELBQAw\n" \
                            "TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh\n" \
                            "cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTUwNjA0MTEwNDM4\n" \
                            "WhcNMzUwNjA0MTEwNDM4WjBPMQswCQYDVQQGEwJVUzEpMCcGA1UEChMgSW50ZXJu\n" \
                            "ZXQgU2VjdXJpdHkgUmVzZWFyY2ggR3JvdXAxFTATBgNVBAMTDElTUkcgUm9vdCBY\n" \
                            "MTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBAK3oJHP0FDfzm54rVygc\n" \
                            "h77ct984kIxuPOZXoHj3dcKi/vVqbvYATyjb3miGbESTtrFj/RQSa78f0uoxmyF+\n" \
                            "0TM8ukj13Xnfs7j/EvEhmkvBioZxaUpmZmyPfjxwv60pIgbz5MDmgK7iS4+3mX6U\n" \
                            "A5/TR5d8mUgjU+g4rk8Kb4Mu0UlXjIB0ttov0DiNewNwIRt18jA8+o+u3dpjq+sW\n" \
                            "T8KOEUt+zwvo/7V3LvSye0rgTBIlDHCNAymg4VMk7BPZ7hm/ELNKjD+Jo2FR3qyH\n" \
                            "B5T0Y3HsLuJvW5iB4YlcNHlsdu87kGJ55tukmi8mxdAQ4Q7e2RCOFvu396j3x+UC\n" \
                            "B5iPNgiV5+I3lg02dZ77DnKxHZu8A/lJBdiB3QW0KtZB6awBdpUKD9jf1b0SHzUv\n" \
                            "KBds0pjBqAlkd25HN7rOrFleaJ1/ctaJxQZBKT5ZPt0m9STJEadao0xAH0ahmbWn\n" \
                            "OlFuhjuefXKnEgV4We0+UXgVCwOPjdAvBbI+e0ocS3MFEvzG6uBQE3xDk3SzynTn\n" \
                            "jh8BCNAw1FtxNrQHusEwMFxIt4I7mKZ9YIqioymCzLq9gwQbooMDQaHWBfEbwrbw\n" \
                            "qHyGO0aoSCqI3Haadr8faqU9GY/rOPNk3sgrDQoo//fb4hVC1CLQJ13hef4Y53CI\n" \
                            "rU7m2Ys6xt0nUW7/vGT1M0NPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNV\n" \
                            "HRMBAf8EBTADAQH/MB0GA1UdDgQWBBR5tFnme7bl5AFzgAiIyBpY9umbbjANBgkq\n" \
                            "hkiG9w0BAQsFAAOCAgEAVR9YqbyyqFDQDLHYGmkgJykIrGF1XIpu+ILlaS/V9lZL\n" \
                            "ubhzEFnTIZd+50xx+7LSYK05qAvqFyFWhfFQDlnrzuBZ6brJFe+GnY+EgPbk6ZGQ\n" \
                            "3BebYhtF8GaV0nxvwuo77x/Py9auJ/GpsMiu/X1+mvoiBOv/2X/qkSsisRcOj/KK\n" \
                            "NFtY2PwByVS5uCbMiogziUwthDyC3+6WVwW6LLv3xLfHTjuCvjHIInNzktHCgKQ5\n" \
                            "ORAzI4JMPJ+GslWYHb4phowim57iaztXOoJwTdwJx4nLCgdNbOhdjsnvzqvHu7Ur\n" \
                            "TkXWStAmzOVyyghqpZXjFaH3pO3JLF+l+/+sKAIuvtd7u+Nxe5AW0wdeRlN8NwdC\n" \
                            "jNPElpzVmbUq4JUagEiuTDkHzsxHpFKVK7q4+63SM1N95R1NbdWhscdCb+ZAJzVc\n" \
                            "oyi3B43njTOQ5yOf+1CceWxG1bQVs5ZufpsMljq4Ui0/1lvh+wjChP4kqKOJ2qxq\n" \
                            "4RgqsahDYVvTH9w7jXbyLeiNdd8XM2w9U/t7y0Ff/9yi0GE44Za4rF2LN9d11TPA\n" \
                            "mRGunUHBcnWEvgJBQl9nJEiU0Zsnvgc/ubhPgXRR4Xq37Z0j4r7g1SgEEzwxA57d\n" \
                            "emyPxgcYxn/eR44/KJ4EBs+lVDR3veyJm+kXQ99b21/+jh5Xos1AnX5iItreGCc=\n" \
                            "-----END CERTIFICATE-----\n"
````

Cut & Paste mit Save funktioniert problemlos.
Ich bereite mal ein pull Request vor. Vielleicht fällt dir was auf. Bin für jeden Tip dankbar.

tbnobody, 2022-07-28:
Hallo @helgeerbe,
danke für den PR. Ich werde versuchen mal am Wochenende drüber zu schauen. Was mir noch nicht ganz klar ist, warum die Partitionsgröße geändert werden muss. aktuell dürften da 280kb frei sein. und wenn ich es richtig überflogen habe darf das zertifikat max 2kb groß sein.

tbnobody, 2022-07-28:
bzgl. dem nicht-anzeigen des zertifikates... ich würde vermuten, dass ein get request auf /api/mqtt/config auch schon ein leeres element anzeigt.

In der onMqttAdminGet Methode wird auch ein AsyncJsonResponse erzeugt. Dies hat u.a. einen optionalen Parameter maxJsonoBufferSize der default auf 1024 eingestellt ist. Mein erster versuch wäre, diesen puffer zu vergrößern.

helgeerbe, 2022-07-29:
Hallo @tbnobody, 
du hast recht. Die Partition muss anscheinend nicht vergrößert werden. Habe das auch schon im PR korrigiert. Ich hatte Speicherfehler, die auf fehlenden Speicher hindeuteten und der esp bootete permanent. Aber da hatte ich auch noch mit anderen Bibliotheken experimentiert, möglich dass es auch memory leaks gab.

Den JsonBuffer habe ich schon auf 3072 hochgesetzt.

tbnobody, 2022-07-29:
> Den JsonBuffer habe ich schon auf 3072 hochgesetzt.

Achso, das wusste ich nicht das du das schon getestet hast. Weil im PR ist es ja nicht drin.

helgeerbe, 2022-07-29:
Reden wir über verschiedene Buffer?

Ich habe das hier in WebApi_mqtt.cpp geändert

````cpp
DynamicJsonDocument root(3072); //TODO check size was 1024 + 2048 for cert
     DeserializationError error = deserializeJson(root, json);
````

helgeerbe, 2022-07-29:
OK,  gefunden. Das habe ich übersehen, da der default nicht explizit gesetzt wurde. Werde ich gleich mal testen.

tbnobody, 2022-07-29:
Ich meinte diese Stelle hier: https://github.com/tbnobody/OpenDTU/blob/12df6029cadbfcaedd315fecd2ef678803dddbc3/src/WebApi_mqtt.cpp#L54

helgeerbe, 2022-07-29:
Ja, das war der entscheidende Hinweis. Ich habe jetzt überall 3072 eingetragen.
Mein Sender ist auch da und ich kann endlich mal richtig testen.

helgeerbe, 2022-08-01:
MQTT mit TLS funktioniert jetzt soweit. Im Status, wird jetzt auch die CertInfo angezeigt, wenn TLS=true ist.

tbnobody, 2022-08-06:
Danke für den Hinweis mit der https://github.com/bertmelis/espMqttClient Bibliothek. Ich habe deine commits noch zusammengefasst und die ein oder andere kleine Änderung vorgenommen.

closed 2022-08-06 by tbnobody