tbnobody/OpenDTU #48: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled

Mearten, 2022-08-05:
Hello,
I have an esp32 after I upload the OpenDTU to the ESP I get the following error: _Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled_
Then the esp goes into a reboot.
Entire log:
```
17:49:17.718 > Starting OpenDTU
17:49:17.718 > Initialize FS... done
17:49:17.718 > Reading configuration... done
17:49:17.718 > Initialize Network... done
17:49:17.718 > Setting Hostname... Configuring WiFi STA using new credentials... done
17:49:17.718 > Initialize NTP... done
17:49:17.718 > Initialize MqTT... done
17:49:17.718 > Initialize WebApi... done
17:49:17.718 > Initialize Hoymiles interface... Connection error!!
17:49:17.718 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
17:49:17.718 > 
17:49:17.718 > Core  1 register dump:
17:49:17.718 > PC      : 0x401720f7  PS      : 0x00060430  A0      : 0x800e1296  A1      : 0x3ffb27b0  
17:49:17.718 > A2      : 0x00000000  A3      : 0x3ffc3aa0  A4      : 0x00000000  A5      : 0x00000000  
17:49:17.718 > A6      : 0x17481253  A7      : 0x00000116  A8      : 0x800ee54a  A9      : 0x3ffb2750  
17:49:17.718 > A10     : 0x00000000  A11     : 0xffffeff5  A12     : 0x00000000  A13     : 0x00000008  
17:49:17.718 > A14     : 0x3ffd0dfc  A15     : 0x3ffd0dfc  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
17:49:17.718 > EXCVADDR: 0x000001d0  LBEG    : 0x4008b3cd  LEND    : 0x4008b3de  LCOUNT  : 0xffffffff  
17:49:17.718 > 
17:49:17.718 > 
17:49:17.718 > Backtrace:0x401720f4:0x3ffb27b00x400e1293:0x3ffb27d0 0x400f3eca:0x3ffb2820 
17:49:17.718 >
```
I can't find where the LoadProhibited variable appears in the code.

At the moment I have not connected the RF24L01+, it doesn't matter whether it is connected or not the same error remains.

Can someone help me with this problem?

tbnobody, 2022-08-06:
Hello,
the "Connection error" means that the RF module is either not connected or the connection is wrong. 

You could change the line (in platformio.ini) from
`monitor_filters = time, colorize, log2file, esp32_exception_decoder`
to
`monitor_filters = log2file, esp32_exception_decoder`

and compile, upload and run the code again. Then the esp32_exception_decoder should decode the core dump and the backtrace and you should see a more detailed exception description.

Mearten, 2022-08-06:
I changed the line, but the result in the serial monitor is still the same:
```
Starting OpenDTU

Initialize FS... done
Reading configuration... done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Hoymiles interface... Connection error!!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4017209b  PS      : 0x00060230  A0      : 0x800e128a  A1      : 0x3ffb27b0  
A2      : 0x00000000  A3      : 0x3ffc3aa0  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x17481253  A7      : 0x00000116  A8      : 0x800ee53e  A9      : 0x3ffb2750  
A10     : 0x00000000  A11     : 0xffffeff5  A12     : 0x00000000  A13     : 0x00000008  
A14     : 0x3ffd0e18  A15     : 0x3ffd0e18  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000001d0  LBEG    : 0x4008b3cd  LEND    : 0x4008b3de  LCOUNT  : 0xffffffff  

Backtrace:0x40172098:0x3ffb27b00x400e1287:0x3ffb27d0 0x400f3e96:0x3ffb2820 
```
I have not connected the RF24L01+, I'm nowhere near the hoymiles either.
But my thought is that I want to be able to get into the configuration without having the RF24L01+ connected without being near my hoymiles.

A noob question, when I upload new code everything is gone, it can't be that something is left behind from a previous project that gets in the way.

ahinrichs, 2022-08-07:
You can leave out the Hoymiles part by commenting out the `Hoymiles.loop()` in `src.main.cpp` line 105 and 106. I used this on a second ESP for testing.

Mearten, 2022-08-07:
Thanks it works
Before the void loop he already went into the reboot, commenting lines 105 and 106 was not enough.
After turning off lines 80 to 98, it will continue to run.
In the configuration of the inverter there was a series of characters (no serial number of the inverter) I have removed this. Lines 80 through 98 turned on. After this, the aspen starts up.

Got an error message buffer full, but after disconnecting the RF24L01+ this message was gone, so here I have the next challenge how to connect it correctly.

Mearten, 2022-08-07:
The buffer full output comes after some time.
There may be more errors in the config.bin
Is it also possible to remove it completely? back to factory settings

tbnobody, 2022-08-07:
Buffer full is a good hint that your wiring is wrong.

Mearten, 2022-08-07:
The buffer full message also comes without the RF24L01+ being connected, I see now

Mearten, 2022-08-07:
This has nothing to do with my initial problem.

Can I also see somewhere when the RF24L01+ is properly connected? or only after I have added an Inverter and start reading?

tbnobody, 2022-08-07:
> Can I also see somewhere when the RF24L01+ is properly connected?

Yes. You will get the following message:
`Initialize Hoymiles interface... done`

Mearten, 2022-08-07:
The message is now: _Initialize Hoymiles interface... Connection successfull_
Tomorrow I'm going to the Hoymiles again so I can try if they are approachable.

For this I had to adjust the platformio.ini, lines 48 to 54 also placed under `[env:generic]`
the lines:
```
build_flags = ${env.build_flags}
     -DHOYMILES_PIN_MISO=15
     -DHOYMILES_PIN_MOSI=2
     -DHOYMILES_PIN_SCLK=14
     -DHOYMILES_PIN_IRQ=13
     -DHOYMILES_PIN_CE=16
     -DHOYMILES_PIN_CS=5
```

Mearten, 2022-08-08:
After adding the serial number of the Hoymiles, it keeps rebooting.
After I disable lines 41 through 53 in the `src.main.cpp` file, it no longer reboots. Error in the config.bin?
If I delete the config.bin (how I don't know yet). Will a new one be created automatically? So I can start again at 0.

tbnobody, 2022-08-08:
I would try to erase the flash. This removes all partitions. (see https://community.platformio.org/t/how-to-erase-flash-of-esp32/27955)

After that you have to flash the esp again. but currently it seams that your littlefs partition contains junk.

Mearten, 2022-08-09:
After a full reset, I was able to add and read both my Hoymiles without any problems.

tbnobody, 2022-08-09:
Nice that it works now. I would say the issue was because of old data on in the flash memory. But not sure. Due to the fact that it's working now I am going to close this issue. Please feel free to open a new one if the error occours again.

closed 2022-08-09 by tbnobody