tbnobody/OpenDTU #64: Hostname not applied correctly

horfic, 2022-08-06:
Hi,
when setting under Settings -> Network Settings the hostname for the device to e.g. "OpenDTU" and saving it. The hostname reports after reboot and also reconnection esp32-F4DF20 instead of the saved and shown hostname in the webui.

[image]

As shown in Ubiquiti Unifi Network GUI.

Does anybody else experience this issue?

cheers,
Horst

tbnobody, 2022-08-07:
Hi,
habe hier das gleiche Problem. Habe mehrere Workarounds usw. versucht aber bekomme keinen Hostnamen gesetzt.

horfic, 2022-08-07:
Servus,
ich hab hier ein interessantes ticket gefunden.

https://github.com/espressif/arduino-esp32/issues/6278

Hier wird beschrieben das bevor WiFi.mode(WIFI_STA) gesetzt wird der hostname gesetzt werden muss.

Kann sein das passiert weil du den Mode zuerst per setupMode in der init methode setzt?

madmartin, 2022-08-23:
Hi,
nachdem ich #102 zusammenprogrammiert habe stelle ich auch dieses Phänomen fest.
Das Setzen des Hostname funktioniert beim allerersten Start, wenn der Accesspoint Modus genutzt wird. Gebe ich dann die WiFi-Daten ein erhält auch der DHCP Server via WiFi den korrekten Hostnamen.
Nach dem nächste Reset des ESP kommt am DHCP Server immer der Hostname esp32-XXXXX .

Ich kümmer mich darum.

madmartin, 2022-08-28:
So, das Ergebnis meiner Untersuchungen:
zusammen mit meinem Patch #102 funktioniert das Setzen des Hostnamen (OpenDTU-xxxxx wobei xxxx durch die letzten 3 Byte der Macadresse in hex ersetzt wird).

Allerdings scheint es wichtig zu sein daß man vor dem flashen einmal "erase flash" durchführt! Danach erscheint der Hostname "esp32-XXXXXX" nicht mehr im Log des DHCP Servers, sondern nur noch "OpenDTU-xxxxxx". 
(Vorher evtl. Inverter-Seriennummern sichern!)

Warum? Kann ich nicht wirklich erklären.
Ich kann aber versichern daß das setzen des Hostname vor dem Aufruf von "WiFi.mode" nicht ausreichend ist.

tbnobody, 2022-09-05:
I think I fixed it... Will commit it tomorrow evening. Testing it over night and day. The hostname will be set reliable on every update. Problem is here:
https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L1122

The hostname is only applied when calling the WiFi.mode with parameter WIFI_MODE_STA

This means after setting the hostname you have to call WiFi.mode(WIFI_MODE_STA)..

madmartin, 2022-09-06:
> This means after setting the hostname you have to call WiFi.mode(WIFI_MODE_STA)..

Yes, and this also means that setting WIFI_AP_STA for the initial admin mode has a negative side-effect. When Wifi is configured and connected, the hostname send to the DHCP server looks wrong. Minutes later, when the AP is turned off by calling  WiFi.mode(WIFI_MODE_STA), the DHCP server receives the correct hostname. If the router and its DNS picks up the new hostname (and when) depends from the router/DNS and the leasetime.

Maybe it is not ideal to open an AP on every poweron of the OpenDTU even if WiFi is configured and able to connect. This exposes a security problem because the AP has a fixed password, which can be looked up in the internet.

In my esp8266 project, I implemented the AdminMode with a DoubleReset detection library.

tbnobody, 2022-09-06:
Should be fixed with f5328100b6f7c1e098a126f5e244ef7fbaae7d0b

closed 2022-09-08 by tbnobody

stefan123t, 2022-09-19:
According to some randomnerd‘s we should set the hostname before calling WiFi.begin():

> To set a custom hostname for your board, call `WiFi.setHostname(YOUR_NEW_HOSTNAME);` before `WiFi.begin();`

see https://randomnerdtutorials.com/esp32-set-custom-hostname-arduino/

tbnobody, 2022-09-20:
Hello @stefan123t, the article you linked seems to be a little bit older. Even the comments below the article say that this method is not working. 
[image]

If you look at the implementation of the `setHostname()` method (https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L774) you can see that this method just calls set_esp_netif_hostname (https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L289) which just sets the internal variable `default_hostname` to the defined hostname.

This variable will be only read by the function called `get_esp_netif_hostname` (https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L281)

The `get_esp_netif_hostname` method gets only called in the `mode` method (https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L1106)

The code (https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L1122) can only be reached if the new mode is **unequal** the current mode (https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L1109) and if the new mode equals WIFI_MODE_STA (https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L1121)

**I hope this helps to understand that with the current implementation of the WiFi class the following order has to be used:**
* First call `setHostname` to the set internal variable `default_hostname`
* Then call mode to set the mode to something else except `WIFI_MODE_STA` 
* Then call mode with parameter `WIFI_MODE_STA` to set the hostname using the `set_esp_interface_hostname` method

stefan123t, 2022-09-20:
Does this work differently in the ESP8266 case ?
While this sounds like a plausible hack just to set the hostname.
Do you think we should address this/our use case upstream ?

tbnobody, 2022-09-20:
To be honest, I did my last esp8266 project is some years ago. At this time it was similiar "unituitive". But I am not sure whether the same logic had to be applied.

stefan123t, 2022-09-20:
@tbnobody obviously this seems to be a requirement of the ESP32 re-implementation of the WiFi stack.
Apparently DHCP does not allow to change the Hostname while we are connected in STA mode.
https://github.com/espressif/arduino-esp32/issues/6099#issuecomment-1166313963

Also note that there has already been quite some discussion upstream about the correct order to which we have to adhere to since 2.0.3:
https://github.com/espressif/arduino-esp32/issues/6700