tbnobody/OpenDTU #66: Feature Request - add second WiFi configuration

wib100, 2022-08-06:
Hi,
would be great if there would be an option to add a second WiFi/Hotspot to the network configuration.
This would be very helpful to pre-configure an Arduino.

Thanks,
Willi

Dom1n1c, 2023-07-10:
I would appreciate it. Some more or less known projects and ESP based devices such as: ESPeasy, Tasmota (depending on the release), Shelly also have the possibility of adding a secondary or backup WiFi.

ortegafernando, 2025-07-14:
> I would appreciate it. Some more or less known projects and ESP based devices such as: ESPeasy, Tasmota (depending on the release), Shelly also have the possibility of adding a secondary or backup WiFi.

I agree you, Tasmota for example has also a secondary or backup Wifi. I have also found it usefull when you want to change your main wifi password, you will never lost connection with opendtu if it has two wifi "configs".