tbnobody/OpenDTU #67: Feature Request - option to reboot device

wib100, 2022-08-06:
Hi,
would be great to have an option in the configuration menu to reboot the device.
Reason: after changing options, would be useful to reboot the device to make sure new options are in place.

Thanks,
Willi

HacksBugsAndRockAndRoll, 2022-08-16:
just did a  quick implementation of a program restart button (also reloads config from my understanding), looks like if you want to hard reset the microcontroller you need to run an extra  wire to an IO pin

sivar2311, 2022-08-16:
> ...  looks like if you want to hard reset the microcontroller you need to run an extra wire to an IO pin

A call to `ESP.restart()` restarts the ESP32, no extra wire is needed.

tbnobody, 2022-08-16:
Just as a thought: Are you closing and reopening a Word document after you've saved it? Just do doublecheck that it's really saved? Would you like to have a feature in Word that closes/reopens the document automatically after each save? Or would you rather a version without such kind of saving issue?

If it's not saved and applied it's a bug which has to be fixed. 
Did it occour that some settings where not saved?

HacksBugsAndRockAndRoll, 2022-08-16:
For me the need for a restart came up, when my inverter config got corrupted during runtime. 
I was assuming, that it  had to do with corrupted RAM, since things were back to normal once I triggered a reloading of the config. As a workaround I had a  restart trigger in place when no data was published by openDTU for a while. For this I had the backend part, I added the UI part since it looked like a good fit for this request.

tbnobody, 2022-08-16:
This problem was propably fixed here: https://github.com/tbnobody/OpenDTU/issues/29
If this issue still occours please open a new issue. If you just reboot your device for your own no one else will know that other people have such kind of issue as well and it will not get fixed.

HacksBugsAndRockAndRoll, 2022-08-16:
> > ...  looks like if you want to hard reset the microcontroller you need to run an extra wire to an IO pin
> 
> A call to `ESP.restart()` restarts the ESP32, no extra wire is needed.

thanks, did not know about that yet!  using this now.

HacksBugsAndRockAndRoll, 2022-08-16:
> This problem was propably fixed here: #29 If this issue still occours please open a new issue. If you just reboot your device for your own no one else will know that other people have such kind of issue as well and it will not get fixed.

I was running yesterdays' master so this fix was in. I was also tunning another openDTU not showing these problems, so currently my assumption is a broken ESP.

madmartin, 2022-08-19:
> But in this discussion it should not matter why people want this functionality but if it should go into the project or not. I'd say it is probably useful and does not influence other functionality.

I agree.

Riza-Aslan, 2022-10-27:
I agree too. I've changed the IP of my ESP32 and was trying to restart the ESP to obtain a new IP from my DHCP-Server. I had to restart the ESP manually now. A restart option is an essential feature for such a device and shouldn't be a problem to implement it.

closed 2022-11-25 by tbnobody

madmartin, 2022-11-25:
#81 can now be closed, too.