tbnobody/OpenDTU #68: Open DTU zeigt keine Daten

J3nuevo, 2022-08-07:
Hallo Zusammen,

Ich habe alles soweit hinbekommen dass ich mir zu OpenDTU melden kann, serial Nr von Hoymiles 600 ist eingegeben und WLan Settings auch...allerdings schein was nicht zu funktionieren....

Bei mir zeigt gar keine Werte bleibt alles auf 0.

Wo kann das Problem liegen?

[image]

hubsi5, 2022-08-07:
Vielleicht einmal den Serial Log auslesen, ob der nRF24L01 überhaupt erkannt wird

J3nuevo, 2022-08-07:
Theoretisch und wenn ich das richtig verstehe der NRF24L01+ ist doch in ordnung weil die OpenDTU erkennt mein WLAN oder?
[image]
....
Immer noch keine Daten...alles auf 0

J3nuevo, 2022-08-07:
Kann sein dass das Problem? Unter Mqtt Info...
[image]

hubsi5, 2022-08-07:
Hat mit WLAN genau nichts zu tun.
Tu was ich dir gesagt habe

Petermann7, 2022-08-07:
> Hallo Zusammen,
> 
> Ich habe alles soweit hinbekommen dass ich mir zu OpenDTU melden kann, serial Nr von Hoymiles 600 ist eingegeben und WLan Settings auch...allerdings schein was nicht zu funktionieren....
> 
> Bei mir zeigt gar keine Werte bleibt alles auf 0.
> 
> Wo kann das Problem liegen?
> 
> [image]

Ich habe das gleiche Problem, openDTU bekommt keine Daten vom Inverter. Wie kann ich den Serial Log auslesen? 
Mit Putty SSH, Telnet und COM-Port geht es wohl nicht.

J3nuevo, 2022-08-07:
> Hat mit WLAN genau nichts zu tun. Tu was ich dir gesagt habe

So ich habbe ein Serial Monitor gemacht und bekommen folgendes....

[image]

kpwg, 2022-08-08:
Zeile 17 sagt, das etwas mit der Verbindung vom ESP32 zum NRF24L01+ nicht stimmt.

J3nuevo, 2022-08-08:
> Zeile 17 sagt, das etwas mit der Verbindung vom ESP32 zum NRF24L01+ nicht stimmt.

Ja, das kann ich selber lesen hahaha. Aber warum oder wo kommt das fehler her?

tbnobody, 2022-08-08:
Bitte Verkabelung prüfen... (https://github.com/tbnobody/OpenDTU/raw/master/docs/Wiring_ESP32_Schematic.png)

Mearten, 2022-08-08:
Ich habe _Hoymiles interface... Connection successfull_. Er konnte den NRF24L01+ bei mir finden, aber ich habe immer noch folgende Meldung:
```
17:51:49.575 > Nothing received, resend whole request
17:51:49.580 > TX Channel: 61 --> 15 74 81 23 53 78 56 34 12 80 B 0 62 F1 31 14 0 0 0 0 0 0 0 0 48 71 9C
17:51:49.884 > RX Period End
```

Also habe ich den RX wahrscheinlich nicht richtig angeschlossen.

@J3nuevo:
Was für ein ESP-Board hast du? vielleicht können wir uns gegenseitig helfen. Ich habe esp32-wroom 32 mit NRF24L01+

tbnobody, 2022-08-08:
> Also habe ich den RX wahrscheinlich nicht richtig angeschlossen.

Glaube ich nicht, weil bei SPI gibt es zwar auch sowas wie RX/TX aber schon beim setzen der Einstellungen des NRF wird darüber kommuniziert. Hätte hier eher den IRQ Pin in verdacht.

J3nuevo, 2022-08-08:
> Ich habe _Hoymiles interface... Connection successfull_. Er konnte den NRF24L01+ bei mir finden, aber ich habe immer noch folgende Meldung:
>
> ```
> 17:51:49.575 > Nothing received, resend whole request [...]

Ich habe genau das selber esp32-wroom 32 mit NFR24L01+
Bei mir habe die pins wie folgt.

DHoymiles spricht die NFR24L01+ pins

DHOYMILES_PIN_CE= D4
DHOYMILES_PIN_SCLK=D10 
DHOYMILES_PIN_MISO=D19
DHOYMILES_PIN_CS=D5
DHOYMILES_PIN_MOSI=D23
DHOYMILES_PIN_IRQ=D15

Bei der D10 bin mir unsicher weil die Esp8266 wird die D18 benutz haben wir aber im die esp32 gar nicht.

[image]

tbnobody, 2022-08-08:
Ich mag mich täuschen... Aber erstens ist D15 auf deinem Bild unbelegt... obwohl du oben schreibst das dort dein IRQ liegt. 

Das Problem ist, das ein ESP32-Wroom alles sein kann. Damit ist nur das aufgelötete kleine Board gemeint. Die Auf deinem Bild sehe ich z.B. gar kein PIN18 nach außen gelegt.

J3nuevo, 2022-08-08:
> Ich mag mich täuschen... Aber erstens ist D15 auf deinem Bild unbelegt... obwohl du oben schreibst das dort dein IRQ liegt.
> 
> Das Problem ist, das ein ESP32-Wroom alles sein kann. Damit ist nur das aufgelötete kleine Board gemeint. Die Auf deinem Bild sehe ich z.B. gar kein PIN18 nach außen gelegt. Oder er verbirgt sich hinter den TX2 / RX2 o.ä. aber dafür müsste man die genaue Board Bezeichnung kennen)

Sorry die Photo war nicht ganz Aktuell aber doch die D15 ist belegt und mit die IRQ verbundet.

Hier vielleicht noch ein paar photos...vielleicht somit kann mir jemand unterstutzen...

Was macht der pin18 die so wichtig ist? Eventuell kann ich mal mit die TX2 oder RX2 probieren?
[image]

[image]

J3nuevo, 2022-08-08:
Mir sheint die zu sein...nur statt D18 bei mir ist D10 geschrieben...scheint aber alle andere gleich zu sein...

esp32 devkit v1 mit 30 pins

kpwg, 2022-08-09:
Dringende Empfehlung: Die Pinheader mit dem Board verlöten, damit es zuverlässigen Kontakt gibt.

hubsi5, 2022-08-09:
Ich hatte einmal das Problem, dass beim mir der eine GND keinen Kontakt hatte.
Kannst auch einmal den anderen GND versuchen.
Und das mit dem löten, wurde eh schon gesagt

Petermann7, 2022-08-11:
Hallo,

ich habe es jetzt ans laufen bekommen.

Was habe ich genacht:

Prüfung ob der GND Anschluß auch wirklich GND ist, wahr ok

 

Neue Bestellung und Austausch des NRF24L01+ von AZ-Delivery, danach:

Initialize Hoymiles interface... Connection successfull,

aber noch immer keine Daten vom Inverter

 

Disable MqTT in openDTU: Daten vom Inverter kommen rein :-)

 

 

 

 
 

Gesendet: Montag, 08. August 2022 um 23:42 Uhr

dmr173, 2022-08-13:
Hallo zusammen,

ich bin ein totaler Newbie was ESP32 und solche Themen angeht und war daher positiv überrascht, wie schnell ich dank der guten Beschreibung die Software auf das Board gekriegt habe. Jetzt stehe ich auch an dem Problem, dass OpenDTU keine Daten von meinem HM-300 bekommt.
Ich nutze dieses ESP32 Board: [Firebeetle](https://www.bastelgarage.ch/firebeetle-board-esp32-e-mit-angeloteten-buchsenleisten?search=firebeetle%20esp32) mit diesem NRF24L01 Modul: [NRF24L01](https://www.bastelgarage.ch/nrf24l01-wireless-funk-modul-2-4ghz?search=nr)

Im Serial Monitor sehe ich ebenfalls, dass keine Messages empfangen werden:

16:57:48.797 > TX Channel: 255 --> 15 71 91 2 7 78 56 34 12 80 B 0 62 F7 BB EA 0 0 0 0 0 0 0 0 6C 0 DB 
16:57:48.797 > RX Period End
16:57:48.797 > All missing
16:57:48.797 > Nothing received, resend whole request

Daher vermute ich auch die Verkabelung.

Bei dem ESP-32 Board finde ich MOSI, MISO und SCK klar bezeichnet mit den erwarteten PINs aber z.B. GPIO16 bzw Pin 27 für IRQ finde ich gar nicht im Layout.

Kann mir jemand sagen wie ich hier die passenden Pins finden kann? Oder welche alternativen Pins ich nutzen könnte? Gibt's da ein Vorgehen für "Dummies"?

dmr173, 2022-08-13:
[image]

dhaude, 2022-08-13:
Hallo bin auch Newbe und total begeistert von diesem Projekt.
Du musst die 'fehlenden' GPIO-Pins durch freie auf deinem Board ersetzen und diese dann in der platformio.ini entsprechend angeben, damit der Compiler die Pin's richtig mappen kann.

Beispiel:
[env:generic]
board = esp32dev
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19  (ist OK bei dir)
    -DHOYMILES_PIN_MOSI=23  (ist OK bei dir)
    -DHOYMILES_PIN_SCLK=18  (ist OK bei dir)
    -DHOYMILES_PIN_IRQ=16 (musst du ändern, z.B. in 14)
    -DHOYMILES_PIN_CE=4  (ist OK bei dir)
    -DHOYMILES_PIN_CS=5 (musst du ändern, z.B. in 13)

dmr173, 2022-08-13:
Vielen Dank für die schnelle Antwort!!! Funktioniert jetzt:

[image]

Und jetzt gleich mal den 3D Drucker anwerfen fürs Gehäuse :-)

CBRocker, 2022-08-19:
Hallo Gemeinde,
ich habe ein ähnliches Problem. Hier meine Daten, vielleicht kann mir jemand einen Tipp geben:

Hardware:
ESP32-WROOM-32 ESP32_Devkitc_V4 (von AZ-Delivery), NRF24L01+

Firmware erfolgreich auf den ESP32 geschrieben.
Keine Daten von meinem Hoymiles HM-600

Serial Monitor ergibt folgendes:

23:20:59.012 > Starting OpenDTU
23:20:59.012 > Initialize FS... done
23:20:59.012 > Reading configuration... done
23:20:59.012 > Initialize Network... done
23:20:59.012 > Setting Hostname... Configuring WiFi STA using new credentials... done
23:20:59.012 > Initialize NTP... done
23:20:59.012 > Initialize MqTT... done
23:20:59.012 > Initialize WebApi... done
23:20:59.012 > Initialize Hoymiles interface... Connection successfull
23:20:59.012 >   Setting radio PA level...
23:20:59.012 >   Setting DTU serial...
23:20:59.012 >   Setting poll interval...
23:20:59.012 >   Adding inverter: [Seriennummer] - HM-600 done
23:20:59.012 > done
23:20:59.012 > Switch to WiFi mode
23:20:59.012 > Setting Hostname... done
23:20:59.012 > Configuring WiFi STA using existing credentials... E (909) wifi:sta is connecting, return error
23:20:59.012 > [   267][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
23:20:59.012 > done
23:20:59.012 > Configuring WiFi STA static IP... done
23:20:59.012 > WiFi connected
23:20:59.012 > WiFi got ip: 192.168.178.100
23:20:59.012 > Network connected
23:21:22.344 > Fetch inverter: [Seriennummer] 
23:21:22.348 > TX Channel: 23 --> 15 81 2 13 3 78 56 34 12 80 B 0 62 FF FE D1 0 0 0 0 0 0 0 0 F9 1F 51
23:21:22.586 > RX Period End
23:21:22.589 > All missing
23:21:22.589 > Nothing received, resend whole request
23:21:22.593 > TX Channel: 40 --> 15 81 2 13 3 78 56 34 12 80 B 0 62 FF FE D1 0 0 0 0 0 0 0 0 F9 1F 51
23:21:22.836 > RX Period End
23:21:22.838 > All missing
23:21:22.840 > Nothing received, resend whole request
23:21:22.844 > TX Channel: 61 --> 15 81 2 13 3 78 56 34 12 80 B 0 62 FF FE D1 0 0 0 0 0 0 0 0 F9 1F 51
23:21:23.086 > RX Period End
23:21:23.090 > All missing
23:21:23.090 > Nothing received, resend whole request
23:21:23.094 > TX Channel: 75 --> 15 81 2 13 3 78 56 34 12 80 B 0 62 FF FE D1 0 0 0 0 0 0 0 0 F9 1F 51
23:21:23.335 > RX Period End
23:21:23.338 > All missing
23:21:23.343 > Nothing received, resend whole request
23:21:23.345 > TX Channel: 3 --> 15 81 2 13 3 78 56 34 12 80 B 0 62 FF FE D1 0 0 0 0 0 0 0 0 F9 1F 51
23:21:23.585 > RX Period End
23:21:23.588 > All missing
23:21:23.591 > Nothing received, resend count exeeded

Hier ein Auszug der platformio.ini, der wohl relevant ist.
So habe ich auch die GPIOs angepasst (mit der originalen Belegung nach Anleitung kommen die gleichen Fehler)

[env:generic]
board = esp32dev
monitor_port = COM3
upload_port = COM3

[env:olimex_esp32_poe]
; https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

board = esp32-poe
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5
    -DOPENDTU_ETHERNET

monitor_port = COM3
upload_port = COM3

Ich habe jetzt schon einige Stunden selber herumexperimentiert, aber nun brauche ich doch externe Hilfe.
Ich freue mich auf Hilfestellung.

Vielen Dank,
Grüße Christoph

madmartin, 2022-08-19:
> Hardware:
> ESP32-WROOM-32 ESP32_Devkitc_V4 (von AZ-Delivery), NRF24L01+

Hi, ich habe auch die Erfahrung gemacht daß es nicht auf Anhieb funktionierte und kann folgende Tips geben:
- probiere den Abstand zwischen der openDTU und dem Wechselrichter zu verringern (z.B. ans Fenster richtung Dach bringen)
- unter Settings -> DTU Settings kann man die Sendeleistung "PA level" verstellen. Steht default auf minimum. Bei mir funktionierts mit "high" wenn ich mit dem Board am Schreibtisch sitze und der Wechselrichter in ca. 10m auf dem Dach liegt.
- der NRF24L01 braucht verhältnismäßig viel Strom. Bei schlechter Stromversorgung (und vorallem schlechten Kabeln!) bringt ein direkt am NRF24L01 angelöteter 10uF Kondensator mehr Stabilität (Pin 1+2 sind die Stromversorgung). Polarität des Kondensators beachten....

Ich drücke die Daumen!
Martin

CBRocker, 2022-08-19:
@madmartin 
Hey madmartin,
danke für die Tipps. Teste ich gleich mal.

Der ESP32 ist auf jeden Fall der richtige?

Was bedeutet aber dieser Fehler?

23:20:59.012 > Configuring WiFi STA using existing credentials... E (909) wifi:sta is connecting, return error
23:20:59.012 > [ 267][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007

Danke,
Grüße Christoph

sivar2311, 2022-08-20:
@CBRocker Das bedeutet, dass die Verbindung zu deinem WLAN nicht hergestellt werden konnte.

CBRocker, 2022-08-20:
@sivar2311 
Hey, also zu meinem Router-wlan? Wenn der ESP per USB am Notebook hängt wegen VSCode, komme ich tatsächlich nicht drauf.
Aber wenn der mit 5V Kabel am Strom verbunden ist, komme ich drauf. 

Ist der Fehler also zu vernachlässigen?

Entfernung zum Hoymiles 5 m direkte Sicht. 
Habe ein Netzteil mit 5,2 V. 

Wäre die Seriennummer des Hoymiles falsch, wäre das irgendwo erkennbar?

CBRocker, 2022-08-20:
Evtl. habe ich das falsche WLAN-Modul, kann das sein?
Ich habe den Kuman nRF24L01+PA+LNA Antenna Drahtloser Sender-Empfänger RF Sender-Empfänger Modul 3 pcs nRF24L01+PA+LNA Antenna Wireless Transceiver RF Transceiver Module for Arduino KY67  
von Amazon. Das Plus-Zeichen hat mich irritiert. Das ist wohl als "und" zu verstehen und nicht als Bezeichnung des RF-Moduls.

Bevor ich jetzt stundenlang weiter probier, hole ich jetzt erst mal einen echten NRF24L01+.
Ich gebe dann Feedback.

sivar2311, 2022-08-21:
Hi @CBRocker 
> Hey, also zu meinem Router-wlan? 

Ja, der Fehler bedeutet, das der ESP keine Verbindung zum WLAN bekommt.

> Wenn der ESP per USB am Notebook hängt wegen VSCode, komme ich tatsächlich nicht drauf.
> Aber wenn der mit 5V Kabel am Strom verbunden ist, komme ich drauf.

Ist der ESP dann ungefähr an der gleichen Position oder ist das 5V Netzteil ganz woanders?

> Ist der Fehler also zu vernachlässigen?

Während der Testphase würde ich dafür sorgen, dass der ESP an deinem Notebook auch eine Verbindung zum WLAN bekommt. So kannst du über dein Notebook dann das Serial-Log auslesen und besser Fehler erkennen.

> Entfernung zum Hoymiles 5 m direkte Sicht.

Ich habe hier ca. 7m durch eine Wand hindurch.

> Wäre die Seriennummer des Hoymiles falsch, wäre das irgendwo erkennbar?

Das kann ich nicht genau beantworten. Aber was sollte man bei einer Seriennumer falsch machen?
Zur Not einfach nochmal löschen und neu eingeben.

> Evtl. habe ich das falsche WLAN-Modul, kann das sein?

Das WLAN-Modul ist der ESP32 - Da kannst du nicht viel falsch machen. 
Ich glaube du meinst evtl. "das falsche NRF24-Modul" ?!

> Ich habe den Kuman nRF24L01+PA+LNA Antenna Drahtloser Sender-Empfänger RF Sender-Empfänger Modul 3 pcs  nRF24L01+PA+LNA Antenna Wireless Transceiver RF Transceiver Module for Arduino KY67
> von [...]

Auf Amazon habe ich dazu die Kundenfrage gefunden:
_Frage: Sind die nrf24l01+ oder nrf24l01?
Antwort: Die nRF24L01_

Es werden also vermutlich die falschen NRF24 Module sein.

CBRocker, 2022-08-21:
Hallo @sivar2311 

danke für die Infos und deine Zeit.

Die Info bei Amazon hatte ich übersehen, daß es nicht der + ist. Ich habe in der Artikelbeschreibung das + gesehen,
dann dachte ich, sei der richtige. Aber das bezieht sich ja auf das Bundle.
Aber der echte + ist schon unterwegs.
Ja, ich meinte natürlich den nRF24 als "WLAN-Modul", was dann Verbindung mit dem WR herstellt.

Wg. der Seriennummer hatte ich gefragt, weil ich es versäumt hatte, die SN vom WR abzufotografieren.
Die SN steht natürlich auch auf der Rechnung. Aber ich hatte dann schon die Befürchtung, die SN wäre evtl.
auf der Rechnung nicht korrekt, weil ich keine Verbindung bekommen hatte.
Und ich hätte keine Lust, nochmal aufs Dach zu steigen, um ein Modul wegzuschrauben, daß ich an den WR
komme um die SN zu fotografieren.

Aber ich denke, der Knackpunkt ist der falsche nRF24.

Vielen Dank.
Ich werde berichten.

Grüße

CBRocker, 2022-08-23:
Hallo,
Ich möchte nun berichten, dass mit dem richtigen nRF24L01+ Modul alles perfekt funktioniert. Lief auf Anhieb gleich. 
Vielen Dank an die Hilfegeber für die Tipps und vor allem an den/die Entwickler von OpenDTU. Tolles Projekt und spart viel Geld.

tbnobody, 2022-08-24:
Schön das es klappt! Würde das Issue hiermit schließen.

closed 2022-08-24 by tbnobody

CBRocker, 2022-10-11:
@kpwg 
Hey, damit ich es richtig verstehe, mit Inverter meinst du den Wechselrichter und mit Panel ein PV-Modul?
Hm, wenn keine Module angeschlossen sind, wird nichts produziert, dann hat imho der Wechselrichter keinen Strom
und kann nichts schicken.
Oder verstehe ich dich falsch?