tbnobody/OpenDTU #70: Live-Daten in Version 0.1.19 werden nach einiger Zeit nicht mehr angezeigt

sivar2311, 2022-08-09:
Hallo und Herzlichen Dank für dieses wunderschöne Tool!

Gestern habe ich ein Update auf die Version 0.1.19 gemacht.
Seit dem gibt es allerdings Probleme mit den Live-Daten.
Diese werden nach einer gewissen Zeit nicht mehr angezeigt.
Ein Reload der Seite hilft nicht. Einzig ein Restart des ESP scheint zu helfen.

Bei der Version 0.1.18 tritt dieser Fehler nicht auf.

Hierzu habe ich ein [Video](https://youtu.be/1YDhuocKNBs) gemacht, auf dem beide Versionen parallel zu sehen sind (2 ESPs).
Links: Version 0.1.19 / Rechts Version 0.1.18

Der Fehler ist ab Minute [2:30](https://youtu.be/1YDhuocKNBs?t=150) zu sehen

Im Serial-Log wird kein Fehler dazu ausgegeben.

Das Update von 0.1.18 auf 0.1.19 habe ich zunächst per OTA durchgeführt (Settings / Firmware Upgrade).
Nachdem ich den Fehler bemerkt hatte, habe ich die Firmware noch einmal direkt geflasht.
Auch ein löschen des .pio Verzeichnis, kompletter re-build und erneuter upload haben nicht geholfen.

Gruß
Sivar2311

hubsi5, 2022-08-09:
Passiert das auch, wenn nur ein Stick läuft?
Wenn ich zwei parallel laufen lasse, stören sich auch beide gegenseitig.

sivar2311, 2022-08-09:
Ja, der Fehler passiert auch bei nur einem laufenden ESP.
Den zweiten ESP hatte ich nur zur Veranschaulichung angeschlossen - sonst läuft bei mir nur einer.

docklugscheiss, 2022-08-09:
> Hallo und Herzlichen Dank für dieses wunderschöne Tool!
>
> Gestern habe ich ein Update auf die Version 0.1.19 gemacht. Seit dem gibt es allerdings Probleme mit den Live-Daten. Diese werden nach einer [...]

Habe exakt dasselbe Problem.

tbnobody, 2022-08-09:
Ich hatte das Problem hier bisher einmal. Aber habe es heute über den Tag beobachtet und auch bei einer Uptime von >23h trat es nicht mehr auf. Ich habe aber die Erfahrung gemacht, das es speziell nach einem neu-flashen des ESPs auftritt wenn zur gleichen Zeit noch die WebGUI offen war. Nicht das hier im ESPAsyncWebserver und den WebSockets etwas durcheinander kommt. 
Wenn das Problem auftritt, was sagt denn die Heap anzeige (Info --> System) in diesem Moment?

sivar2311, 2022-08-09:
Der war m. E. nicht auffällig. 
Genaue Werte kann ich allerdings erst morgen wieder liefern.

tbnobody, 2022-08-09:
Interessant wäre in diesem Zusammenhang (neben den Heap Daten) dann ebenfalls der Git-Hash der auf der System Seite angezeigt wird.
Hast du MQTT aktiv?

sivar2311, 2022-08-09:
Nein, MQTT nutze ich (bisher) noch nicht.
Git-Hash ist 778cce2

sivar2311, 2022-08-10:
Der Fehler tritt auch auf, wenn noch keine Verbindung zum HM besteht - also kann ich doch schon Werte liefern:
Version 0.1.19 zum Zeitpunkt des Fehlers:
[image]

Zum Vergleich was der ESP mit 0.1.18 anzeigt:
[image]

Beide ESP's wurden ungefähr zur selben Zeit heute morgen eingeschaltet.

tbnobody, 2022-08-10:
Hm ich muss noch etwas mit Fragen nerven um das genau nachzuvollziehen.
@docklugscheiss verwendest du MQTT?

Aber das Bild mit dem erhöhten Heap verbrauch hatte ich mir schon gedacht. Die Frage ist wo es herkommt.
Nachdem im Serial Log keine Meldung bzgl. "Too many messages queued" erscheint (käme vom WebSocket Server) würde ich das ausschließen.

Die einzige relevante Änderung die zwischen den beiden Versionen eingeflossen ist, ist eigentlich der wechsel des MQTT Clients damit dieser TLS unterstützt.
https://github.com/tbnobody/OpenDTU/compare/12df602...778cce2

Du meinstest oben um 5:xx Uhr, das beide ESPs heute morgen zur gleichen Zeit eingeschaltet wurden. Ich würde daraus schließen das das Problem relativ schnell auftritt? (Frage für eine Nachvollziehbarkeit, wenn ich bei mir MQTT deaktiviere)

Hattest du den Source selbst kompiliert oder ein fertiges .bin verwendet?

hubsi5, 2022-08-10:
Hilft zwar nicht, aber ich verwende MQTT und die neueste Version und habe dieses Problem nicht

tbnobody, 2022-08-10:
> Hilft zwar nicht, aber ich verwende MQTT und die neueste Version und habe dieses Problem nicht

Hilft auf jeden Fall wenn @docklugscheiss sagt das er kein MQTT verwendet. Dann gibt es nämlich ein Problem bei deaktiviertem MQTT und ich weiß eher wo ich suchen muss :)

docklugscheiss, 2022-08-10:
Hallo Zusammen, ich verwende MQTT nicht. 

Danke, dass ihr an dem Thema dran seid. VG!

MG850, 2022-08-10:
Hallo,

habe das gleiche Verhalten (noch keine WR in Betrieb).
Der Heap sieht hier im Zeitverlauf so aus:

20 min = 43% OK
34 min = 46% OK
38 min = 47% OK
42 min = 48% OK
46 min = 49% OK
49 min = 50% OK
52 min = 51% OK
56 min = 52% OK
60 min = 53% OK
64 min = 54% OK
68 min = 55% OK
78 min = 57% nur noch Buttons da
80 min = 58% ab hier kommt nix mehr
82 min = 59% 
86 min = 59% 
87 min = 60% 
91 min = 61%
96 min = 62%
98 min = 63%
102 min= 64%
118 min= 68%
127 min= 71%
144 min= 76%
172 min= 79%
175 min= Absturz, Webserver/Ping anwortet nicht mehr

Vielleicht hilft das was.

Grüße an die Programmier-Helden

tbnobody, 2022-08-10:
@MG850 hast du aktuell MQTT aktiviert?

MG850, 2022-08-10:
MQTT ist nicht aktiviert.

sivar2311, 2022-08-10:
Als der WR heute früh noch inaktiv war, hat es bei mir auch wesentlich länger gedauert, bis der Fehler aufgetreten ist.
Wenn der WR aktiv ist und Antworten sendet, dauert es nur ein paar Minuten.

MQTT hatte ich versucht zu konfigurieren, aber der Hostname war länger als die max. erlaubten Zeichen. Es ist bei mir also eine halbe MQTT Konfiguration vorhanden, aber nicht aktiv.

tbnobody, 2022-08-10:
Glaube ich hab den Fehler gefunden. Sollte hier gefixed sein: 80f7ba86d7df21554a7b95f2df63a3601e48ca8c

Ggf. bitte um Rückmeldung ob das jetzt noch auftritt.

tbnobody, 2022-08-10:
> Off-Topic: Ich überlege ob man das als Library "umverpacken" kann (natürlich ohne das Webinterface)

Das komplette Hoymiles Interface ist als Library implementiert. Der RF Teil sowie alle Commands, Parser usw. liegen alle unter lib/Hoymiles. In den anderen Ordnern liegt nur der Kram um ein Interface nach außen zu generieren.

sivar2311, 2022-08-10:
> Glaube ich hab den Fehler gefunden. Sollte hier gefixed sein

Erster Test sieht sehr gut aus! 
Der WR bekommt um diese Zeit (20:55 Uhr) gerade noch genug Saft und sendet fleißig Antworten :)
Der Heap-Wert pendelt zwischen 37% und 38%.

> Das komplette Hoymiles Interface ist als Library implementiert.

Perfekt, dann werde ich mich mal diesen Teil des Sourcodes bedienen und damit ein wenig spielen :)

docklugscheiss, 2022-08-11:
... auch bei mir kann ich vermelden: funktioniert nun tadellos. **Danke** an das Fixing - und auch für die Mühe und Arbeit, die im gesamten Thema steckt! (und wenn die Leistungsreduzierung in OpenDTU funktioniert, mache ich Luftsprünge!)

tbnobody, 2022-08-11:
Dann mache ich den Issue hier erstmal zu...

closed 2022-08-11 by tbnobody

sivar2311, 2022-08-11:
Sorry, bin gerade erst wieder zu Hause.
Bisher sind keine weiteren Probleme aufgetreten. Vielen Dank für den schnellen fix!

commit 80f7ba86 (2022-08-10): Fix #70: MQTT Messages where published even if no broker connection was established