tbnobody/OpenDTU #71: Feature Request - Lange Servername

juergen-dietz, 2022-08-10:
Hi,
euer Projekt läuft sehr gut mit meinem neuen HM-600, vielen Dank.

Könnt Ihr die länge des Servernamen ( MqTT Broker Parameter ) auf 64 Zeichen erweitern ?
Leider weiss ich nicht wie man den webapp Teil neu erzeugt.

tbnobody, 2022-08-10:
Hallo,
der webapp Teil ist sogar der einfachere Teil. Ich muss gucken wie ich die config so umbauen kann, das sie für bestehende Benutzer kompatibel ist. (Normal kein Problem aber man darf nicht einfach die Länge dort ändern)
Bei der Gelegenheit würde ich die Länge vmtl. direkt auf 128 Zeichen erhöhen. 
Max DNS Name Länge ist 253 Zeichen aber das fände ich dann doch etwas übertrieben.

closed 2022-08-11 by tbnobody

juergen-dietz, 2022-08-11:
Hi,
super es klappt, vielen Dank. 
Ich kann bestätigen das jetzt die MQTT Verbindung zu HiveMQ-Cloudinstanzen funktioniert.
Zur Info, nach dem Firmwareupdate auf [217fddf] hatten sich beim Inverter die "Max Power" Werte verändert.

commit 35a72acf (2022-08-11): Fix #71: Extend mqtt hostname from 31 characters to 128 characters