tbnobody/OpenDTU #73: Victron ve.direct interface und Leistungsbegrenzung

helgeerbe, 2022-08-11:
Hallo Thomas,

das ist kein Issue. Vielleicht magst du in GitHub die Discussions aktivieren?

Ich habe aktuell meinen Batterielader von Victron über das ve.direct Protokoll (HardwareSerial) in OpenDTU integriert. 

Von der Idee:
- Der virtuelle Lastausgang des Baterieladers schaltet den Inverter an/aus (aktuell mache ich das über einen Shelly auf der AC Seite). So wird die Tiefenentladung der Batterie verhindert. Bzw. muss ein gewisser Ladestand erreicht werden, bevor der Lausausgang wieder freigeschaltet wird.
- In Abhängigkeit des exportierten Stroms (gemessen über Shelly EM3) würde ich dann gerne die Leistung des Hoymiles regeln, so dass möglichst kein Strom zum Stromversorger abfliesst.

Dazu hätte ich 2 Fragen. 
1. Die Batterie liefert 24 V. Laut Datenblatt ist da MPPT nicht aktiv. Läuft ein Hoymiles HM-300 sauber mit dem entsprechenden Batteriestrom an?
2. Man kann neben der Leistungsbegrenzung den Inverter auch abschalten (laut ahoy code). Funktioniert das? Und könnte ich damit auf den Shelly verzichten?

Testen kann ich das ganze noch nicht, da ich für die Batterie einen China-Inverter habe, der über ein Poti fix auf 160 Watt gestellt ist.

Aber einen Hoymiles-300 für die Grundlast könnte ich einfach an mein HM-600 Balkonkraftwerk anschliessen. Der esp32 könnte das Gesamtsystem prima regeln.

FSys77, 2022-08-12:
> Vielleicht magst du in GitHub die Discussions aktivieren?

Ja, das wäre sehr gut; dann hätte ich dort mir dir reden können :-)

Zwei Frage: 

> Ich habe aktuell meinen Batterielader von Victron über das ve.direct Protokoll (HardwareSerial) in OpenDTU integriert.

(1) Wie genau hast du das in OpenDTU integriert?

> Der virtuelle Lastausgang des Baterieladers schaltet den Inverter an/aus (aktuell mache ich das über einen Shelly auf der AC Seite).

Wie machst du das wenn du die Batterie wieder zuschaltest? Hast du da einen Vorschaltwiderstand? Oder klatscht du gleich die Batterie voll auf den Hoymiles drauf?

helgeerbe, 2022-08-12:
@tbnobody  Danke, dass du die Discussons aktiviert hast. Du müsstest auch irgendwo einen Button haben, um das Issue hier in eine Diskussion zu verwandeln.

helgeerbe, 2022-08-12:
@FSys77 
1. Ich hatte schon ein ve.direct interface auf einem esp32 laufen. Habe es jetzt mal quick und dirty in OpenDTU als lib eingebunden.
  - Über die Weboberfläche kannst du das interface aktivieren.
  - Entscheiden, ob bei jedem read der volle Datensatz per MQTT oder nur geänderte Daten gesendet werden sollen.
- Es fehlt die Anzeige der live Daten
- Daten werden so gesendet, wie sie ausgelesen werden. D. h. für live Daten muss man noch die Rohdaten interpretieren. Das mache ich aktuell in node-red.
- Es fehlt mqtt discovery

Zu 2. da ich die AC Seite schalte muss man da nicht viel beachten. Der Inverter fährt hoch und fängt dann irgendwann an zu arbeiten. Ist wie mit einem normalen Balkonkraftwerk. Der Inverter ist aber auch so eingestellt, dass er aktuell maximal ca. 150 W liefert.

FSys77, 2022-08-14:
> Zu 2. da ich die AC Seite schalte muss man da nicht viel beachten. Der Inverter fährt hoch und fängt dann irgendwann an zu arbeiten. Ist wie mit einem normalen Balkonkraftwerk. Der Inverter ist aber auch so eingestellt, dass er aktuell maximal ca. 150 W liefert.

D.h, du stellst sicher, dass der Akku immer geladen ist - und nie offline geht?

helgeerbe, 2022-08-15:
Nein. Der Batterielader hat einen sogenannten virtuellen Lastausgang an/aus, der über ve.direct signalisiert, ob die Batterie entladen werden darf. Untere Schranke ist der Tiefentladungsschutz. Obere Schranke habe ich bei ca. 70% Kapazität eingestellt.

Wenn also der virtuelle Lastausgang auf Aus schaltet, schalte ich den Shelly aus und der Inverter ist abgeschaltet. Steigt die Batterieladung auf 70%, geht der virtuelle Lastausgang auf An und ich schalte den Shelly ein. Der Inverter fährt hoch und liefert 150W.

Um keine Energie zu verschwenden schalte ich den Shelly auch ab, wenn mehr als 50W zum Energieversorger abfliessen.

Leider kann ich die Leistung meines Inverters nicht dynamisch regeln. Daher würde ich ihn gerne mit einem HM-300 ersetzen. So dass er je nach Bedarf zwischen 0-300 Watt liefert. Spannend wäre es eben zu wissen, ob ich ihn über mqtt auch komplett abschalten kann, so daß die Batterie nicht weiter belastet wird (Standby Verbrauch von ca. 1W wäre OK).

FSys77, 2022-08-15:
> Nein. Der Batterielader hat einen sogenannten virtuellen Lastausgang an/aus, der über ve.direct signalisiert, ob die Batterie entladen werden darf.

Die Wechselrichter haben ja Kondensatoren und anderes Zeugs verbaut. Wenn du da die Batterie auf einen Schlag zuschaltest fließen da 100+ Ampere auf einen Schlag rein. Darum meine Frage ob du das mittels Vorschaltwiderstand machst.

Crashdown1977, 2022-08-15:
Hallo. :)

> Die Wechselrichter haben ja Kondensatoren und anderes Zeugs verbaut. Wenn du da die Batterie auf einen Schlag zuschaltest fließen da 100+ Ampere auf einen Schlag rein. Darum meine Frage ob du das [...]

Ich betreib einen HM-600 bereits seit zwei Jahren an einem 24V Akku LiFePo4 an einem Eingang des HM600. Am anderen hängt ein Solarmodul. Ein Rlelais schaltet die 24v mit einem 100W Strombegrenzer, oder halt unbegrenzt. Einen hohen Einschaltstrom gibt es nicht. Der Inverter produziert nach etwa 15 Sekunden 325W. Oder halt die 100W. Wenn die Batteriespannung bei 100W unter 25V sank hat er sich aber manchmal abgeschaltet. Ich nutze jetzt Ahoy mit limitierung. Das klappt super bis auf die Tatsache dass der esp8266 oft neu startet. Ich nutze OpenDtu vorher zur Überwachung. Der hat sich nie aufgehängt. Daher hoffe ich auch auf die Limitierung hier. 
Gruß Timo

FSys77, 2022-08-15:
> Ein Rlelais schaltet die 24v mit einem 100W Strombegrenzer, oder halt unbegrenzt.

Welches Relais genau verwendest du da?

Crashdown1977, 2022-08-15:
> > Ein Rlelais schaltet die 24v mit einem 100W Strombegrenzer, oder halt unbegrenzt.
> 
> Welches Relais genau verwendest du da?

[image]

Diese hier...

Und einen step-down auf 12V für die Spulen. 
Geschaltet über NodeRed flow am Esphome mit Mosfet Treiber. 
[image]

Im Moment schalte ich aber nur noch VS ab wenn der Akku runter ist. Über MQTT> Inverter aus, wenn Strom = 0 > Relais aus. Wenn Akku 70% Relais an und Leistung nachführen in drei Stufen. Damit schaffe ich ca 4 kWh am Tag ohne Einspeisung wenn die Sonne brutzelt.

ntfrnd, 2022-08-19:
Hallo, 
ich probiere aktuell auch OpenDTU und ahoy aus, weil ich auch meinen HM-600 auslesen und die Leistungsabgabe begrenzen bzw. ausschalten will. 
Auch bei mir läuft ahoy sehr instabil und bootet immer wieder neu - egal was ich alles versucht habe. 
OpenDTU hingegen läuft recht gut. 
Von daher bin ich auch an einer Leistungsbegrenzung mittels MQTT interessiert. 
Angeblich soll das in ahoy funktionieren, wo finde ich den entsprechenden Eintrag? Dann könnte ich das dort probieren, bis es hoffentlich in OpenDTU implementiert wird 😀

Crashdown1977, 2022-08-19:
> Hallo, ich probiere aktuell auch OpenDTU und ahoy aus, weil ich auch meinen HM-600 auslesen und die Leistungsabgabe begrenzen bzw. ausschalten will. Auch bei mir läuft ahoy sehr instabil und bootet [...]

Das ist im User_Manual.md beschrieben

ntfrnd, 2022-08-20:
Ich finde nur eine Readme.md (https://github.com/grindylow/ahoy/blob/main/README.md) aber da steht diesbezüglich nichts drin. 
Hast Du für das User_Manual.md eingeben Link?

fila612, 2022-08-20:
https://github.com/grindylow/ahoy/blob/b8ed52b4a2ea2867197ceab711ee0015bf4d1e30/tools/esp8266/User_Manual.md

ntfrnd, 2022-08-20:
👍 Danke

helgeerbe, 2022-08-24:
So mein ve.direct interface läuft jetzt mit auf opendtu. Der esp32 scheint genug power zu haben.

Das Rest-Api `/api/vedirectlivedata/status/`:

````json
{
"data_age":4,
"age_critical":false,
"PID":"SmartSolar MPPT 100|30",
"SER":"HQ2210UxxxH",
"FW":"159",
"LOAD":"ON",
"CS":"Bulk",
"ERR":"No error",
"OR":"Not off",
"MPPT":"MPP Tracker active",
"HSDS":{"v":41,"u":"Days"},
"V":{"v":26.7,"u":"V"},
"I":{"v":1.7,"u":"A"},
"VPV":{"v":37.48,"u":"V"},
"PPV":{"v":50,"u":"W"},
"H19":{"v":75.45,"u":"kWh"},
"H20":{"v":2.16,"u":"kWh"},
"H21":{"v":660,"u":"W"},
"H22":{"v":1.96,"u":"kWh"},
"H23":{"v":734,"u":"W"}
}
````

Und die Livedata:
[image]

Per MQTT werden die Rohdaten gesendet.

Was fehlt:
- Home Assistant discovery
- MQTT Empfang Einspeisung der Energie ins Grid (Energieversorger)
- Lokale Steuerung des Inverters 
  - Virtueller Lastausgang des Batterieladers soll den Inverter einschalten, wenn die Batterie genug gladen ist, bzw. den Inverter abschalten, um die Tiefenentladung zu verhindern.
  - Regelung der Leistungsbegrenzung je nach Einspeisung zum Netzbetreiber (Ziel: Null-Einspeisung)