tbnobody/OpenDTU #74: Verwendung der Hoymiles.h als Library

sivar2311, 2022-08-11:
Hallo Thomas!

Eigentlich ist das kein Issue, sondern mehr eine Frage :)

Ich habe versucht die Hoymiles.h als Library in einem kleinen [Testsketch](https://github.com/sivar2311/HoymilesTest/blob/main/src/main.cpp) zu benutzen.
Allerdings bekomme ich keine Daten geliefert. 
Kannst du mir einen Tipp geben, was ich falsch mache?

(Ich verwende die gleichen Seriennummern wie im openDTU).

Gruß
Boris

tbnobody, 2022-08-11:
Hi, 
ich vermute das du in der Konsole nicht mal TX Meldungen bekommst. Das liegt wohl daran das du keine Zeit setzt. Es muss die aktuelle Zeit hinterlegt sein da diese in jedem Paket mitgeschickt wird.

sivar2311, 2022-08-11:
Ah, stimmt :) Da war ja was!
Danke für den Hinweis - Ich implementiere schnell mal NTP dazu.

sivar2311, 2022-08-11:
Ja, das war des Rätsels Lösung!
Vielen Dank 👍

closed 2022-08-11 by sivar2311