tbnobody/OpenDTU #76: Crash after setting Inverter

andreash-esp, 2022-08-14:
My opendtu testcase reboots after setting an (invalid) inverter number/name.

I can integrate it into WLAN and then try to set Inverter serial to "1" and name to "Michael". My real inverter will only arrive next week, but I have tried with valid serial numbers as well.

The ESP32 immediately goes into a boot loop:
11:31:21.438 > Starting OpenDTU
11:31:21.438 > Initialize FS... done
11:31:21.438 > Reading configuration... done
11:31:21.438 > Initialize Network... done
11:31:21.438 > Setting Hostname... Configuring WiFi STA using new credentials... done
11:31:21.438 > Initialize NTP... done
11:31:21.438 > Initialize MqTT... done
11:31:21.438 > Initialize WebApi... done
11:31:21.438 > Initialize Hoymiles interface... Connection successfull
11:31:21.438 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
11:31:21.438 >
11:31:21.438 > Core  1 register dump:
11:31:21.438 > PC      : 0x40188f5b  PS      : 0x00060930  A0      : 0x800e19aa  A1      : 0x3ffb27b0
11:31:21.438 > A2      : 0x00000000  A3      : 0x3ffc413c  A4      : 0x00000000  A5      : 0x00000999
11:31:21.438 > A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800ee13e  A9      : 0x3ffb2750
11:31:21.438 > A10     : 0x00000000  A11     : 0xffffeedf  A12     : 0x00000001  A13     : 0x3ffc470c
11:31:21.438 > A14     : 0x0000004b  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
11:31:21.438 > EXCVADDR: 0x000001d0  LBEG    : 0x4008464d  LEND    : 0x40084655  LCOUNT  : 0x00000027

Reflashing the code via platformIO does not fix the boot loop. I have to clear the flash with some other code and only then can reapply openDTU.

tbnobody, 2022-08-14:
Beside the issue (I will add some additional debug output) you can, instead of uploading another code, also just execute the "Erase Flash" command in the Project Tasks
[image]

(By uploading the opendtu bin again, the data partition which contains the configuration data will not be overwritten. Otherwise all settings will be lost after an upgrade.

andreash-esp, 2022-08-16:
This is what I found:
Setting an invalid Inverter Serial crashes openDTU. I must have used an invalid S/N two days ago as well.

I tried using "1" and "100000000000" (a valid S/N has 12 digits). Both crash at this step:

18:34:36.064 > Initialize Hoymiles interface... Connection successfull
18:34:36.077 >   Setting radio PA level...
18:34:36.080 >   Setting DTU serial...
18:34:36.082 >   Setting poll interval...
18:34:36.084 >   Adding inverter: 1 - MichaelGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

tbnobody, 2022-08-16:
Thank you for reporting this issue. Please feel free to open it again if it still occours in the latest version

commit 14569746 (2022-08-16): Fix #76: Null Pointer exception when adding a inverter with unkonwn serial number
closed 2022-08-16 by tbnobody