tbnobody/OpenDTU #84: TSUN TSOL-M800 - keine Daten

Depio, 2022-08-16:
Hallo zusammen,

ich habe einen TSUN TSOL-M800 WR, welchen ich gerne überwachen würde.

Leider bekomme ich keine Daten aus dem WR.

`done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Hoymiles interface... Connection successfull
  Setting radio PA level... 
  Setting DTU serial... 
  Setting poll interval... 
  Adding inverter: 114182207126 - TSUN TSOL-M800 done
done
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using existing credentials... E (917) wifi:Association refused temporarily, comeback time 1024 mSec
E (922) wifi:sta is connecting, return error
[   276][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
done
WiFi disconnected
Try reconnecting
Network lost connection
WiFi connected
WiFi got ip: 192.168.178.68
Network connected
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C1 0 0 0 0 0 0 0 0 47 34 E5 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 40 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C1 0 0 0 0 0 0 0 0 47 34 E5 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 61 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C1 0 0 0 0 0 0 0 0 47 34 E5 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 75 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C1 0 0 0 0 0 0 0 0 47 34 E5 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 3 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C1 0 0 0 0 0 0 0 0 47 34 E5 
RX Period End
All missing
Nothing received, resend count exeeded
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C6 0 0 0 0 0 0 0 0 77 12 F4 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 40 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C6 0 0 0 0 0 0 0 0 77 12 F4 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 61 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C6 0 0 0 0 0 0 0 0 77 12 F4 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 75 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C6 0 0 0 0 0 0 0 0 77 12 F4 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 3 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C6 0 0 0 0 0 0 0 0 77 12 F4 
RX Period End
All missing
Nothing received, resend count exeeded
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD CB 0 0 0 0 0 0 0 0 E7 4A 31 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 40 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD CB 0 0 0 0 0 0 0 0 E7 4A 31 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 61 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD CB 0 0 0 0 0 0 0 0 E7 4A 31 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 75 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD CB 0 0 0 0 0 0 0 0 E7 4A 31 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 3 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD CB 0 0 0 0 0 0 0 0 E7 4A 31 
RX Period End
All missing
Nothing received, resend count exeeded
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D0 0 0 0 0 0 0 0 0 17 F4 64 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 40 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D0 0 0 0 0 0 0 0 0 17 F4 64 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 61 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D0 0 0 0 0 0 0 0 0 17 F4 64 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 75 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D0 0 0 0 0 0 0 0 0 17 F4 64 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 3 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D0 0 0 0 0 0 0 0 0 17 F4 64 
RX Period End
All missing
Nothing received, resend count exeeded
Websocket: [/livedata][1] connect
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D5 0 0 0 0 0 0 0 0 47 CB E 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 40 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D5 0 0 0 0 0 0 0 0 47 CB E 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 61 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D5 0 0 0 0 0 0 0 0 47 CB E 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 75 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D5 0 0 0 0 0 0 0 0 47 CB E 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 3 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD D5 0 0 0 0 0 0 0 0 47 CB E 
RX Period End
All missing
Nothing received, resend count exeeded
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DA 0 0 0 0 0 0 0 0 B7 8A B0 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 40 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DA 0 0 0 0 0 0 0 0 B7 8A B0 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 61 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DA 0 0 0 0 0 0 0 0 B7 8A B0 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 75 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DA 0 0 0 0 0 0 0 0 B7 8A B0 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 3 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DA 0 0 0 0 0 0 0 0 B7 8A B0 
RX Period End
All missing
Nothing received, resend count exeeded
Websocket: [/livedata][1] disconnect
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DF 0 0 0 0 0 0 0 0 E7 B5 DA 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 40 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DF 0 0 0 0 0 0 0 0 E7 B5 DA 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 61 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DF 0 0 0 0 0 0 0 0 E7 B5 DA 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 75 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DF 0 0 0 0 0 0 0 0 E7 B5 DA 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 3 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD DF 0 0 0 0 0 0 0 0 E7 B5 DA 
RX Period End
All missing
Nothing received, resend count exeeded
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD E4 0 0 0 0 0 0 0 0 D6 92 F7 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 40 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD E4 0 0 0 0 0 0 0 0 D6 92 F7 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 61 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD E4 0 0 0 0 0 0 0 0 D6 92 F7 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 75 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD E4 0 0 0 0 0 0 0 0 D6 92 F7 
RX Period End
All missing
Nothing received, resend whole request
TX Channel: 3 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD E4 0 0 0 0 0 0 0 0 D6 92 F7 
RX Period End
All missing
Nothing received, resend count exeeded
`

HacksBugsAndRockAndRoll, 2022-08-16:
Hi, so sieht das bei mir aus, wenn ich die DTU ausserhalb Reichweite zum WR aufstelle.

Depio, 2022-08-16:
Hi, danke für deine Rückmeldung.
leider kann ich das ausschließen. Es scheint irgendwas mit der Kommunikation zum WR zu tun zu haben 👎

Solateur, 2022-08-19:
Kann es sein, dass dieser TSUN TSOL-M800 mit einer Firmware läuft, die hier nicht unterstützt wird?

Siehe readme.md:
"Currently supported Inverters

    Hoymiles HM-300
   .
   .
    Hoymiles HM-1500
    TSUN TSOL-M800 (**Maybe depending on firmware on the inverter**)"

tbnobody, 2022-08-19:
> Kann es sein, dass dieser TSUN TSOL-M800 mit einer Firmware läuft, die hier nicht unterstützt wird?

Auf das würde ich aktuell auch schätzen

stefan123t, 2022-09-19:
Stimmt die Serial Id 1141 und vor allem auch der Rest ? 
Klappt es denn mit der aktuellen Ahoy Version ? 
Was für eine Firmware Version liefert das InverterDevInform_All 0x01 Kommando zB in Ahoy ?

stefan123t, 2022-10-07:
@Depio warum kommen denn die Fehlermeldungen des WiFi STA Modes am Anfang ?
```
 Configuring WiFi STA using existing credentials... E (917) wifi:Association refused temporarily, comeback time 1024 mSec
E (922) wifi:sta is connecting, return error
[ 276][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
done
WiFi disconnected
Try reconnecting
Network lost connection
WiFi connected
```

Die NTP Abfrage scheint aber funktioniert zu haben zumindest sendet er die richtige Zeit an den WR:
0x`62 FB AD C1` = 1660661185 und das ist als UNIX Epoch timestamp: Tuesday, August 16, 2022 2:46:25 PM
```
Fetch inverter: 114182207126
TX Channel: 23 --> 15 82 20 71 26 78 56 34 12 80 B 0 62 FB AD C1 0 0 0 0 0 0 0 0 47 34 E5
RX Period End
All missing
Nothing received, resend whole request
```

@tbnobody wovon stammen eigentlich die folgende WebSocket-Meldung zwischen den Kommunikations-Versuchen:
```
Nothing received, resend count exeeded
Websocket: [/livedata][1] disconnect
```

tbnobody, 2022-10-07:
> @tbnobody wovon stammen eigentlich die folgende WebSocket-Meldung zwischen den Kommunikations-Versuchen:

Wenn man halt im Browser auf eine anderen Ansicht  umschaltet oder den Tab schließt wird der Websocket zu gemacht. Der ist nur in der LiveView aktiv.

tbnobody, 2022-10-22:
Will close this issue because of no further response

closed 2022-10-22 by tbnobody