tbnobody/OpenDTU #92: Home Assistant MQTT Auto Discovery - Sensor "flattern"

itn3rd77, 2022-08-19:
Hallo Thomas,

mit ist heute erst aufgefallen, dass die erkannten Sensoren per Home Assistant MQTT Auto Discovery  ständig "flatten". D.h. anstatt einfach nur zu Updaten sind die immer kurz "Nicht verfügbar". Da passt wohl irgentwas mit dem Update der Topics nicht. Bei anderen Sensoren die ich per Home Assistant MQTT Auto Discovery  in meinem System haben zeigt sich das Verhalten nicht. Hier mal ein Video das ist wohl sprechender:

https://user-images.githubusercontent.com/2655169/185608858-cb625f68-fd3a-4e2c-ac9b-94b2eeb43a15.mp4

Hast Du eine Idee? Ich gehe auch mal auf die Suche im Code.

Beste Grüße
Ingo

itn3rd77, 2022-08-19:
Hallo Thomas,

ich denke ich habe den Grund gefunden. Du hast die Option "expire_after" genutzt und auf 10 Sekunden eingestellt.
Hatte bisher in dieser Form noch nicht gesehen. Die Option macht ja durchaus Sinn (Sensor in der Nacht nicht verfügbar), aber evtl. sollten der Wert etwas großzügiger gewählt werden, oder?

[image]

Macht es evtl. auch Sinn die Option "force_update" zu aktivieren? Anhand der Beschreibung finde ich es sinnvoll.

itn3rd77, 2022-08-19:
**Update**

Das DTU Poll Interval: 5
Das MQTT Publish Interval: 5

Ich habe 3 Inverter in OpenDTU registriert.
In der Live-Ansicht sehe ich das "Data Age" immer bis 15 hoch läuft.

tbnobody, 2022-08-19:
Hallo Ingo,

ich merke den Fehler. Das DTU Poll Interval ist eigentlich die Waiting Time zwischen den anfragen an die Wechselrichter. D.h. um alle 3 Wechselrichter abzufragen, benötigt man 3x5 Sek + ca. 1 Sek pro Wechselrichter zum Schreiben/Lesen der Werte. Also knapp 18Sek.

Leider ist das expire interval für HASS aktuell so implementiert: https://github.com/tbnobody/OpenDTU/blob/231c1903ff3586373cc564a090d908506f55a245/src/MqttHassPublishing.cpp#L119

Das ist ein Fehler. Ich bin nur leider gerade nicht in Rechner nähe. Wird bis nächste Woche dauern.

itn3rd77, 2022-08-19:
Hallo Thomas,

danke für die Rückmeldung und mach Dir keinen Stress wegen einem möglichen Fix. Ursache identifiziert - perfekt!

Beste Grüße
Ingo

szepnorbee, 2022-08-21:
> Hallo Thomas,
> 
> mit ist heute erst aufgefallen, dass die erkannten Sensoren per Home Assistant MQTT Auto Discovery ständig "flatten". D.h. anstatt einfach nur zu Updaten sind die immer kurz "Nicht verfügbar". Da passt wohl irgentwas mit dem Update der Topics nicht. Bei anderen Sensoren die ich per Home Assistant MQTT Auto Discovery in meinem System haben zeigt sich das Verhalten nicht. Hier mal ein Video das ist wohl sprechender:
> 
>  19-08-_2022_13-22-24.mp4 
> Hast Du eine Idee? Ich gehe auch mal auf die Suche im Code.
> 
> Beste Grüße Ingo

Hallo! Kann ich Ihre configuration.yaml-Datei haben? Leider kann ich es nicht unter homeassistant auf MQTT zum Laufen bringen. Das würde mir sehr helfen (nur dieser Teil der Datei reicht aus).

itn3rd77, 2022-08-23:
Hallo Thomas,

ich habe jetzt noch ein bisschen mehr in Home Assistant gemacht, als reine Anzeige von Werten. Da wurde mir schnell klar, das es wohl besser wäre komplett auf die "expire_after" Option zu verzichten. Ich mache z.B .jetzt verschiedene Berechnungen und das alles fliegt einem um die Ohren sobald die Sensoren "unavaiable" werden.

Hättest Du was bestimmtes im Sinn als Du dich für die Option entschieden hättest?

tbnobody, 2022-08-23:
Hallo Ingo, das expire_after macht schon sinn. Man möchte ja z.B. wissen wenn keine gültigen Daten mehr angezeigt werden. Also z.B. nach 15min ohne Aktualisierung kann man wohl davon ausgehen das kein Wert der angezeigt wird mehr passt. In Regeln usw. macht es sowieso sinn Sensoren auf gültige Werte abzufragen. Das kann nämlich auch passieren wenn der MQTT Broker weg ist, nach einem Neustart von HASS mit nicht retainten Nachrichten usw...

closed 2022-08-27 by tbnobody

battifreund, 2022-08-30:
Das Thema hat mir keine Ruhe gelassen. ;) Ich habe mir eine Option eingebaut, mit der ich das Setzen von "expire_after" ein/aus-schalten kann. Ich schiebe es mal in einen Pull Request. Vielleicht möchte es ja sonst noch jemand haben.

commit dbf692d9 (2022-08-27): Fix #92: Expiration time depends on the amount of inverters