tbnobody/OpenDTU #117: HM800 keine Antwort

daja64, 2022-09-01:
Hallo, nachdem ich mich mehr recht als schlecht hier durchgewurschtelt habe, vieles ist immer noch unklar, 
läuft nun mein ESP32-Pico und ein NRF24l01+(laut Label) so einigermaßen.
So startet er 
22:39:22.611 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:39:22.611 > configsip: 188777542, SPIWP:0xee
22:39:22.611 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:39:22.611 > mode:DIO, clock div:2
22:39:22.611 > load:0x3fff0030,len:1184
22:39:22.611 > load:0x40078000,len:13132
22:39:22.611 > load:0x40080400,len:3036
22:39:22.611 > entry 0x400805e4
22:39:22.611 > 
22:39:22.611 > Starting OpenDTU
22:39:22.611 > Initialize FS... done
22:39:22.611 > Reading configuration... done
22:39:22.611 > Initialize Network... E (1211) esp.emac: emac_esp32_init(349): reset timeout
22:39:22.611 > E (1212) esp_eth: esp_eth_driver_install(214): init mac failed
22:39:22.611 > [  1216][E][ETH.cpp:323] begin(): esp_eth_driver_install failed
22:39:22.611 > done
22:39:22.611 > Setting Hostname... Configuring WiFi STA using new credentials... done
22:39:22.611 > Initialize NTP... done
22:39:22.611 > Initialize MqTT... done
22:39:22.611 > Initialize WebApi... done
22:39:22.611 > Initialize Hoymiles interface... Connection successfull
22:39:22.611 >   Setting radio PA level.. 
22:39:22.611 >   Setting DTU serial... 
22:39:22.611 >   Setting poll interval... 
22:39:22.611 >   Adding inverter: 114180124306 - hm800 done
22:39:22.611 > done
22:39:22.611 > Switch to WiFi mode
22:39:22.611 > E (1284) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb26e0 (config or if_key is NULL or duplicate key)
22:39:22.611 > E (1285) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
22:39:22.611 > [  1295][E][ETH.cpp:270] begin(): esp_eth_mac_new_esp32 failed
22:39:22.611 > Setting Hostname... WiFi connected
22:39:22.611 > done
22:39:22.611 > Configuring WiFi STA using existing credentials... done
22:39:22.611 > Configuring WiFi STA static IP... WiFi got ip: 192.168.178.66
22:39:22.611 > Network connected
22:39:22.611 > done
22:39:22.611 > Fetch inverter: 114180124306

jetzt sind aber Fehlermeldungen, die ich nicht vertsehe. Dann gehts so weiter:
22:39:22.611 > TX Channel: 23 --> 15 80 12 43 6 78 56 34 12 80 B 0 63 F C6 F3 0 0 0 0 0 0 0 0 8D E0 75 
22:39:22.611 > RX Period End
22:39:22.611 > All missing
22:39:22.611 > Nothing received, resend whole request
22:39:22.611 > TX Channel: 40 --> 15 80 12 43 6 78 56 34 12 80 B 0 63 F C6 F3 0 0 0 0 0 0 0 0 8D E0 75 
22:39:22.611 > RX Period End
22:39:22.611 > All missing
22:39:22.611 > Nothing received, resend whole request
22:39:22.611 > TX Channel: 61 --> 15 80 12 43 6 78 56 34 12 80 B 0 63 F C6 F3 0 0 0 0 0 0 0 0 8D E0 75 
22:39:22.611 > RX Period End
22:39:22.611 > All missing
22:39:22.611 > Nothing received, resend whole request
22:39:22.611 > TX Channel: 75 --> 15 80 12 43 6 78 56 34 12 80 B 0 63 F C6 F3 0 0 0 0 0 0 0 0 8D E0 75 
22:39:22.611 > RX Period End
22:39:22.611 > All missing
22:39:22.611 > Nothing received, resend whole request
22:39:22.611 > TX Channel: 3 --> 15 80 12 43 6 78 56 34 12 80 B 0 63 F C6 F3 0 0 0 0 0 0 0 0 8D E0 75 
22:39:22.611 > RX Period End
22:39:22.611 > All missing
22:39:22.611 > ..... und von vorn wieder

Egal auch wenn ich unmittelbar neben dem WR stehe, kein Kontakt. Wer kann mir helfen und gibt es eine möglichkeit zu prüfen, on es wirklich so ein NRF-Chip ist. Danke schon mal.

tbnobody, 2022-09-01:
Ist es bei dir um 22:39 noch hell? Der Inverter ist nur aktiv wenn die Sonne scheint.

daja64, 2022-09-01:
Nein das ist nur ein Beispiel, das sah heute Vormittag genau so aus.

tbnobody, 2022-09-01:
Es scheint als hättest du OpenDTU mit Ethernet Support compiliert. Hast du selbst ein environment angelegt? Hier kann es ggf. zu Pin Konflikten zwischen den Ethernet Pins und den NRF pins kommen. Gibt es einen Grund warum du nicht einfach das Generic Profil verwendet hast?

Der IRQ Pin ist korrekt verdrahtet? Hast du ggf. einen Link zum verwendeten NRF Modul um zu prüfen ob es ein NRF24+ ist.

daja64, 2022-09-02:
Sooo, jetzt weiter.
Ich habe dein Projekt komplett entpackt, die Software so wie beschrieben installiert. Starte ich upload gab es danach eine Fehlermeldung. Also versuche ich diese Platformio-Geschicht zu verstehen, Basis ist etwas Assembler, Basic(Bascom),und etwas C, und habe dann die Platformio.ini für meine Zwecke angepasst (Nicht schon wieder neue Hardware kaufen).
Es war wahrscheinlich diese Zeile in der ini "-DOPENDTU_ETHERNET", die eine korrekte Funktion verhindert hat.
So sieht meine ini aus:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = esp32

[env]
framework = arduino
platform = espressif32@>=5

build_flags =
    -D=${PIOENV}
    -DCOMPONENT_EMBED_FILES=webapp_dist/index.html.gz:webapp_dist/zones.json.gz:webapp_dist/favicon.ico:webapp_dist/js/app.js.gz

lib_deps =
    https://github.com/me-no-dev/ESPAsyncWebServer.git
    bblanchon/ArduinoJson @ ^6.19.4
    https://github.com/bertmelis/espMqttClient.git
    nrf24/RF24 @ ^1.4.5

extra_scripts =
    pre:auto_firmware_version.py

board_build.partitions = partitions_custom.csv
board_build.filesystem = littlefs
monitor_filters = time, colorize, log2file, esp32_exception_decoder
monitor_speed = 115200
upload_protocol = esptool

[env:esp32]

board = pico32
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=21
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5

monitor_port = COM3
upload_port = COM3

Und heute hatte ich ersten Kontakt mit dem WR.

09:33:26.156 > rst:0x1 (POWERON_RESET),boot:0x13 (S[37mP[37mI_FAST_FLASH_BOOT)
09:33:26.161 > configsip: 188777542, SPIWP:0xee
09:33:26.161 > clk_drv:0[37mx[37m00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[37m09:33:26.171 > m[37mode:DIO, clock div:2
09:33:26.172 > load:0x3fff0030,len:1184
09:33:26.172 > load:0x40078000[37m,[37mlen:13132
09:33:26.178 > load:0x40080400,len:3036
09:33:26.178 > entry 0x400805e4
[37m09:33:26.763 > 
[37m
09:33:26.764 > Starting OpenDTU
09:33:26.764 > Initialize FS... [37md[37mone
09:33:26.770 > Reading configuration... [37md[37mone
09:33:26.818 > Initialize Network... [37md[37mone
09:33:26.942 > Setting Hostname... Configuring WiFi STA using new credent[37mials... [37md[37mone
09:33:26.952 > Initialize NTP... done
09:33:26.952 > Initialize MqTT... done
09:33:26.952 > Initializ[37me[37m WebApi... [37md[37mone
09:33:26.964 > Initialize Hoymiles interface... [37mC[37monnection successfull
09:33:26.978 >   Setting radio PA level...
09:33:26.978 >   Setting [37mD[37mTU serial... 
09:33:26.983 >   Setting poll interval... 
09:33:26.983 >   Adding inverter: [37m1[37m14181300270 - hm800 done
09:33:26.989 > done
09:33:26.989 > Switch to WiFi mode
09:33:26.989 > Setting Ho[37ms[37mtname... done
09:33:26.994 > Configuring WiFi STA using existing credentials.[37m.[37m. E (916) wifi:sta is connecting, return error
09:33:27.002 > [   271][E][WiF[37mi[37mSTA.cpp:317] begin(): connect failed! 0x3007
09:33:27.008 > done
[37m09:33:27.012 > W[37miFi connected
[37m09:33:31.043 > W[37miFi got ip: 192.168.178.47
09:33:31.044 > Network connected
[37m09:33:31.730 > F[37metch inverter: 114181300270
[37m09:33:33.894 > W[37mebsocket: [/livedata][1] connect
[37m09:33:46.751 > F[37metch inverter: 114181300270
[37m09:34:01.773 > F[37metch inverter: 114181300270
09:34:01.774 > TX Channel: 23 --> 15 81 30 2 70 7[37m8[37m 56 34 10 80 B 0 63 11 B1 E9 0 0 0 0 0 0 0 0 34 80 C9 
[37m09:34:01.820 > I[37mnterrupt received
09:34:01.821 > RX 95 81 30 2 70 81 30 2 70 1 0 1 1 52 2 6D [37m8[37m 2F 1 52 2 74 8 47 0 0 E4 
[37m09:34:01.859 > I[37mnterrupt received
09:34:01.859 > RX 95 81 30 2 70 81 30 2 70 2 10 E9 0 0 11 E[37m [37m0 97 0 99 9 6C 13 89 F AC 23 
[37m09:34:01.896 > I[37mnterrupt received
09:34:01.897 > RX 95 81 30 2 70 81 30 2 70 83 0 0 0 A7 3 E8[37m [37m1 1E 0 5 BE 6B 95 
[37m09:34:02.012 > R[37mX Period End
09:34:02.012 > Success
[37m09:34:06.773 > F[37metch inverter: 114181300270
09:34:06.774 > Request device info
09:34:06.774 > TX Channel: 4[37m0[37m --> 15 81 30 2 70 78 56 34 10 80 B 0 63 11 B1 EE 0 0 0 0 0 0 0[37m [37m0 4 A6 D8 
[37m09:34:06.825 > I[37mnterrupt received
09:34:06.826 > RX 95 81 30 2 70 81 30 2 70 1 0 1 1 51 2 6E [37m8[37m 30 1 52 2 75 8 4A 0 0 F7 
[37m09:34:06.863 > I[37mnterrupt received
09:34:06.865 > RX 95 81 30 2 70 81 30 2 70 2 10 E9 0 0 11 E[37m [37m0 97 0 99 9 6A 13 88 F AF 27 
[37m09:34:06.922 > I[37mnterrupt received
09:34:06.925 > RX 95 81 30 2 70 81 30 2 70 83 0 1 0 A7 3 E8[37m 1 1E 0 5 A6 FA 1D 
[37m09:34:07.015 > R[37mX Period End
09:34:07.016 > Success

Bitt was hat es mit dem "Git Hash" Wert in der "System Info" auf sich, bei mir alles Nullen.

daja64, 2022-09-02:
Neue Frage: wo kann ich einstellen das die Zeit aus meinem Wlan geholt wird ?

tbnobody, 2022-09-02:
> Bitt was hat es mit dem "Git Hash" Wert in der "System Info" auf sich, bei mir alles Nullen.

Du hast vmtl. nicht wie in der README.md beschrieben den Sourcecode mit `git clone` gecloned sondern die ZIP Datei herunter geladen.

> wo kann ich einstellen das die Zeit aus meinem Wlan geholt wird ?

Uhrzeiten werden nicht aus dem WLAN geholt. Uhrzeiten werden mit einem NTP Server synchronisiert. NTP Einstellungen findest du in den Settings.

madmartin, 2022-09-04:
@daja64 
Scheinbar ist das Problem gelöst - wäre nett hier kurz die Ursache für das Problem zu schildern. Andere Leute können daraus vielleicht was lernen.

daja64, 2022-09-04:
Sorry, also ich dachte wenn ich die ini und das Log poste das reicht. Aber gut. 
In der ini habe ich diese Zeile entfernt "-DOPENDTU_ETHERNET" und die Verbindung an den ESP32pico angepasst.
Achso, bei der SSID hatt ich ein Leerzeichen unterschlagen.