tbnobody/OpenDTU #119: Firmware upgrade

gd35, 2022-09-03:
I have had to change the pin assignment due to a different ESP32 module (Joy-It). I generated a new firmware and tried to upload it with the "firmware Upgrade" item in the "Settings" menu. The upload stopped at 4%. Any idea why the upload stopped ? Thanks.

FWallenwein, 2022-09-03:
Same here. Tried to upload, and it stopped after 9%.

And a second question. 
When selection the firmware file, the browser points to ".bin" files. 
(     <input class="form-control" type="file" ref="file" accept=".bin,.bin.gz" @change="uploadOTA" />   ) 
In the readme file, it says choose the ".elf " file. 
Shouldn't it be     accept=".elf"     ?

madmartin, 2022-09-03:
@FWallenwein I am sorry that I made a mistake in my pull request #116 . Please pick the *.bin file !!!!
PR is fixed in the meantime.

gd35, 2022-09-03:
I did not questioned myself about the file type... Thanks to have pointed this. Now the upgrade ran fine !

tbnobody, 2022-09-04:
> @tbnobody It will be fixed with https://github.com/tbnobody/OpenDTU/pull/116.

Thank you, just merged it.

closed 2022-09-04 by tbnobody