tbnobody/OpenDTU #120: HM-350

gd35, 2022-09-03:
As I got recently my solar panel equipped with the HM-350. I uploaded OpenDTU to the ESP32. I added the inverter with its id (I checked it twice). But I only received values with zero everywhere. Does it mean that the communication failed between the the HM-350 and the ESP32 ?

madmartin, 2022-09-03:
Hi, I have also had the experience that it did not work right away and can give the following tips:
- try to reduce the distance between the openDTU and the inverter (e.g. move it to the window towards the roof)
- under Settings -> DTU Settings you can adjust the transmit power "PA level". Default is minimum. For me it works with "high" when I sit with the board at the desk and the inverter is in about 10m on the roof.
- the NRF24L01 needs relatively much current. If the power supply is bad (and especially bad cables!) a 10uF capacitor soldered directly to the NRF24L01 brings more stability (pin 1+2 are the power supply). Pay attention to the polarity of the capacitor....

I keep my fingers crossed!
Martin

Translated with www.DeepL.com/Translator (free version)

gd35, 2022-09-03:
Thanks for the tips. I already tested the first one as the DTU is very close to the solar panel (3m). For the second tip, I used High but nothing has changed. I will have thus to test the third tip but I need to get the capacitor.

But may be I made a mistake in the configuration of the ESP32. It is different from the one shown in the OpenDTU Doc:

[image]

[](https://joy-it.net/en/products/SBC-NodeMCU-ESP32
)
I use the following flags to build the firmware:
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=2
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5

I am not sure about the PIN_IRQ. It is connected to D2.

tbnobody, 2022-09-04:
The output of the serial console would also be helpfull.

madmartin, 2022-09-04:
> I will have thus to test the third tip but I need to get the capacitor.

I de-soldered some capacitors from an old motherboard. There were plenty of capacitors on it.

gd35, 2022-09-05:
> The output of the serial console would also be helpfull.

Herewith is the output of the serial console:

```
19:07:23.534 > rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
19:07:23.534 > configsip: 0, SPIWP:0xee
19:07:23.534 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:07:23.534 > mode:DIO, clock div:2
19:07:23.534 > load:0x3fff0030,len:1184
19:07:23.534 > load:0x40078000,len:13132
19:07:23.534 > load:0x40080400,len:3036
19:07:23.534 > entry 0x400805e4
19:07:23.534 > 
19:07:23.534 > Starting OpenDTU
19:07:23.534 > Initialize FS... done
19:07:23.534 > Reading configuration... done
19:07:23.534 > Initialize Network... done
19:07:23.534 > Setting Hostname... Configuring WiFi STA using new credentials... done
19:07:23.534 > Initialize NTP... done
19:07:23.534 > Initialize MqTT... done
19:07:23.534 > Initialize WebApi... done
19:07:23.534 > Initialize Hoymiles interface... Connection successfull
19:07:23.534 >   Setting radio PA level... 
19:07:23.534 >   Setting DTU serial... 
19:07:23.534 >   Setting poll interval... 
19:07:23.534 >   Adding inverter: 112182053057 - SUNOLOGY done
19:07:23.534 > done
19:07:23.534 > Switch to WiFi mode
19:07:23.534 > Setting Hostname... done
19:07:23.534 > Configuring WiFi STA using existing credentials... E (934) wifi:sta is connecting, return error
19:07:23.534 > [   280][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
19:07:23.534 > done
19:07:23.534 > WiFi connected
19:07:23.534 > WiFi got ip: 192.168.1.20
19:07:23.534 > Network connected
19:07:26.278 > Fetch inverter: 112182053057
19:07:41.299 > Fetch inverter: 112182053057
19:07:56.319 > Fetch inverter: 112182053057
19:08:11.339 > Fetch inverter: 112182053057
```

There is a "connect failed" but I can access the device using my web browser without any problem. I did the test very close to the solar panel (less than 1m).

Thanks for your help

tbnobody, 2022-09-05:
hm ok it seems that you dont have any internet access using this wifi. Problem is, that the ESP requires access to a working NTP server to get the current time. The time is included in every data package sent to the inverter.

gd35, 2022-09-05:
You are right 200% ;-) I changes the NTP server address by putting its IP number instead of its symbolic name and ...

[image]

Thanks a lot for having solved the problem !

tbnobody, 2022-09-05:
Nice to hear that its working now :)
I am going to close this issue. I also added a hint regarding the NTP server to the README.md file

closed 2022-09-05 by tbnobody