tbnobody/OpenDTU #121: Event ID 36

horfic, 2022-09-04:
Servus,
ich hab im Event Log ID 36 mit Message = Unknown.
Sonst ist nur der Inverter start mit ID 1 drin.

[image]

Wie weiß man was das wirklich wäre bzw wie kann ich helfen es herauszufinden.

lg,
Horst

tbnobody, 2022-09-04:
Habe Event ID 36 auch schon öfter in meinem Event Log gesehen. Habe schon diverse Datenblätter usw. durchgesehen aber bisher nichts weiter gefunden

crapp, 2022-09-07:
Ich hatte heute auch noch Event 12, gibt es dazu weitere Infos?

tbnobody, 2022-09-07:
Für alle Events die als "Unknown" markiert werden kenne zumindest ich keine Bedeutung. Falls hier jemand Infos hatte nehme ich aber gerne eine Event Beschreibung auf :)

crapp, 2022-09-07:
Die Frage wäre ja wie wir an die Info kommen könnten. In dem Thread auf Mikrocontroller wurde ja das ursprüngliche Reverse Engineering des Protokolls besprochen. Dabei haben ja Leute geholfen die den original DTU haben. Gibt es da Kontakte? Wäre es denn möglich den openDTU und original DTU paralell zu betreiben?

stefan123t, 2022-09-19:
Die HM Fehlercodes gehen von 121..308 einige der bekannten Fehlercodes stammen noch von den MI Wechselrichtern aber auch da ist mW nichts mit 36 und 12 bekannt.
Man könnte evtl einen Dummy Wechselrichter bauen/simulieren der auf die bekannten DTU Anfeagen die entsprechenden WR Antworten liefert und dabei auch die Events 12 und 36 zurückgeben. Dann könnte man in einer DTU lite/Pro die Bedeutung ablesen, das ist aber mE etwas aufwändig. Die Übersetzung findet vermutlich erst in der S-Miles Cloud statt.

stefan123t, 2022-10-07:
@De-Ichirou wäre das etwas für Dich zum Reverse-Engineeren ?

De-Ichirou, 2022-10-09:
In der CLoud sehe ich auf meinem MI keine Fehlercodes, nur den Fehler an sich. Der HM bringt aktuell keine Fehler, die über PV Über-/Unterspannung oder Netzausfall hinausgehen. Wobei selbst die aktuell nicht kommen. 
Für Reverse können wir gerne mal schauen, aber damit die Daten in der Cloud landen, ist noch ein bissel mehr nötig.

joschtl, 2023-01-30:
Vielleicht hilft's…
ich hatte in den letzten Stunden mehrfach kurze Stromausfälle – und denen ging immer ein 14 und/oder ein 36 voraus:

Start | Stop | ID | Message
-- | -- | -- | --
08:25:11 | 08:25:11 | 1 | Inverter start
11:27:58 | 11:27:58 | 14 | Unknown
11:27:58 | 11:29:14 | 143 | Grid: Grid undervoltage
11:27:59 | 11:29:14 | 145 | Grid: Grid underfrequency
11:51:49 | 11:51:49 | 36 | Unknown
11:51:49 | 11:53:16 | 143 | Grid: Grid undervoltage
11:51:50 | 11:53:16 | 145 | Grid: Grid underfrequency
12:14:50 | 12:14:50 | 14 | Unknown
12:14:50 | 12:16:03 | 143 | Grid: Grid undervoltage
12:14:51 | 12:16:03 | 145 | Grid: Grid underfrequency
12:16:03 | 12:16:03 | 36 | Unknown
12:22:54 | 12:22:54 | 14 | Unknown
12:22:54 | 12:24:08 | 143 | Grid: Grid undervoltage
12:22:54 | 12:24:08 | 145 | Grid: Grid underfrequency

Ein 36 scheint "überlebbar" zu sein, aber nach 14 ist immer ein Ausfall gekommen.

stefan123t, 2023-02-03:
Es gibt im Power Profile diverse Angaben wie lange ein bestimmter Zustand anhalten kann / darf. Vielleicht ist das eine Meldung dass eines der Limits (Power/Freq, etc)  in einen Timeout gelaufen ist ?
Siehe https://github.com/lumapu/ahoy/wiki/Protocol#wie-wird-das-down_pro-0x0e--down_dat-0x0a-verwendet

stefan123t, 2023-02-03:
@tbnobody Du hast Doch auch schon ein paar custom / unbekannte Alarm IDs in OpenDTU. Kennst Du 14 und 36 schon ?

tbnobody, 2023-02-03:
Nein, ist noch nichts bekannt.

stefan123t, 2023-02-03:
Interessant wäre evtl auch hierfür dieses Feature das mit einer DTU im Simulations Modus einen Dummy Wechselrichter simuliert. D.h. die DTU antwortet dann wie ein normaler Wechselrichter und man könnte beliebige Werte in der von der DTU zurückgelieferten Payload an zB eine Original Hoymiles DTU zurückgeben.
Da würde man dann einfach mal die ganzen Events angeben und könnte sich die Original Alarm IDs und Code Beschreibung in der SMiles Cloud ausgeben lassen.

https://github.com/lumapu/ahoy/issues/612

stefan123t, 2023-02-16:
@tbnobody heute hat mein Shelly 1pm+ der am HM-600 hängt rumgesponnen.
Er hat von 5:32 morgens bis ca. 8:55 Over voltage detected Fehler geworfen.
Nach einem Firmware Update habe ich auch mal ein wenig damit rumgespielt
und den Inverter mit dem Relais im Shelly vom AC Anschluß getrennt.

Es kamen die üblichen o.g. Fehler
* 136 Grid: Grid undervoltage
* 147 Grid: Power grid outage
* 145 Grid: Grid underfrequency
und dann kam noch
* 13 Unknown

Ich nehme daher an dies ist entweder der NA-Schutz der in einem solchen Fall (136, 145) einsetzen muß.
Oder ein Ähnliches Event das die drei anderen Events einschließt.

Nachdem ich das Relais wieder durchgeschalten habe, dauerte es eine kurze Weile 5-15 Sekunden bis wieder AC Spannung und Frequenz im OpenDTU Live Fenster gemeldet wurden. Aber erst nach weiteren 30-90 Sekunden wurde auch etwas eingespeist. Mehr oder minder gleichzeitig wurde dann auch das 13 Event gestoppt. Zusätzlich wurde dann das Event 36 geloggt, das m.E. sowas wie WR speist wieder ein bedeutet. In der nächsten Minute hat dann der WR angefangen auch die eingespeiste Leistung wieder anzuheben und sich langsam seinem PowerLimit (100%) bzw. der an den beiden DC Eingängen verfügbaren Leistung genähert.

Das Prozedere sollte sich relativ leicht nachstellen lassen. Nach dem zweiten / dritten Versuch wurden jedoch keine Events mehr mitgeschrieben, ich vermute das liegt daran dass wir bis jetzt nur das AlarmData 0x11 und noch nicht das AlarmUpdate 0x12 Command in OpenDTU implementiert haben. Das benötigen wir m.E. um die Events als gelesen bzw. überschreibbar zu markieren.
Mehr als 15 Events scheint der WR nämlich nicht länger speichern zu können, d.h. er schiebt alte Events mit einer ggf. niedrigeren Priorität (wie z.B. 2 DTU command failed) aus dem Event log, behält sich aber viele der og. Events und kann daher vermutlich auch keine neuen Events mehr ablegen.

[image]
[image]
[image]
[image]

stefan123t, 2023-02-16:
Siehe auch #590

stefan123t, 2023-10-08:
@horfic Danke für die Meldung, ja es gibt noch ein paar Unbekannte / unknown Fehlermeldungen.
U.a. 
* 12 (https://github.com/tbnobody/OpenDTU/issues/1068),
* 14 (https://github.com/tbnobody/OpenDTU/issues/131),
* 12, 13, 14, 36,  (https://github.com/tbnobody/OpenDTU/issues/121),
* 12, 14, 36, 46, 47, 73 https://github.com/tbnobody/OpenDTU/discussions/590

Ich bin in den `assets/warn_code.json` der Hoymiles S-Miles Installer App fündig geworden. 🥳 

Hier mal die bisher aufgetretenen Fehler oder Warnings aus dem angehängten warn_code.json (GitHub versteht nur ZIP ?).

11: Grid voltage surge
12: Grid voltage sharp drop
13: Grid frequency mutation
14: Grid phase mutation
36: INV overvoltage or overcurrent
46: FB overvoltage
47: FB overcurrent
71: VW function enable
72: FW function enable
73: TW function enable

@tbnobody vielleicht wollt Ihr das in den beiden Projekten einbauen, dann kann man die o.g. drei Issues auf OpenDTU und bestimmt auch einige auf Ahoy Seite schließen. Piece of Cake: 🍰 

[warn_code.json.zip](https://github.com/tbnobody/OpenDTU/files/12841298/warn_code.json.zip)

stefan123t, 2023-10-08:
Hier die warn_code.json als Markdown Tabelle:

Event ID | Description
--- | ---
1 | Reset
2 | Time calibration
3 | EEPROM reading and writing error during operation
4 | Offline
11 | Grid voltage surge
12 | Grid voltage sharp drop
13 | Grid frequency mutation
14 | Grid phase mutation
15 | Grid transient fluctuation
36 | INV overvoltage or overcurrent
46 | FB overvoltage
47 | FB overcurrent
48 | FB clamp overvoltage
49 | FB clamp overvoltage
61 | Calibration parameter error
62 | System configuration parameter error
63 | Abnormal power generation data
71 | VW function enable
72 | FW function enable
73 | TW function enable
95 | PV module in suspected shadow
96 | PV module in suspected shadow
97 | PV module in suspected shadow
98 | PV module in suspected shadow
121 | Over temperature protection
122 | Microinverter is suspected of being stolen
123 | Locked by remote control
124 | Shut down by remote control
125 | Grid configuration parameter error
126 | EEPROM reading and writing error
127 | Firmware error
128 | Hardware configuration error
129 | Abnormal bias
130 | Offline
141 | Grid overvoltage
142 | 10 min value grid overvoltage
143 | Grid undervoltage
144 | Grid overfrequency
145 | Grid underfrequency
146 | Rapid grid frequency change rate
147 | Power grid outage
148 | Grid disconnection
149 | Island detected
150 | DCI exceeded
181 | Abnormal insulation impedance
182 | Abnormal grounding
205 | Input overvoltage
206 | Input overvoltage
207 | Input undervoltage
208 | Input undervoltage
209 | PV1 No input
210 | PV2 No input
211 | PV3 No input
212 | PV4 No input
213 | PV-1&PV-2 abnormal wiring
214 | PV-3&PV-4 abnormal wiring
301 | FB short circuit failure
302 | FB short circuit failure
303 | FB overcurrent protection failure
304 | FB overcurrent protection failure
305 | FB clamp circuit failure
306 | FB clamp circuit failure
307 | INV power device failure
308 | INV overcurrent or overvoltage protection failure
1111 | Repeater
2000 | Standby
2001 | Standby
2002 | Standby
2003 | Standby
2004 | Standby
3001 | Reset
3002 | Reset
3003 | Reset
3004 | Reset
5011 | PV1 Error code-01
5012 | PV2 Error code-01
5013 | PV3 Error code-01
5014 | PV4 Error code-01
5020 | Error code-02
5041 | PV1 Error code-04
5042 | PV2 Error code-04
5043 | PV3 Error code-04
5044 | PV4 Error code-04
5051 | PV1 Input Overvoltage/Undervoltage
5052 | PV2 Input Overvoltage/Undervoltage
5053 | PV3 Input Overvoltage/Undervoltage
5054 | PV4 Input Overvoltage/Undervoltage
5060 | Abnormal bias
5070 | Over temperature protection
5080 | Grid Overvoltage/Undervoltage
5090 | Grid Overfrequency/underfrequency
5100 | Island detected
5110 | GFDI
5120 | EEPROM reading and writing error
5141 | FB clamp overvoltage
5142 | FB clamp overvoltage
5143 | FB clamp overvoltage
5144 | FB clamp overvoltage
5150 | 10 min value grid overvoltage
5160 | Grid transient fluctuation
5200 | Firmware error
5511 | Error code-01
5512 | Error code-01
5513 | Error code-01
5514 | Error code-01
5520 | Error code-02
8310 | Shut down by remote control
8320 | Locked by remote control
9000 | Microinverter is suspected of being stolen

tbnobody, 2023-10-09:
Ggf hilft das aus der spec. Vom grid profile 

[image]

tbnobody, 2023-10-09:
Akutell habe ich  auch sowas drin stehen:
`{ AlarmMessageType_t::ALL, 73, "Temperature >80°C" }, // https://github.com/tbnobody/OpenDTU/discussions/590#discussioncomment-6049750`

TW = Temperature Warning?

tbnobody, 2023-10-09:
Auch interessant finde ich ID 2... Das ist bisher ja das "DTU Command Failed"... Jetzt eher ein Harmloses "Time calibration"

stefan123t, 2023-10-09:
> Akutell habe ich auch sowas drin stehen: `{ AlarmMessageType_t::ALL, 73, "Temperature >80°C" }, // https://github.com/tbnobody/OpenDTU/discussions/590#discussioncomment-6049750`
> 
> TW = Temperature Warning?

Ja ich bin auch auf VW, FW, TW -> Voltage, Frequency oder Temperature WatchDog gekommen. Warning geht aber sicher auch.
FB steht eventuell für FlyBack Transformer.

Siehe unsere Diskussion in Ahoy mit Lenni und homeautomation von gestern:
https://github.com/tbnobody/OpenDTU/discussions/590

stefan123t, 2023-10-09:
> Ggf hilft das aus der spec. Vom grid profile. Das sind verschiedene Software Module welche man im Grid Profile aktivieren kann.
> 
> [image]

Ja danke was das Grid Profile angeht habe ich bereits diese "Topcis" hier dokumentiert:
https://github.com/lumapu/ahoy/wiki/Protocol#file-name-de_vde4105_2018

Ich habe auch von @lumapu und Flole bereits einen Grid Profile Response bekommen und ansatzweise ausgewertet.
Das kommt sogar einigermaßen hin mit den Topics. 
Ich bin nur noch nicht ganz sicher wie die Topic Überschriften (fett & gelb) zu interpretieren sind. 
Mit Flole hatte ich ein Argument es könnten evtl. Offsets im Config-Speicher des TMS320F28034 von Texas Instruments sein, vielleicht sind es aber auch irgendwelche Binary IDs.

[DE_VDE4105_2018_v2.0.0_20230924.xlsx](https://github.com/tbnobody/OpenDTU/files/12850517/DE_VDE4105_2018_v2.0.0_20230924.xlsx)

stefan123t, 2023-11-22:
@tbnobody ich glaube das Issue mit Event ID 36 ist damit gelöst und kann geschlossen werden.

closed 2023-11-22 by tbnobody