tbnobody/OpenDTU #122: Daten visualisieren

daja64, 2022-09-04:
Kann mir mal jemand erklären, wie die Daten auf einem Handy zeigen kann. So mit Diagramm oder so, also was der WR so den ganzen Tag gemacht hat. Diese MQTT Sache kapier ich nicht. Und was ich im Netz gefunden habe, hat nicht wirklich geholfen. Hilfe ist von Nöten. Danke

ahinrichs, 2022-09-05:
Der ESP speichert - aktuell und auf absehbar - keine Datenreihen, sondern kennt nur den aktuellen Ist-Zustand, also Leistung jetzt gerade und Spannung jetzt gerade. Diese Werte kannst Du dir in der Weboberfläche angucken. Für Diagramme musst Du die Daten extern sammeln. Also auf irgendeinem Server die Daten regelmäßig per HTTP abholen (oder eben per MQTT dorthin verschicken lassen) und dort in eine Datenbank oder ähnliches schreiben. Und von dort mit einer Weboberfläche Deiner Wahl visualisieren. Viele haben schon eine Steuerzentrale, die das kann. Sonst wird oft die Kombi Influx und Grafana genutzt. Aber wenn Du Dich mit MQTT schwer tutst, wird das nicht ganz leicht.

Das ist bei Hoymiles ja nicht anders. Die Orginal-DTU sammelt die Daten und schickt sie an die Cloud. Charts gibt es dann nur von dort. Vielleicht findest Du einen anderen Cloud-Dienst, in den Du Deine Daten reinfüttern kannst.

ottelo9, 2022-09-06:
Mhhh könnte man das Projekt irgendwie mit Tasmota verbinden dann könnte man die Daten sogar via deren Script Support mit Google Chart visualisieren. So habe ich es jedenfalls bei mir mit Tasmota und meinem Stromzähler gemacht. https://ottelo.jimdofree.com/stromz%C3%A4hler-auslesen/

battifreund, 2022-09-06:
Ich mag Tasmota nicht so richtig. Es ist mittlerweile ein ganz schöner Verhau. 

Ich finde ESPHome wesentlich strukturierter und durch die Anbindung an HomeAssistant / ioBroker können die Daten dort sehr gut visualisiert oder langfristig gespeichert werden.

Wenn, dann sollte in eine ESPHome-Komponente investiert werden. Ich weiß allerdings nicht, ob das aufgrund der Architektur von ESPHome immer problemlos funktioniert.

----

I don't really like Tasmota. It's quite a mess now.

I find ESPHome much more structured and through the connection to HomeAssistant / ioBroker the data can be visualized very well there or stored long-term.

However, I do not know whether this always works without problems due to the architecture of ESPHome.

ottelo9, 2022-09-06:
Ja klar gibt es Alternativen. Es war auch nur ein Wunsch für eine Option, die man ja z.b. extra aktivieren muss um sich ein kleines Diagramm anzeigen zu lassen.

Ich finde aber so Einzellösungen wie Tasmita + Google Chart ganz gut, wenn man keine Hausautomatisierung ala Home Assistant und Co hat. Wie bei meinen Eltern z.B.. Mein Vater wollte eigentlich nur den Stromzähler bequem via Handy auslesen. Und da kommt ein einzelner Lesekopf inkl. ESP + Tasmota + Webseite mit Google Chart genau richtig. Wenn es nicht mehr sein soll ist doch solch eine einfache Lösung ideal. Für alles weitere hast du natürlich Recht, ESPHome + HA.

c1328, 2022-09-09:
Wer doch die MQTT Schiene geht und außerdem noch Telegraf/Grafana nutzen möchte, für den gibt es hier vielleicht ein interessantes Dashboard (für Ahoy gebaut, aber kann man sicher auch für openDTU nutzbar machen): https://grafana.com/grafana/dashboards/16850-pv-power-ahoy/

HacksBugsAndRockAndRoll, 2022-09-24:
Ich habe die Daten in meinem Setup über einen [MQTT Broker](https://mosquitto.org/) in [Home Assistant](https://www.home-assistant.io/) eingebunden. OpenDTU unterstützt das Einbinden in HomeAssistant hier ziemlich gut, so dass man die einzelnen MQTT topics nicht manuell interpretieren muss. Visualisierung findet dann über die HomeAssistant dashboards statt, die Handy App gibts dann umsonst dazu.
Um so ein setup aufzubauen hilft es, wenn ein SoC (wie ein raspberry >= 3), eine NAS mit Docker Support (mein setup) oder ein anderer home server zur Verfügung steht.