tbnobody/OpenDTU #128: Bootet nicht

mpl1337, 2022-09-06:
Hallo,

habe eben die Firmware auf den ESP32 https://www.az-delivery.de/products/esp32-developmentboard übertragen, jedoch startet es nicht.

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r>
<\n>flash read err, 1000<\r>
<\n>ets_main.c 371 <\r>
<\n>ets Jun  8 2016 00:22:57`

ESPHOME hingegen startet nach dem Flashen ohne Fehler.

was könnte da das Problem sein?

tbnobody, 2022-09-06:
Hast du versucht den Flashinhalt komplett zu löschen? "Erase Flash"

mpl1337, 2022-09-06:
ja habe ich - "yes, wipes all data"

tbnobody, 2022-09-06:
Könntest du bitte die Ausgabe des Terminal Fensters posten in dem der Upload stattfindet? Die Meldung `flash read err, 1000` scheint zu bedeuten das der Bootloader an Adresse 0x1000 nichts sinnvolles findet. Ein weiterer Versuch wäre es, das NRF Modul erst einmal nicht anzuschließen um eine falsche Verkabelung auszuschließen.

tbnobody, 2022-09-19:
Für den Fall, dass du die fertigen .bin Dateien verwendet hast. Die ZIP's enthalten jetzt 4 Dateien. Die jeweiligen Speicheradressen sind in  der README.md dokumentiert.

mpl1337, 2022-10-06:
Hab's hinbekommen. Hab direkt per Vscode geflasht.