tbnobody/OpenDTU #129: Live data disappeared

gd35, 2022-09-11:
Today, I am not able to display the live data. However, data are still transmitted via MQTT. 

[image]

I will reboot the device this evening to see if live data can be displayed again for the next day.

Anyway, I am a happy user of OpenDTU. You did a very nice piece of software/hardware !

tbnobody, 2022-09-11:
I am also facing this problem sometimes. There seems to be a issue with the WebSocket implementation of the ESPAsyncWebserver library. Are you able to see the output of the serial console without rebooting the ESP? Or can you maybe take a screenshot of the Info --> System view? The most interessting part would be the Memory Information and the Git Hash.

gd35, 2022-09-11:
The device is connected to my Synology NAS through a USB hub. I am not sure I can connect to the serial console with the NAS. Herewith is the screenshot:

[image]

Thanks for your help

gd35, 2022-09-12:
This morning Live Data appeared again. Without doing anything.

tbnobody, 2022-09-12:
Could you please double check what the Heap usage is now displaying? I would guess a value in the range of 110-120kbyte (instead of the 170).

gd35, 2022-09-13:
This morning, the heap usage is:

[image]

tbnobody, 2022-09-14:
@gd35 what browser are you using? A Chromium based on? (Chrome, Vivaldi etc?)

gd35, 2022-09-15:
I use Firefox on a Mac. If it will happen again I will try another web browser. So far, it works.

S-Przybylski, 2022-09-16:
From time to time i have the same issue. I use Android browser (samsung and chrome) and on PC Firefox.

tbnobody, 2022-09-16:
I've changed the WebServer yesterday (using another fork). Let's see if this fixes the WebSocket issue.

S-Przybylski, 2022-09-16:
I just tried to compile the todays version and got an error. Any hint?

````
Compiling .pio\build\d1 mini esp32\src\main.cpp.o
src/WebApi_ws_live.cpp: In member function 'void WebApiWsLiveClass::loop()':
src/WebApi_ws_live.cpp:60:9: error: 'AsyncWebSocketMessageBuffer' was not declared in this scope
         AsyncWebSocketMessageBuffer* buffer = _ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/WebApi_ws_live.cpp:60:9: note: suggested alternative: 'AsyncWebSocketMessage'
         AsyncWebSocketMessageBuffer* buffer = _ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
         AsyncWebSocketMessage
src/WebApi_ws_live.cpp:60:38: error: 'buffer' was not declared in this scope
         AsyncWebSocketMessageBuffer* buffer = _ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
                                      ^~~~~~
src/WebApi_ws_live.cpp:60:38: note: suggested alternative: 'setbuffer'
         AsyncWebSocketMessageBuffer* buffer = _ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
                                      ^~~~~~
                                      setbuffer
src/WebApi_ws_live.cpp:60:51: error: 'class AsyncWebSocket' has no member named 'makeBuffer'
         AsyncWebSocketMessageBuffer* buffer = _ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
                                                   ^~~~~~~~~~
*** [.pio\build\d1 mini esp32\src\WebApi_ws_live.cpp.o] Error 1
````

S-Przybylski, 2022-09-16:
Sorry my fault - again cloned / now works (Danke).

tbnobody, 2022-09-16:
(Hint: `git pull` should load the current version of the code without the need of a complete clone)

tbnobody, 2022-09-19:
I am closing this issue  as (at least for me) this issue didn't occour any more... feel free to open it again

closed 2022-09-19 by tbnobody