tbnobody/OpenDTU #131: new Events not viewable

locke981, 2022-09-12:
I have two HM-1500 (West, Ost) connected to the OpenDTU. I can see the red numbers for both of the inverters in case of new events.
If the numbers on the Ost (which is implemented later as the 2nd inverter) exceed a certain number of events they are not listed when trying to open them, it's just an empty list. On the first inverter (West) this always works flawless. It just happens on the 2nd inverter. If there are only small number of events it also works on the 2nd inverter without any problem. I do not exactly know what is the breaking number. I must be someting greater than 3 or so.
Stefan

[image]

tbnobody, 2022-09-12:
Could you please open the URL: `http://<OpenDTU IP>/api/eventlog/status` in a browser and post the output? I assume that maybe the buffer for the json output is to small and therefor the entries get cut off.

locke981, 2022-09-12:
{"116180204736":{"count":13,"events":[{"message_id":1,"message":"Inverter start","start_time":22526,"end_time":22526},{"message_id":12,"message":"Unknown","start_time":27684,"end_time":27684},{"message_id":14,"message":"Unknown","start_time":27813,"end_time":27813},{"message_id":148,"message":"Grid disconnection","start_time":27814,"end_time":27900},{"message_id":12,"message":"Unknown","start_time":28024,"end_time":28024},{"message_id":12,"message":"Unknown","start_time":28116,"end_time":28116},{"message_id":12,"message":"Unknown","start_time":28147,"end_time":28147},{"message_id":12,"message":"Unknown","start_time":29115,"end_time":29115},{"message_id":12,"message":"Unknown","start_time":29279,"end_time":29279},{"message_id":12,"message":"Unknown","start_time":29403,"end_time":29403},{"message_id":12,"message":"Unknown"}]}}

tbnobody, 2022-09-12:
If you have already compiled it yourself and if you want to test it you can try to replace the line:
https://github.com/tbnobody/OpenDTU/blob/85d08e0aba937860c1eb96808cab5fd439228244/src/WebApi_eventlog.cpp#L25
by the following:
`AsyncJsonResponse* response = new AsyncJsonResponse(false, 2048); `

This is only a workaround as for more (up to 10 supported inverters) inverters the 2048 will be also to small. I think I have to rework the API a bit.

locke981, 2022-09-12:
Ok, I will test that and update here, thanks for the great support and your work!

locke981, 2022-09-12:
Looks good now....I will post again here if there are any issues.
[image]

closed 2022-09-12 by tbnobody

stefan123t, 2023-10-08:
@locke981 Danke für die Meldung, ja es gibt noch ein paar Unbekannte / unknown Fehlermeldungen.
U.a. 
* 12 (https://github.com/tbnobody/OpenDTU/issues/1068),
* 14 (https://github.com/tbnobody/OpenDTU/issues/131),
* 12, 13, 14, 36,  (https://github.com/tbnobody/OpenDTU/issues/121),
* 12, 14, 36, 46, 47, 73 https://github.com/tbnobody/OpenDTU/discussions/590

Ich bin in den `assets/warn_code.json` der Hoymiles S-Miles Installer App fündig geworden. 🥳 

Hier mal die bisher aufgetretenen Fehler oder Warnings aus dem in #121 angehängten warn_code.json (GitHub versteht nur ZIP ?).

11: Grid voltage surge
12: Grid voltage sharp drop
13: Grid frequency mutation
14: Grid phase mutation
36: INV overvoltage or overcurrent
46: FB overvoltage
47: FB overcurrent
71: VW function enable
72: FW function enable
73: TW function enable

Es wird von @tbnobody im Rahmen von #121 in OpenDTU implementiert und daher kann dieses Issue als Duplicate geschlossen werden.