tbnobody/OpenDTU #144: Kein Live Data über AP

andreash-esp, 2022-09-17:
Hi,

Live Data wird nicht angezeigt, wenn man initial über den AP verbunden ist. 
Wurde mal ein Wifi angelegt, funktioniert es auch bei Verbindung über den AP.

Andreas

tbnobody, 2022-09-17:
Das ist ganz logisch. Ohne eingetragenes wlan kein Zugriff zu einem NTP Server. Ohne Zugriff zu einem NTP Server keine Uhrzeit. Ohne Uhrzeit keine Anfragen an den wechselrichter da in jedem Paket die Uhrzeit enthalten ist.

closed 2022-09-17 by tbnobody

lumapu, 2022-09-18:
@tbnobody du kannst dich für diesen Betriebsmodus einfach eine 'Fake Zeit' nehmen, zB die des Firmware complies.
In der Ahoy Version wird aktuell einfach 1970 verwendet

stefan123t, 2022-09-19:
Ich fände für diesen Fall AP-only mode eine Option zum manuellen setzen von Datum & Zeit sinnvoll.