tbnobody/OpenDTU #163: TSUN TSOL M1600 Support

tobox, 2022-09-26:
Weiß jemand, ob der TSUN TSOL M1600 unterstützt wird? Scheint von der "Baureihe" her sehr ähnlich zu sein wie der M800 bzw. die noch kleineren...

tbnobody, 2022-09-26:
Beginnt deine Seriennummer mit 11? Das wäre ein gutes Anzeichen das es klappen könnte. Ich schreibe "könnte" weil es wohl verschiedene Firmware Versionen gibt. Die Einen funktionieren, die Anderen nicht.

tobox, 2022-09-27:
Ja, Seriennummer beginnt mit 11617... Bin gespannt. WR ist da, ESP32 auch, Funkplatine im Zulauf.

tbnobody, 2022-09-28:
Würde mich dann über eine Rückkmeldung freuen ob es klappt oder nicht. Dann könnte ich den Inverter auch noch in die Readme.md aufnehmen.

tobox, 2022-09-28:
Ich habe den Wechselrichter mal temporär mit AC versorgt ohne das Solarmodule dran sind. Die OpenDTU habe ich auch nur mit Jumperkabeln zusammengesteckt. Aber ich befürchte fast, es funktioniert - oder was meint ihr? Ich habe nur die Seriennummer eingegeben, die Anzahl der Module hat er dadurch scheinbar schon selber rausgefunden.

[image]

tobox, 2022-09-28:
Hier noch ein paar Nicht-Infos:
[image]

tbnobody, 2022-09-28:
Hm ne. Also die Anzahl der Module bekommt er über die Seriennummer raus. Nur mit AC aktiviert den Inverter nicht. Dieser wird NUR von der DC Seite her mit Strom versorgt. Also ohne Solarmodule ist noch alles offen :-)

tobox, 2022-09-28:
Mist... Dann muss ich den Aufbau wohl morgen Abend mal nach draußen verlegen... Sollte ein PV-Modul reichen? Die stehen nämlich noch im Schuppen...

tbnobody, 2022-09-28:
Mit einem Modul sollte er schon aktiv werden ja. Ich bin mir jetzt aber nicht 100% sicher ob man die AC Seite benötigt.

tobox, 2022-09-29:
Kannst einen Haken dran machen! Läuft! Vielen Dank für das tolle Projekt!

[image]

tobox, 2022-09-29:
Falls das hier noch hilft:

[image]

closed 2022-09-29 by tobox