tbnobody/OpenDTU #174: WebUI doesn't show phase voltage anymore

MFV, 2022-10-02:
Based on a firmware build on the commit [2871be3](https://github.com/tbnobody/OpenDTU/commit/2871be36cfc5f392d4f7debebf2e527d0cc013bb) the phase voltage is't shown anymore in the WebUI. 
Line only shows Voltage NaN V.

The value is still correctly reporting on the MQTT topic (solar/xyz/0/voltage).

[image]

tbnobody, 2022-10-02:
How many inverters do you monitor with the ESP? And what is the output of `http://[ip]/api/livedata/status`

MFV, 2022-10-02:
Two inverters.
Output of the API doesn't show for both inverts the phase voltage:
`[
  {
    "serial" : "xx1",
    "events" : 3,
    "data_age" : 3,
    "0" : {
      "YieldTotal" : {
        "v" : 20.8390007,
        "u" : "kWh"
      },
      "Temperature" : {
        "v" : 37.400001529999997,
        "u" : "Â°C"
      },
      "ReactivePower" : {
        "v" : 0,
        "u" : "var"
      },
      "Efficiency" : {
        "v" : 95.481826780000006,
        "u" : "%"
      },
      "Frequency" : {
        "v" : 50.009998320000001,
        "u" : "Hz"
      },
      "PowerFactor" : {
        "v" : 100,
        "u" : "%"
      },
      "Current" : {
        "v" : 1.1299999949999999,
        "u" : "A"
      },
      "YieldDay" : {
        "v" : 785,
        "u" : "Wh"
      },
      "Power DC" : {
        "v" : 283.2999878,
        "u" : "W"
      },
      "Power" : {
        "v" : 270.5,
        "u" : "W"
      },
      "Voltage" : {
        "v" : null,
        "u" : "V"
      }
    },
    "1" : {
      "Irradiation" : {
        "v" : 77.616432189999998,
        "u" : "%"
      },
      "YieldTotal" : {
        "v" : 20.8390007,
        "u" : "kWh"
      },
      "Power" : {
        "v" : 283.2999878,
        "u" : "W"
      },
      "Current" : {
        "v" : 8.6599998469999999,
        "u" : "A"
      },
      "Voltage" : {
        "v" : 32.799999239999998,
        "u" : "V"
      },
      "YieldDay" : {
        "v" : 785,
        "u" : "Wh"
      }
    },
    "reachable" : true,
    "producing" : true,
    "name" : "BVK-BKW-365"
  },
  {
    "serial" : "xx2",
    "events" : 3,
    "data_age" : 0,
    "0" : {
      "YieldTotal" : {
        "v" : 4.6620001789999996,
        "u" : "kWh"
      },
      "Temperature" : {
        "v" : 33.900001529999997,
        "u" : "Â°C"
      },
      "ReactivePower" : {
        "v" : 0.10000000100000001,
        "u" : "var"
      },
      "Efficiency" : {
        "v" : 95.445549009999993,
        "u" : "%"
      },
      "Frequency" : {
        "v" : 50.009998320000001,
        "u" : "Hz"
      },
      "PowerFactor" : {
        "v" : 100,
        "u" : "%"
      },
      "Current" : {
        "v" : 0.61000001400000003,
        "u" : "A"
      },
      "YieldDay" : {
        "v" : 582,
        "u" : "Wh"
      },
      "Power DC" : {
        "v" : 151.5,
        "u" : "W"
      },
      "Power" : {
        "v" : 144.6000061,
        "u" : "W"
      },
      "Voltage" : {
        "v" : null,
        "u" : "V"
      }
    },
    "1" : {
      "Irradiation" : {
        "v" : 40.945945739999999,
        "u" : "%"
      },
      "YieldTotal" : {
        "v" : 4.6620001789999996,
        "u" : "kWh"
      },
      "Power" : {
        "v" : 151.5,
        "u" : "W"
      },
      "Current" : {
        "v" : 4.3400001530000001,
        "u" : "A"
      },
      "Voltage" : {
        "v" : 34.900001529999997,
        "u" : "V"
      },
      "YieldDay" : {
        "v" : 582,
        "u" : "Wh"
      }
    },
    "reachable" : true,
    "producing" : true,
    "name" : "BVK-BKW-370"
  }
]`

tbnobody, 2022-10-02:
There is something strange. The order of the web UI should be the same as in the json response. In your json output YieldTotal is the first value while on your screenshot it's somewhere in the middle. Interessting thing is, I didn't change anything regarding the parsing and the byte assignment of the statistics request. I assume you have a inverter serial beginning with 1121xxxxxx? 

How did you request the json output mentioned above? Because also reachable and producing should occour after data_age. In your output it's at the end of the output.

mesingerold, 2022-10-02:
The same issue with 2 working inverters.
Inverter 1 shows Voltage : 0, the second shows Voltage right - both transmit to Mqtt correctly!!

[{"serial“:“xxx“,“name":"HM400","data_age":4,"reachable":true,"producing":true,"0":{"Power":{"v":35.79999924,"u":"W"},"Voltage":{"v":null,"u":"V"},"Current":{"v":0.150000006,"u":"A"},"Power DC":{"v":37.59999847,"u":"W"},"YieldDay":{"v":299,"u":"Wh"},"YieldTotal":{"v":7.252999783,"u":"kWh"},"Frequency":{"v":49.97999954,"u":"Hz"},"Temperature":{"v":20.29999924,"u":"Â°C"},"PowerFactor":{"v":100,"u":"%"},"ReactivePower":{"v":0,"u":"var"},"Efficiency":{"v":95.21276855,"u":"%"}},"1":{"Power":{"v":37.59999847,"u":"W"},"Voltage":{"v":34,"u":"V"},"Current":{"v":1.110000014,"u":"A"},"YieldDay":{"v":299,"u":"Wh"},"YieldTotal":{"v":7.252999783,"u":"kWh"},"Irradiation":{"v":9.766233444,"u":"%"}},"events":3},{"serial“:“xxx“,“name":"HM600","data_age":9,"reachable":true,"producing":true,"0":{"Power":{"v":53.40000153,"u":"W"},"Voltage":{"v":234.1999969,"u":"V"},"Current":{"v":0.230000004,"u":"A"},"Power DC":{"v":56,"u":"W"},"YieldDay":{"v":449,"u":"Wh"},"YieldTotal":{"v":63.83499908,"u":"kWh"},"Frequency":{"v":49.99000168,"u":"Hz"},"Temperature":{"v":25.10000038,"u":"Â°C"},"PowerFactor":{"v":100,"u":"%"},"ReactivePower":{"v":0,"u":"var"},"Efficiency":{"v":95.35714722,"u":"%"}},"1":{"Power":{"v":28.89999962,"u":"W"},"Voltage":{"v":31.89999962,"u":"V"},"Current":{"v":0.910000026,"u":"A"},"YieldDay":{"v":232,"u":"Wh"},"YieldTotal":{"v":33.32799911,"u":"kWh"},"Irradiation":{"v":9.633333206,"u":"%"}},"2":{"Power":{"v":27.10000038,"u":"W"},"Voltage":{"v":30.79999924,"u":"V"},"Current":{"v":0.879999995,"u":"A"},"YieldDay":{"v":217,"u":"Wh"},"YieldTotal":{"v":30.50699997,"u":"kWh"},"Irradiation":{"v":9.033333778,"u":"%"}},"events":4}]
[image]

mesingerold, 2022-10-02:
The same issue with 2 working inverters.
Inverter 1 shows Voltage : 0, the second shows Voltage right - both transmit to Mqtt correctly!!

[{"serial“:“xxx“,“name":"HM400","data_age":4,"reachable":true,"producing":true,"0":{"Power":{"v":35.79999924,"u":"W"},"Voltage":{"v":null,"u":"V"},"Current":{"v":0.150000006,"u":"A"},"Power DC":{"v":37.59999847,"u":"W"},"YieldDay":{"v":
[image]
299,"u":"Wh"},"YieldTotal":{"v":7.252999783,"u":"kWh"},"Frequency":{"v":49.97999954,"u":"Hz"},"Temperature":{"v":20.29999924,"u":"Â°C"},"PowerFactor":{"v":100,"u":"%"},"ReactivePower":{"v":0,"u":"var"},"Efficiency":{"v":95.21276855,"u":"%"}},"1":{"Power":{"v":37.59999847,"u":"W"},"Voltage":{"v":34,"u":"V"},"Current":{"v":1.110000014,"u":"A"},"YieldDay":{"v":299,"u":"Wh"},"YieldTotal":{"v":7.252999783,"u":"kWh"},"Irradiation":{"v":9.766233444,"u":"%"}},"events":3},{"serial“:“xxx“,“name":"HM600","data_age":9,"reachable":true,"producing":true,"0":{"Power":{"v":53.40000153,"u":"W"},"Voltage":{"v":234.1999969,"u":"V"},"Current":{"v":0.230000004,"u":"A"},"Power DC":{"v":56,"u":"W"},"YieldDay":{"v":449,"u":"Wh"},"YieldTotal":{"v":63.83499908,"u":"kWh"},"Frequency":{"v":49.99000168,"u":"Hz"},"Temperature":{"v":25.10000038,"u":"Â°C"},"PowerFactor":{"v":100,"u":"%"},"ReactivePower":{"v":0,"u":"var"},"Efficiency":{"v":95.35714722,"u":"%"}},"1":{"Power":{"v":28.89999962,"u":"W"},"Voltage":{"v":31.89999962,"u":"V"},"Current":{"v":0.910000026,"u":"A"},"YieldDay":{"v":232,"u":"Wh"},"YieldTotal":{"v":33.32799911,"u":"kWh"},"Irradiation":{"v":9.633333206,"u":"%"}},"2":{"Power":{"v":27.10000038,"u":"W"},"Voltage":{"v":30.79999924,"u":"V"},"Current":{"v":0.879999995,"u":"A"},"YieldDay":{"v":217,"u":"Wh"},"YieldTotal":{"v":30.50699997,"u":"kWh"},"Irradiation":{"v":9.033333778,"u":"%"}},"events":4}]
[image]

mesingerold, 2022-10-02:
The same issue with 2 working inverters.
Inverter 1 shows Voltage : 0, the second shows Voltage right - both transmit to Mqtt correctly!!

[{"serial“:“xxx“,“name":"HM400","data_age":4,"reachable":true,"producing":true,"0":{"Power":{"v":35.79999924,"u":"W"},"Voltage":{"v":null,"u":"V"},"Current":{"v":0.150000006,"u":"A"},"Power DC":{"v":37.59999847,"u":"W"},"YieldDay":{"v":299,"u":"Wh"},"YieldTotal":{"v":7.252999783,"u":"kWh"},"Frequency":{"v":49.97999954,"u":"Hz"},"Temperature":{"v":20.29999924,"u":"Â°C"},"PowerFactor":{"v":100,"u":"%"},"ReactivePower":{"v":0,"u":"var"},"Efficiency":{"v":95.21276855,"u":"%"}},"1":{"Power":{"v":37.59999847,"u":"W"},"Voltage":{"v":34,"u":"V"},"Current":{"v":1.110000014,"u":"A"},"YieldDay":{"v":299,"u":"Wh"},"YieldTotal":{"v":7.252999783,"u":"kWh"},"Irradiation":{"v":9.766233444,"u":"%"}},"events":3},{"serial“:“xxx“,“name":"HM600","data_age":9,"reachable":true,"producing":true,"0":{"Power":{"v":53.40000153,"u":"W"},"Voltage":{"v":234.1999969,"u":"V"},"Current":{"v":0.230000004,"u":"A"},"Power DC":{"v":56,"u":"W"},"YieldDay":{"v":449,"u":"Wh"},"YieldTotal":{"v":63.83499908,"u":"kWh"},"Frequency":{"v":49.99000168,"u":"Hz"},"Temperature":{"v":25.10000038,"u":"Â°C"},"PowerFactor":{"v":100,"u":"%"},"ReactivePower":{"v":0,"u":"var"},"Efficiency":{"v":95.35714722,"u":"%"}},"1":{"Power":{"v":28.89999962,"u":"W"},"Voltage":{"v":31.89999962,"u":"V"},"Current":{"v":0.910000026,"u":"A"},"YieldDay":{"v":232,"u":"Wh"},"YieldTotal":{"v":33.32799911,"u":"kWh"},"Irradiation":{"v":9.633333206,"u":"%"}},"2":{"Power":{"v":27.10000038,"u":"W"},"Voltage":{"v":30.79999924,"u":"V"},"Current":{"v":0.879999995,"u":"A"},"YieldDay":{"v":217,"u":"Wh"},"YieldTotal":{"v":30.50699997,"u":"kWh"},"Irradiation":{"v":9.033333778,"u":"%"}},"events":4}]
[image]
[image]

tbnobody, 2022-10-02:
Hm difference is, that the first inverter is a 1 channel, second one is a 2 channel inverter....

With some luck it's fixed with this commit: a3d9c6ae0fd90799642d7f0ef00198119aca9ef7

MFV, 2022-10-02:
Fixed with a3d9c6a. Thanks!

closed 2022-10-02 by MFV

tbnobody, 2022-10-02:
Sorry for inconvenience