tbnobody/OpenDTU #192: Support for HMT-2250-6T

ctandi, 2022-10-05:
Hi @tbnobody ,

first of all, thank you very much for this great piece of software!

@pinae, @jamct and I will be installing our first roof-scale photovoltaic array in two weeks. We already have two Hoymiles HMT-2250-6T and as they are not yet supported with OpenDTU, would like to contribute if possible.

Do you have any advice or by chance documentation that could help us getting started?

Thanks
Andrijan

haus-bus, 2022-10-07:
Hi,
I have two of HMT-2250-6T, too. This project looks very promising and I would like to contribute to support them. What exactly are the differences and does someone know what hardware is needed to communicate with the HMT series?

stefan123t, 2022-10-07:
Dear @ctandi (and @pinae & @jamct ) as well as @haus-bus 
we are discussing support for HMT in this issue over at Ahoy and have already identified several modules which may work for communication.
https://github.com/lumapu/ahoy/issues/233

Please also make sure to join our [Discord](https://discord.gg/WzhxEY62mB) server in order to discuss matters for HMT and HMS inverters normally managed by the DTU Pro-S in the `#hms-serie` channel.

@tbnobody I would suggest to close this issue and continue in Ahoy / Discord until someone has proven that the same HM-commands can be used via the CMT2300A or other compatible models or implemented a Proof of concept.

closed 2022-10-22 by tbnobody