tbnobody/OpenDTU #193: Bekomme keine Verbindung aufgebaut

vo5tr0, 2022-10-06:
Hi, 
ich hoffe, hier hat mir jemanden einen Tipp, leider bekomme ich einfach keine Verbindung zum Wechselrichter aufgebaut.

Gekauft habe ich die folgenden Produkte:
"Funkmodul NRF24L01+ kompatibel" (Ich hoffe, es liegt nicht daran, dass es nicht das originale Funkmodul ist)
https://www.roboter-bausatz.de/p/funkmodul-nrf24l01-kompatibel?number=RBS10024

 "ESP32 NodeMcu"
https://www.roboter-bausatz.de/p/esp32-nodemcu-entwicklungsboard-mit-bluetooth-und-wifi?number=RBS12259

"Olimex ESP32-POE-ISO IoT"
https://www.roboter-bausatz.de/p/olimex-esp32-poe-iso-iot-entwicklungsboard?number=RBS13830

Weder mit dem ESP32 NodeMcu noch mit der POE Version war ich erfolgreich. Ich habe eine Direkte Sichtverbindung zum Wechselrichter und bin vielleicht 5 Meter davon entfernt. Netzwerk ist da, Zeit synchronisiert. Seriennummer auch mehrfach geprüft. 
Das USB Kabel habe ich getauscht und das Netzteil sollte auch genug Power liefern. Das Funkmodul ist mittlerweile auf Maximaler Leistung eingestellt. Es ist allerdings ganz kalt. Sollte das nicht wenigstens etwas warm werden?

Jetzt ist leider die Sonne weg und ich kann nicht weiter testen. Aber vielleicht kann mir hier jemand für morgen den Entscheidenden Hinweis liefern :)

Danke und Grüße
Julian

tbnobody, 2022-10-06:
Hallo Julian, 
was siehst du denn auf der Seriellen Konsole?

vo5tr0, 2022-10-06:
Durchgehend nur das

19:29:58.773 > RX Period End
19:29:58.773 > All missing
19:29:58.773 > Nothing received, resend whole request
19:29:58.773 > TX Channel: 255 --> 15 81 85 26 70 78 56 34 12 80 5 0 63 3F 10 94 0 0 0 0 0 0 0 0 C7 7A AF 
19:29:58.773 > RX Period End
19:29:58.773 > All missing
19:29:58.773 > Nothing received, resend count exeeded
19:30:01.205 > Fetch inverter: 114180404196
19:30:01.206 > Request SystemConfigPara
19:30:01.208 > TX Channel: 255 --> 15 80 40 41 86 78 56 34 12 80 B 0 63 3F 10 99 0 0 0 0 0 0 0 0 99 2D F0 
19:30:01.410 > RX Period End
19:30:01.411 > All missing
19:30:01.411 > Nothing received, resend whole request
19:30:01.416 > TX Channel: 255 --> 15 80 40 41 86 78 56 34 12 80 B 0 63 3F 10 99 0 0 0 0 0 0 0 0 99 2D F0 
19:30:01.624 > RX Period End
19:30:01.625 > All missing
19:30:01.625 > Nothing received, resend whole request
19:30:01.632 > TX Channel: 255 --> 15 80 40 41 86 78 56 34 12 80 B 0 63 3F 10 99 0 0 0 0 0 0 0 0 99 2D F0 
19:30:01.838 > RX Period End
19:30:01.838 > All missing
19:30:01.838 > Nothing received, resend whole request
19:30:01.843 > TX Channel: 255 --> 15 80 40 41 86 78 56 34 12 80 B 0 63 3F 10 99 0 0 0 0 0 0 0 0 99 2D F0 
19:30:02.052 > RX Period End
19:30:02.053 > All missing
19:30:02.053 > Nothing received, resend whole request
19:30:02.058 > TX Channel: 255 --> 15 80 40 41 86 78 56 34 12 80 B 0 63 3F 10 99 0 0 0 0 0 0 0 0 99 2D F0 
19:30:02.266 > RX Period End
19:30:02.267 > All missing
19:30:02.267 > Nothing received, resend count exeeded
19:30:02.272 > TX Channel: 255 --> 15 80 40 41 86 78 56 34 12 80 5 0 63 3F 10 99 0 0 0 0 0 0 0 0 57 22 3F 
19:30:02.481 > RX Period End
19:30:02.482 > All missing

tbnobody, 2022-10-06:
1. Du verwendest eine ziemlich alte Version. Bitte Updaten
2. Drücke mal auf Reset am ESP. Die Zeilen die beim Starten zu sehen sind wären wichtiger.

tbnobody, 2022-10-06:
Außerdem ist mit ziemlicher Sicherheit deine Verkabelung falsch. Bitte hier einen Blick drauf werfen.

vo5tr0, 2022-10-06:
Ich habs gerade nochmal geklont und auch die Verkabelung geprüft. 
-DHOYMILES_PIN_IRQ=16 habe ich auf die 22 gelegt, weil es bei meinem Modell die 16 nicht gibt
(https://www.roboter-bausatz.de/media/pdf/0b/cd/28/SchematicsforESP32.pdf)

Hier nochmal der Monitor ab reset

20:17:46.954 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:17:46.959 > configsip: 0, SPIWP:0xee
20:17:46.961 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:17:46.967 > mode:DIO, clock div:2
20:17:46.969 > load:0x3fff0030,len:1184
20:17:46.972 > load:0x40078000,len:13132
20:17:46.974 > load:0x40080400,len:3036
20:17:46.976 > entry 0x400805e4
20:17:47.578 > 
20:17:47.578 > Starting OpenDTU
20:17:47.578 > Initialize FS... done
20:17:47.582 > Reading configuration... done
20:17:47.632 > Initialize Network... done
20:17:47.758 > Setting Hostname... Configuring WiFi STA using new credentials... done
20:17:47.769 > Initialize NTP... done
20:17:47.771 > Initialize MqTT... done
20:17:47.773 > Initialize WebApi... done
20:17:47.778 > Initialize Hoymiles interface... Connection error!!
20:17:47.792 >   Setting radio PA level... 
20:17:47.795 >   Setting DTU serial... 
20:17:47.797 >   Setting poll interval... 
20:17:47.799 >   Adding inverter: 114180404196 - Garage done
20:17:47.804 > done
20:17:47.804 > Switch to WiFi mode
20:17:47.806 > Setting Hostname... done
20:17:47.822 > Configuring WiFi STA using new credentials... done
20:17:47.832 > Configuring WiFi STA static IP... done
20:17:47.852 > WiFi disconnected
20:17:47.854 > Try reconnecting
20:17:47.855 > Network lost connection
20:17:47.931 > WiFi connected
20:17:47.934 > WiFi got ip: 192.168.178.154
20:17:47.938 > Network connected
20:17:52.544 > Fetch inverter: 114180404196
20:17:52.547 > Request SystemConfigPara
20:17:52.549 > TX RealTimeRunData Channel: 8 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 3F 1B D0 00 00 00 00 00 00 00 00 EF 07 43 
20:17:52.844 > RX Period End
20:17:52.846 > All missing
20:17:52.847 > Nothing received, resend whole request
20:17:52.850 > TX RealTimeRunData Channel: 8 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 3F 1B D0 00 00 00 00 00 00 00 00 EF 07 43 
20:17:53.153 > RX Period End
20:17:53.155 > All missing
20:17:53.156 > Nothing received, resend whole request
20:17:53.159 > TX RealTimeRunData Channel: 8 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 3F 1B D0 00 00 00 00 00 00 00 00 EF 07 43 
20:17:53.462 > RX Period End
20:17:53.464 > All missing
20:17:53.465 > Nothing received, resend whole request
20:17:53.468 > TX RealTimeRunData Channel: 8 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 3F 1B D0 00 00 00 00 00 00 00 00 EF 07 43 
20:17:53.771 > RX Period End
20:17:53.772 > All missing
20:17:53.774 > Nothing received, resend whole request
20:17:53.777 > TX RealTimeRunData Channel: 8 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 3F 1B D0 00 00 00 00 00 00 00 00 EF 07 43 
20:17:54.080 > RX Period End
20:17:54.081 > All missing
20:17:54.083 > Nothing received, resend count exeeded
20:17:54.086 > TX AlarmData Channel: 8 --> 15 81 10 12 29 78 56 34 12 80 11 00 63 3F 1B D0 00 00 00 00 00 00 00 00 35 1C 98 
20:17:54.696 > RX Period End
20:17:54.697 > All missing

tbnobody, 2022-10-06:
`20:17:47.778 > Initialize Hoymiles interface... Connection error!!`

Irgendwas passt an der Verkabelung nicht...

vo5tr0, 2022-10-06:
Also ich hab es jetzt nochmal mit dem POE Modell versucht und bekomme schonmal ein 
20:57:56.648 > Initialize Hoymiles interface... Connection successfull

Kann ich denn bei dem anderen einfach so das DHOYMILES_PIN_IRQ flag auf die 22 setzen? 
Wenn ich IRQ mit Pin 6 (GPIO16/RXD2) verbinde, ändert das allerdings auch nichts.

Immerhin bin ich mit dem POE Modell einen Schritt weiter :)

vo5tr0, 2022-10-07:
So siehts aktuell aus, leider noch keine Verbindung :(
Aber ich wüsste auch nicht, was ich an der Verkabelung noch ändern sollte. Es ist alles genau nach der vorgegebenen Pinbelegung angeschlossen.

[env:olimex_esp32_poe]
board = esp32-poe
build_flags = 
	${env.build_flags}
	-DHOYMILES_PIN_MISO=15
	-DHOYMILES_PIN_MOSI=2
	-DHOYMILES_PIN_SCLK=14
	-DHOYMILES_PIN_IRQ=13
	-DHOYMILES_PIN_CE=16
	-DHOYMILES_PIN_CS=5
	-DOPENDTU_ETHERNET
monitor_port = /dev/cu.usbserial-210
upload_port = /dev/cu.usbserial-210

Hab hier Mal zwei Fotos gemacht. Die Farben entsprechen deiner "Symbolic View" Grafik in der Readme
https://bilderupload.org/image/9ad954841-pxl-20221007-145537149.jpg
https://bilderupload.org/image/2ecd54879-pxl-20221007-145646416.jpg

16:49:29.126 > rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
16:49:29.131 > configsip: 0, SPIWP:0xee
16:49:29.134 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:49:29.141 > mode:DIO, clock div:2
16:49:29.144 > load:0x3fff0030,len:1184
16:49:29.145 > load:0x40078000,len:13132
16:49:29.148 > load:0x40080400,len:3036
16:49:29.148 > entry 0x400805e4
16:49:29.764 > 
16:49:29.764 > Starting OpenDTU
16:49:29.764 > Initialize FS... done
16:49:29.767 > Reading configuration... done
16:49:29.813 > Initialize Network... ETH start
16:49:31.478 > ETH connected
16:49:31.527 > done
16:49:31.528 > Setting Hostname... Configuring WiFi STA using new credentials... done
16:49:31.537 > Initialize NTP... done
16:49:31.537 > Initialize MqTT... done
16:49:31.540 > Initialize WebApi... done
16:49:31.551 > Initialize Hoymiles interface... Connection successfull
16:49:31.563 >   Setting radio PA level... 
16:49:31.566 >   Setting DTU serial... 
16:49:31.569 >   Setting poll interval... 
16:49:31.571 >   Adding inverter: 114180404196 - Garage done
16:49:31.583 > done
16:49:31.587 > Switch to Ethernet mode
16:49:31.587 > WiFi disconnected
16:49:31.622 > Configuring Ethernet static IP... ETH got IP: 192.168.178.147
16:49:31.629 > Network connected
16:49:31.632 > Connecting to MQTT...
16:49:31.634 > done
16:49:31.634 > Setting Hostname... done
16:49:31.643 > Connected to MQTT.
16:49:34.733 > Fetch inverter: 114180404196
16:49:34.734 > Request SystemConfigPara
16:49:34.736 > TX RealTimeRunData Channel: 23 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 40 3C 7E 00 00 00 00 00 00 00 00 ED 99 29 
16:49:34.977 > RX Period End
16:49:34.977 > All missing
16:49:34.977 > Nothing received, resend whole request
16:49:34.983 > TX RealTimeRunData Channel: 40 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 40 3C 7E 00 00 00 00 00 00 00 00 ED 99 29 
16:49:35.228 > RX Period End
16:49:35.229 > All missing
16:49:35.229 > Nothing received, resend whole request
16:49:35.234 > TX RealTimeRunData Channel: 61 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 40 3C 7E 00 00 00 00 00 00 00 00 ED 99 29 
16:49:35.480 > RX Period End
16:49:35.481 > All missing
16:49:35.481 > Nothing received, resend whole request
16:49:35.486 > TX RealTimeRunData Channel: 75 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 40 3C 7E 00 00 00 00 00 00 00 00 ED 99 29 
16:49:35.732 > RX Period End
16:49:35.732 > All missing
16:49:35.732 > Nothing received, resend whole request
16:49:35.737 > TX RealTimeRunData Channel: 3 --> 15 81 10 12 29 78 56 34 12 80 0B 00 63 40 3C 7E 00 00 00 00 00 00 00 00 ED 99 29 
16:49:35.983 > RX Period End
16:49:35.984 > All missing
16:49:35.984 > Nothing received, resend count exeeded
16:49:35.989 > TX AlarmData Channel: 23 --> 15 81 10 12 29 78 56 34 12 80 11 00 63 40 3C 7E 00 00 00 00 00 00 00 00 37 82 F2 
16:49:36.535 > RX Period End
16:49:36.536 > All missing
16:49:36.536 > Nothing received, resend whole request
16:49:36.541 > TX AlarmData Channel: 40 --> 15 81 10 12 29 78 56 34 12 80 11 00 63 40 3C 7E 00 00 00 00 00 00 00 00 37 82 F2 
16:49:37.088 > RX Period End
16:49:37.088 > All missing
16:49:37.088 > Nothing received, resend whole request
16:49:37.093 > TX AlarmData Channel: 61 --> 15 81 10 12 29 78 56 34 12 80 11 00 63 40 3C 7E 00 00 00 00 00 00 00 00 37 82 F2 
16:49:37.640 > RX Period End
16:49:37.641 > All missing
16:49:37.641 > Nothing received, resend whole request
16:49:37.645 > TX AlarmData Channel: 75 --> 15 81 10 12 29 78 56 34 12 80 11 00 63 40 3C 7E 00 00 00 00 00 00 00 00 37 82 F2 
16:49:38.191 > RX Period End
16:49:38.192 > All missing
16:49:38.192 > Nothing received, resend whole request
16:49:38.197 > TX AlarmData Channel: 3 --> 15 81 10 12 29 78 56 34 12 80 11 00 63 40 3C 7E 00 00 00 00 00 00 00 00 37 82 F2 
16:49:38.743 > RX Period End
16:49:38.743 > All missing
16:49:38.743 > Nothing received, resend count exeeded
16:49:38.748 > TX SystemConfigPara Channel: 23 --> 15 81 10 12 29 78 56 34 12 80 05 00 63 40 3C 7E 00 00 00 00 00 00 00 00 23 96 E6 
16:49:38.996 > RX Period End
16:49:38.996 > All missing
16:49:38.996 > Nothing received, resend whole request
16:49:39.001 > TX SystemConfigPara Channel: 40 --> 15 81 10 12 29 78 56 34 12 80 05 00 63 40 3C 7E 00 00 00 00 00 00 00 00 23 96 E6 
16:49:39.247 > RX Period End
16:49:39.247 > All missing

tbnobody, 2022-10-07:
Also das senden sieht jetzt erstmal korrekt aus. D.h. die Verbindung zum NRF Modul ist vorhanden. (IRQ kann man so erstmal aber nicht prüfen). Folgende Dinge würden mir noch einfallen:
* Bitte prüfe ob es wirklich ein nRF24L01+ Modul ist (das PLUS ist ganz entscheidend und muss auch auf dem IC abgebildet sein)
* Prüfe ob der IRQ Pin korrekt angeschlossen ist (laut Bild muss das aber eigentlich passen)
* Prüfe die Seriennummer

Gitsaibot, 2022-10-07:
> Ich habs gerade nochmal geklont und auch die Verkabelung geprüft. -DHOYMILES_PIN_IRQ=16 habe ich auf die 22 gelegt, weil es bei meinem Modell die 16 nicht gibt (https://www.roboter-bausatz.de/media/pdf/0b/cd/28/SchematicsforESP32.pdf)

Also ich habe genau die gleiche Hardware mit den 30 Pins und da gibt es den Pin 16. Hier ist ein Link zum Pinout der 30er Version: https://www.etechnog.com/2021/12/esp32-pinout-diagram-.html

Pinout vom nRF24L01+ Modul https://howtomechatronics.com/tutorials/arduino/arduino-wireless-communication-nrf24l01-tutorial/

Hab mir dann schnell die Tabelle erstellt(fand ich einfacher als den Farbelinien zu folgen ;)): 

**ESP** -------- **nRF24L01+**
GPIO 23 -> 6 MOSI
GPIO 19 -> 7 MISIO
GPIO 18 -> 5 SCK
GPIO 16 -> 8 IRQ
GPIO 5 -> 4 CS
GPIO 4 -> CE
3,3V -> 2 VCC
GND -> 1 GND

stefan123t, 2022-10-07:
@vo5tr0 es gibt wohl auch nRF24 Module mit einer etwas falschen / spiegelverkehrten Beschriftung (auf der Unterseite).
Schließe mal GND an den Pin 1 der auf dem nRF24 Modul mit einem kleinen Quadrat einge-"kreist" ist.
Direkt daneben ist VCC / +3.3V also Pin 2 und so weiter. In der äußeren Reihe ungerade Pins und näher zur Antenne die geraden Pins.

tbnobody, 2022-10-07:
> @vo5tr0 es gibt wohl auch nRF24 Module mit einer etwas falschen / spiegelverkehrten Beschriftung (auf der Unterseite).

Die SPI Verbindung klappt aber ja. Sonst käme beim Startup ein Connection Failed und auch die Channels bei den TX Ausgaben währen undefiniert (so wie es am Anfang war)

bwired-nl, 2022-10-08:
I have the exact same problem as Above
Pinsetting exact the same and had to change the pin setting in the platformio.ini as stated to
-DHOYMILES_PIN_MISO=19
-DHOYMILES_PIN_MOSI=23
-DHOYMILES_PIN_SCLK=18
-DHOYMILES_PIN_IRQ=16
-DHOYMILES_PIN_CE=4
-DHOYMILES_PIN_CS=5

also getting
16:49:38.743 > Nothing received, resend count exeeded
16:49:38.748 > TX SystemConfigPara Channel: 23 --> 15 81 10 12 29 78 56 34 12 80 05 00 63 40 3C 7E 00 00 00 00 00 00 00 00 23 96 E6

btw: what  DTU Serial, is this Auto generated?

Thanks

vo5tr0, 2022-10-08:
> Also das senden sieht jetzt erstmal korrekt aus. D.h. die Verbindung zum NRF Modul ist vorhanden. (IRQ kann man so erstmal aber nicht prüfen). Folgende Dinge würden mir noch einfallen:
>
> * Bitte [...]

Mir wurde wohl das falsche Modul geliefert... 
Aufgedruckt ist folgendes:
Si24R1
703HH11
 
So viel zu "Funkmodul NRF24L01+ kompatibel"... :(

bwired-nl, 2022-10-09:
I used the pin setting as in the drawings, this is my INI file setting now, i think the problem lays there (see below)
Still not working and i get this in the monitor

Starting OpenDTU
Initialize FS... done
Reading configuration... done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Hoymiles interface... Connection successfull
  Setting radio PA level... 
  Setting DTU serial... 
  Setting poll interval... 
  Adding inverter: 116181852181 - Hoymiles1 done
done
Switch to WiFi mode
WiFi connected
Setting Hostname... done
Configuring WiFi STA using existing credentials... done
Configuring WiFi STA DHCP IP... done
WiFi got ip: 10.0.0.145
Network connected
Connecting to MQTT...
Connected to MQTT.
Fetch inverter: 116181852181
Request SystemConfigPara
TX RealTimeRunData Channel: 23 --> 15 81 85 21 81 78 56 34 12 80 0B 00 63 42 88 7D 00 00 00 00 00 00 00 00 67 C0 41 
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 40 --> 15 81 85 21 81 78 56 34 12 80 0B 00 63 42 88 7D 00 00 00 00 00 00 00 00 67 C0 41 
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 61 --> 15 81 85 21 81 78 56 34 12 80 0B 00 63 42 88 7D 00 00 00 00 00 00 00 00 67 C0 41 
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 75 --> 15 81 85 21 81 78 56 34 12 80 0B 00 63 42 88 7D 00 00 00 00 00 00 00 00 67 C0 41 
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 3 --> 15 81 85 21 81 78 56 34 12 80 0B 00 63 42 88 7D 00 00 00 00 00 00 00 00 67 C0 41 
RX Period End
All missing
Nothing received, resend count exeeded
TX AlarmData Channel: 23 --> 15 81 85 21 81 78 56 34 12 80 11 00 63 42 88 7D 00 00 00 00 00 00 00 00 BD DB 9A 
RX Period End
All missing
Nothing received, resend whole request
TX AlarmData Channel: 40 --> 15 81 85 21 81 78 56 34 12 80 11 00 63 42 88 7D 00 00 00 00 00 00 00 00 BD DB 9A 
RX Period End
All missing
Nothing received, resend whole request
TX AlarmData Channel: 61 --> 15 81 85 21 81 78 56 34 12 80 11 00 63 42 88 7D 00 00 00 00 00 00 00 00 BD DB 9A 
RX Period End
All missing

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
PLATFORMIO INI

[platformio]
default_envs = generic

[env]
framework = arduino
platform = espressif32@>=5.2.0

build_flags =
    -D=${PIOENV}
    -DCOMPONENT_EMBED_FILES=webapp_dist/index.html.gz:webapp_dist/zones.json.gz:webapp_dist/favicon.ico:webapp_dist/js/app.js.gz
    -Wall

lib_deps =
    https://github.com/yubox-node-org/ESPAsyncWebServer
    bblanchon/ArduinoJson @ ^6.19.4
    https://github.com/bertmelis/espMqttClient.git#v1.3.2
    nrf24/RF24 @ ^1.4.5

extra_scripts =
    pre:auto_firmware_version.py

board_build.partitions = partitions_custom.csv
board_build.filesystem = littlefs
monitor_filters = time, colorize, log2file, esp32_exception_decoder
monitor_speed = 115200
upload_protocol = esptool

[env:generic]
board = esp32dev
monitor_port = COM3
upload_port = COM3

[env:olimex_esp32_poe]
; https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

board = esp32-poe
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5
    -DOPENDTU_ETHERNET

monitor_port = COM3
upload_port = COM3

[env:olimex_esp32_evb]
; https://www.olimex.com/Products/IoT/ESP32/ESP32-EVB/open-source-hardware

board = esp32-evb
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5
    -DOPENDTU_ETHERNET

monitor_port = /dev/tty.usbserial-1450
upload_port = /dev/tty.usbserial-1450

[env:d1 mini esp32]
board = wemos_d1_mini32
build_flags =
    ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5

monitor_port = /dev/cu.usbserial-01E68DD0
upload_port = /dev/cu.usbserial-01E68DD0

bwired-nl, 2022-10-09:
it makes me crazy, tried everything bit not working
it even created a platform. ini like below
use the exact pin gpio setting as on the website drawing
the webpage system info tells connected, all looks good, also ntp etc,
but still getting this in the log

[14:01:18]Nothing received, resend count exeeded
[14:01:18]TX AlarmData Channel: 23 --> 15 81 85 21 81 78 56 34 12 80 11 00 63 42 B8 0C 00 00 00 00 00 00 00 00 10 B3 1E 
[14:01:18]RX Period End
[14:01:18]All missing

========================================================
[platformio]
default_envs = generic

[env]
framework = arduino
platform = espressif32@>=5.2.0

build_flags =
D=${PIOENV}
    -DCOMPONENT_EMBED_FILES=webapp_dist/index.html.gz:webapp_dist/zones.json.gz:webapp_dist/favicon.ico:webapp_dist/js/app.js.gz
    -Wall
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5

wib100, 2022-10-09:
Hi,
what's the distance to your inverter?
Did you already try to raise the PA Level?
Regarding DTU Serial: it's defined in defaults.h and can also be changed via WebGUI.

wib100, 2022-10-09:
If you want to compile for the D1 mini, you can either switch to the environment via VSC or just change the entries in the platform.ini from:
[platformio]
default_envs = generic
to:
[platformio]
default_envs = d1 mini esp32

Depending on your OS, you may also have to change:
monitor_port = /dev/cu.usbserial-01E68DD0
upload_port = /dev/cu.usbserial-01E68DD0

HTH!

bwired-nl, 2022-10-09:
> Hi, what's the distance to your inverter? Did you already try to raise the PA Level? Regarding DTU Serial: it's defined in defaults.h and can also be changed via WebGUI.

thanks
yes tried everything, putting it closer, raise power outputs etc.
i know everything should be able to work, i get a Mini D1 working on the Ahoy-dtu firmware, but its not stable
busy for day now allready :-(

stefan123t, 2022-10-10:
@bwired-nl what is the current source for your OpenDTU build, maybe you can post some more details ?

Did you add the 100uF electrolytic cap to the nRF24 modules Vcc/3.3V and GND pins ?
We have seen several outages in the past because sending mode drains the voltage regulator on the ESP board which supplies the 3.3V input of the nRF communications board.

What type of nRF module do you use, one with external Antenna or the cheap ones with a circuit board antenna ? The latter one need to be set to HIGH PowerLevel to have the same yield as the ones with ext Antenna (PA+LNA) at LOW/MIN PowerLevels. That is also the consumption may increase with the cheap modules.

Regarding your inverter based on the serial Id its most likely a HM-1000..1500 manufactured in calendar week 18 of 2022. I understood it works with AhoyDTU intermittently but Ahoy is too flaky for your purpose. Hence we can assume the inverter to be supported in principle.

Do you run AhoyDTU and OpenDTU at the same time they may impact each other especially if they use the same DTU Serial and query the same inverter. But also they are using the same RF channels. Can you try solely with OpenDTU connected ?

I can see that your commands contain a proper timestamp which sometimes is a problem with AhoyDTU to begin communications.
[14:01:18]TX AlarmData Channel: 23 --> 15 81 85 21 81 78 56 34 12 80 11 00 >63 42 B8 0C< 00 00 00 00 00 00 00 00 10 B3 1E

Did you try to change the CE and CS pins? I believe I had to habe them assigned the other way around in my own lolin32 ini env.

Kind regards,
Stefan

vo5tr0, 2022-10-11:
Heute kam das richtige Modul an:
24L01+
2219AH

Leider nach wie vor keine Verbindung und den Identischen Output wie schon oben :(
Hast du noch eine Idee? Die aktuelle Firmware ce9c0e8 habe ich heute auch installiert

bwired-nl, 2022-10-12:
> @bwired-nl what is the current source for your OpenDTU build, maybe you can post some more details ?
>
> Did you add the 100uF electrolytic cap to the nRF24 modules Vcc/3.3V and GND pins ? We have seen [...]

put the 100uf in but no diff, 
unbelievable is this, makes me crazy
see below images of my settings

![IMG_2033](https://user-images.githubuserc
[image]
ontent.com/5882794/195283688-60b93b43-caec-4ac5-8d7e-476412617776.jpg)

bwired-nl, 2022-10-12:
Yesyesyes  i got it working, dont use the transmitter a in above picture
I bought the standard one (picture on the mainpage schematic) without the antenna and its working directly :-)

tbnobody, 2022-10-12:
I am going to push a new version this evening which also detects the Chip Model (Plus / non-Plus). If you like you can test your previous model with this version. 

I am using the module with external antenna since a lot of weeks without any problems. But bought mine from amazon.

Sprinterfreak, 2022-10-16:
> Yesyesyes i got it working, dont use the transmitter a in above picture I bought the standard one (picture on the mainpage schematic) without the antenna and its working directly :-)

I don't agree. The ones with antenna do generally work. What we already discovered in the mc forum's thread is, that there is some margin of chips with defects in stock. These are cheap chineese clones. As always with stuff from china. If you need one, buy ten.

Also

  * try different tx power settings
  * orient antenna perpendicular to the line of sight with inverter
  * put the NRF as far as possible away from the ESP's wifi antenna (same frequency, overloads hf receiver)
  * expect bad connections from jumper wires.  Basically you can observe communication issues when blowing at it.

I recently switched to Ebyte ML01DP5 modules. They come with proper shielding and seem to be of much better quality. Of course a little more expensive. But fully pin and protocol compatible.

vo5tr0, 2022-10-20:
Gestern kam bei mir das Modul mit externer Antenne an. Damit klappt es jetzt endlich :)

Also entweder die kleinen waren alle defekt, oder die Reichweite ist extrem schlecht :/
5 Meter direkte Sichtverbindung sollten die doch auf der maximalen Sendeleistung packen, oder?

Naja, wie auch immer, ich bin froh, dass ich endlich loslegen kann

tbnobody, 2022-10-20:
Meiner läuft auf minimaler Leistung geht vom Keller aus über mehrere Räume hinweg in den Garten raus ohne Probleme.

Aber schön das es läuft! Würde den issue hier erstmal schließen.

closed 2022-10-20 by tbnobody

Sprinterfreak, 2022-10-20:
@vo5tr0 
> 5 Meter direkte Sichtverbindung sollten die doch auf der maximalen Sendeleistung packen, oder?

Maximale Sendeleistung funktioniert bei mir auch nicht. Ds ist so wie Musik, wenn man am Mischpult alle Regler auf Anschlag hoch schiebt, klingts nicht mehr gut. Man muss den richtigen Pegel finden, dass sich das Signal "gut anhört". Bei Funk kommt noch hinzu, dass man mit viel zu hoher Sendeleistung sinnlos Bandbreite in teils Kilometern Umkreis illegal(!) blockiert. Thema re-use, fair-use. Die Luft ist ein knappes shared-medium.

FrodoVDR, 2022-10-21:
Ich habe das Modul ohne externe Antenne von AZ-Delivery und betreibe es in der Standard Einstellung Minimum , bisher hatte ich keine Problem.
Die Wände sind aber zumeist Holz oder Gipskarton mit Metall Konstruktion und das OpenDTU ist in ca. 6 Meter Entfernung.

AlfAlex, 2022-10-25:
Vielleicht wurde die PIN Konfiguration in der platformio.ini falsch eingetragen?
Es gibt in der Default-Datei mehrere Blöcke (für jede Art von ESP-Device) eines und ich hatte die Konfiguration vom eps32-poe verändert, bzw gesteckt, aber nicht bemerkt, dass es noch einen genersichen Block gibt, der auch genutzt wird!
Nachdem ich die PIN-Konfiguration dorthin kopiert habe und das neue Build hochgeladen habe lief es sofort einwandfrei.