tbnobody/OpenDTU #215: WLAN Reconnect after Pwoer loss AP

tiff1112, 2022-10-12:
Hi,
after Power loss of the WLAN Infrastructure and the DTU Device, the device cant connect because AP need more time to start. 
Is it possible to try to reconnect to wlan after x seconds? maybe only a SSID is set in Config?

thanks

tbnobody, 2022-10-12:
Thats already implemented... When WiFi gets disconnected it tries to reconnect for 15 seconds. After this time the reconnect will stop for 10 minutes. This is done because during the connection phase the integrated AP gets very unstable (due to frequency cycling).

closed 2022-10-12 by tbnobody

tiff1112, 2022-10-12:
Sry maybe i dont explain it right. I had Power loss at the Router and Dtu. After Power on the device try to connect But the Router is Not ready. After half an hour the Router is still back But dtu does Not connect.
After reboot dtu it connect again.