tbnobody/OpenDTU #218: Build error after latest commits

szepnorbee, 2022-10-13:
I can't build the project after the latest commits. Have anyone issue with it?

In file included from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:103,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from include/Configuration.h:4,
                 from src/Configuration.cpp:5:
C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:10: fatal error: xtensa/hal.h: No such file or directory
 #include <xtensa/hal.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic\src\Configuration.cpp.o] Error 1
In file included from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:103,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from include/Configuration.h:4,
                 from include/MqttPublishing.h:4,
                 from src/MqttPublishing.cpp:5:
C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:10: fatal error: xtensa/hal.h: No such file or directory
 #include <xtensa/hal.h>
          ^~~~~~~~~~~~~~
compilation terminated.
In file included from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:103,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from include/Configuration.h:4,
                 from include/MqttHassPublishing.h:4,
                 from src/MqttHassPublishing.cpp:5:
C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:10: fatal error: xtensa/hal.h: No such file or directory
 #include <xtensa/hal.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic\src\MqttPublishing.cpp.o] Error 1
*** [.pio\build\generic\src\MqttHassPublishing.cpp.o] Error 1
In file included from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:103,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.h:38,
                 from C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/libraries/DNSServer/src/DNSServer.h:3,
                 from include/NetworkSettings.h:4,
                 from include/MqttSettings.h:4,
                 from src/MqttSettings.cpp:5:
C:/Users/Szep Norbert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:10: fatal error: xtensa/hal.h: No such file or directory
 #include <xtensa/hal.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic\src\MqttSettings.cpp.o] Error 1
================================================================================================ [FAILED] Took 24.53 seconds ================================================================================================
Environment    Status    Duration
-------------  --------  ------------
generic        FAILED    00:00:24.529
=========================================================================================== 1 failed, 0 succeeded in 00:00:24.529 ===========================================================================================

Solateur, 2022-10-13:
With the latest workflow (see below) I was able to build the project without any error messages.

[image]

D3R-ST3FAN, 2022-10-13:
just compiled [d7fe495] without issues for generic and olimex

tbnobody, 2022-10-14:
It seems that the issue is somewhere at your platformio installation as the preprocessor does not find include files:
`fatal error: xtensa/hal.h: No such file or directory` which belong to the framework and not to OpenDTU.
Did you already try a clean build before rebuilding?

tbnobody, 2022-10-26:
Closing this issue due to no response (and it seem to work within the actions etc.)

closed 2022-10-26 by tbnobody