tbnobody/OpenDTU #222: Feature Request: Battery Mode

johannesosterberg, 2022-10-15:
First of all, thank you for the helpful work, I had always been looking for something to avoid having to install a Sun1000 or something similar in my system.

Now the feature idea, at the moment OpenDTU can measure the voltage, in my case it is the voltage of the 24v LFT battery. Couldn't you now develop a battery mode where you set the inverter to switch off at a certain voltage? You can also do this via a node red or something like that, but I actually want to have as little house automation as possible.

Maybe this is an idea for further development. thank you very much.

Translated with www.DeepL.com/Translator (free version)

Sprinterfreak, 2022-11-27:
I think OpenDTU should remain a gateway. If we start to implement such nish circumstances, we would cause the project to explode quickly. We should also keep in mind the very limited storage sizes and room for future implementations for interfacing with a wider variety of inverters.

OpenDTU still provides a powerful interface which enables flexible and simple external development.
Such custom requirements are arguably easier to implement externally and communicate via MQTT.

johannesosterberg, 2022-11-28:
You are right, I have since built a simple Pyrthon sqript that does exactly what I want and it works fine for my setup. I was able to develop it myself without much skill.

https://github.com/johannesosterberg/hm_nulleinspeisung

closed 2022-11-28 by johannesosterberg