tbnobody/OpenDTU #224: Olimex ESP32-POE - Errors if Ethernet is not connected (WiFi-Mode)

wib100, 2022-10-15:
Hi,
I did some test with the current version and noted that there are some errors reported in the serial console if the Ethernet port is not connected during startup:
[esp32-eth-error during startup.txt](https://github.com/tbnobody/OpenDTU/files/9791913/esp32-eth-error.during.startup.txt)

I compiled the firmware with the default platform.ini (no debug switched used).

Thanks!

stromibaer, 2022-10-21:
Im running Olimex too with WiFi enabled. It's working. What problems do you have with it?

wib100, 2022-10-21:
It's working fine, but I noted that there are some errors during device startup if no ethernet is connected and just wanted to report this.

tbnobody, 2022-10-22:
Yeah I also saw this output during my tests. Seems to come from the deep of the esp-idf. Would ignore it for now as it doesn't really affect the function.

closed 2022-10-22 by tbnobody