tbnobody/OpenDTU #226: Nach Flashen kein Access Point

MaxiKing007, 2022-10-15:
Hallo,
flashen über Visual Studio Code hat geklappt, doch dann wird kein Access Point aufgemacht? Zumindest unter http://192.168.4.1/ kommt ein Seiten-Ladefehler....habe diesen ESP32: https://www.ebay.de/itm/255283221996
Wäre für jeden Kommentar dankbar.
LG

tbnobody, 2022-10-15:
Was sagt denn die Serielle Konsole?

MaxiKing007, 2022-10-15:
Ahh mein Fehler. Klappt alles!!!
Aber trotzdem kurze andere Frage (da ich mich noch nie mit einem ESP32 beschäftigt habe): Wie schauen ich mir die serielle Konsole an?

tbnobody, 2022-10-15:
Im visual studio in der statusleiste unten das symbol mit dem stecker (vgl. hier: https://community.platformio.org/uploads/default/original/2X/5/53cf9531bead9a5354aa1a75c0b2ff8d5f4b6313.png)

closed 2022-10-15 by tbnobody

meystervie, 2022-11-09:
Hallo, habe mit VS Studio erfolgreich geflasht, AP OpenDTU-* wird angezeigt, aber login mit Passwort openDTU42 nicht möglich. Konsole meldet "interrupt received". Ideen? Chip ist ESS32 Wroom
LG

tbnobody, 2022-11-09:
Du erhältst "interrupt received" obwohl es Nachts ist und dein WR dadurch ausgeschaltet ist? --> IRQ Pin falsch angeschlossen

meystervie, 2022-11-09:
ja genau. den WR habe ich noch gar nicht einstellen können, weil ich nichts in WIFI komme und damit nicht ins Interface. Also die PINs noch einmal checken?

meystervie, 2022-11-09:
War in der Tat ein PIN falsch belegt (am Radio Board sind sie nicht bezeichnet. "interrupt received" ist weg. Passwort openDTU42 wird noch immer nicht angenommen. 
Konsole sagt: 
--- Terminal on COM6 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
22:41:37.202 > q_drv:0x00,
22:41:37.202 > Starting OpenDTU
22:41:37.202 > Initialize FS... done
22:41:37.202 > Reading configuration... done
22:41:37.202 > Initialize Network... done
22:41:37.202 > Setting Hostname... Initialize NTP... done
22:41:37.202 > Initialize MqTT... done
22:41:37.202 > Initialize WebApi... done
22:41:37.202 > Check for default DTU serial... done
22:41:37.202 > Initialize Hoymiles interface... Connection successfull
22:41:37.202 >   Setting radio PA level...
22:41:37.202 >   Setting DTU serial...
22:41:37.202 >   Setting poll interval...
22:41:37.202 > done
22:41:37.202 > Switch to WiFi mode
22:41:37.202 > Setting Hostname... done
22:41:48.859 > Disable search for AP... done

meystervie, 2022-11-09:
den ESP habe ich heute bekommen. War augenscheinlich neu von Amazon geliefert. Die Firmware habe ich selbst mit VS Studio compiliert.

meystervie, 2022-11-09:
habe gerade die vorkomplilierte BIN mit dem Flash Download Tool nach Anleitung überspielt inkl. Erase. Gleiches Problem. Kann es mit den PINs zu tun haben zum Radio Board oder eher ESP Chip falsch? Hast Du weitere Ideen?

tbnobody, 2022-11-09:
Ne, kann normal nicht mit den pins zu tun haben. v.a. diese zeile:
`22:41:37.202 > Initialize Hoymiles interface... Connection successfull`
bedeutet eigentlich das das nrf24 modul nicht so falsch angeschlossen sein kann. (IRQ kann immer noch falsch sein aber das sollte keinen einfluss auf die passwort eingabe haben)

hast du die möglichkeit dich mit einem anderen gerät Mobiltelefon/Laptop zum esp zu verbinden? (beim Passwort ist die Groß/Kleinschreibung wichtig und auch darauf achten, dass am ende kein leerzeichen oder so eingegeben wird.)

meystervie, 2022-11-09:
auch mit iPhone und Android das Gleiche. Passwort habe ich kontrolliert. 
Habe das gleiche Hailege ESP von Amazon wie GoGitl
https://www.amazon.de/gp/product/B07YKBY53C/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
Liegt vmtl dann daran?

tbnobody, 2022-11-09:
Hm das muss dann schon ein Hardware Defekt am ESP sein. Es gibt zumindest in den Rezessionen auch sowas hier:

> Zwei Stück gekauft, zwei verbinden sich nicht mit dem Rechner.

Da kann man ein Qualitätsproblem jetzt leider nicht ausschließen. Aber 100% sagen kann ich das leider auch nicht.

meystervie, 2022-11-09:
Alles klar. schicke den zurück und bestell mal einen anderen.

meystervie, 2022-11-11:
der hier wurde heute geliefert. 
**https://www.amazon.de/gp/product/B071P98VTG/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1**
läuft auf Anhieb. Super Software. Danke!