tbnobody/OpenDTU #230: When OpenDTU starts DTU Pro crashes

gogogommez, 2022-10-16:
Whenever i activate the OpenDTU, after a view minutes the DTU Pro does not work correct anymore. S-Miles-Cloud doesn’t get any data anymore. I have to switch the OpenDTU off and have to unpower/power the DTU Pro. After a view minutes the DTU Pro works again.

Solateur, 2022-10-16:
If this blog entry 
https://www.mikrocontroller.net/topic/525778?page=8#7113046
is correct, parallel operation of DTU pro and OpenDTU is not possible.

gogogommez, 2022-10-16:
Ahoy DTU and DTU Pro work well at the same time.

Sprinterfreak, 2022-10-16:
Inverters are only designed to talk with one master. There are errors to be expected, because inverters keep track of for instance log messages already being send. If two masters talking to the device, one of them gets communication errors for requesting information already send to the other master.
Ahoy working together with DTU Pro maybe boils down to lower poll rates and less deep protocol implementation.

OpenDTU is a replacement for DTU Pro. Not an extension.

tbnobody, 2022-11-01:
I am going to close this issue. It seems that 2 masters is in general a bad idea.

closed 2022-11-01 by tbnobody

mi-hol, 2022-11-02:
> 2 masters is in general a bad idea.

Maybe adding a section "Limitations" in readme would help newbies to avoid this issue?