tbnobody/OpenDTU #246: Temperaturwert plötzlich falsch

ce3rd, 2022-10-20:
Hi, seit heute in der Früh wird die Temperatur falsch angezeigt - liegts am hm1500 oder an opendtu? Kann sein, dass es bei uns heute das erste mal minus Grade hatte - ev. das schuld? 
json "Temperature":{"v":6552.200195,"u":"°C"}
mqtt und web im anhang
[image]
[image]

ce3rd, 2022-10-20:
ok, jetzt wo wir ins + rutschen passt es wieder: "Temperature":{"v":0.200000003,"u":"°C"}

tbnobody, 2022-10-20:
Hallo,

damit bist du wohl der erste der jemals je negative temperaturen gemessen hat :)
Muss mal gucken wo da eine Konvertierung schief geht.

stefan123t, 2022-10-20:
@tbnobody sieht nach einem fixed point integer und keinem uint aus.
65536 bzw. 6553.6 wäre demnach -0.1 °C wenn ich es richtig interpretiere.

homeautomation2022, 2022-10-20:
quick fix für den eigentlich int16_t Wert mit einer Kommastelle, der aber als uint16_t zu float verarbeitet wird:
```
if (TEMP > 3276.7)
    TEMP -= 6553.6;
```

commit 3ab7d9f6 (2022-10-20): Fix #246: Allow negative float values

This is required for the temperature field which can be negative as well
closed 2022-10-20 by tbnobody