tbnobody/OpenDTU #250: Build errors on Debian system with new compile flags

b-a-9-0, 2022-10-20:
New complie flags cause build errors on debian build system (docker image: python:3.11-rc-bullseye). Most probably this issue is caused by the build flags in platform.ini.

`Building in release mode
Converting .pio/build/esp32doit-devkit-v1/index.html.gz.txt.o
Converting .pio/build/esp32doit-devkit-v1/zones.json.gz.txt.o
Converting .pio/build/esp32doit-devkit-v1/favicon.ico.txt.o
Converting .pio/build/esp32doit-devkit-v1/app.js.gz.txt.o
Compiling .pio/build/esp32doit-devkit-v1/src/Configuration.cpp.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
Compiling .pio/build/esp32doit-devkit-v1/src/MqttHassPublishing.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/MqttPublishing.cpp.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
Compiling .pio/build/esp32doit-devkit-v1/src/MqttSettings.cpp.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
Compiling .pio/build/esp32doit-devkit-v1/src/NetworkSettings.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/NtpSettings.cpp.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
Compiling .pio/build/esp32doit-devkit-v1/src/WebApi.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/WebApi_config.cpp.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
Compiling .pio/build/esp32doit-devkit-v1/src/WebApi_devinfo.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/WebApi_dtu.cpp.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
Compiling .pio/build/esp32doit-devkit-v1/src/WebApi_eventlog.cpp.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
Compiling .pio/build/esp32doit-devkit-v1/src/WebApi_firmware.cpp.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/NtpSettings.cpp.o] Error 1
cc1plus: all warnings being treated as errors
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/MqttPublishing.cpp.o] Error 1
*** [.pio/build/esp32doit-devkit-v1/src/NetworkSettings.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/WebApi.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/Configuration.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/MqttSettings.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/MqttHassPublishing.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/WebApi_firmware.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/WebApi_eventlog.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/WebApi_devinfo.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/WebApi_config.cpp.o] Error 1
cc1plus: all warnings being treated as errors
*** [.pio/build/esp32doit-devkit-v1/src/WebApi_dtu.cpp.o] Error 1
========================= [FAILED] Took 40.93 seconds =========================`

tbnobody, 2022-10-20:
Make shure your environments don't contain spaces or dashes (-)... Just use underscores instead (_)

b-a-9-0, 2022-10-21:
That solved it, thanks!

closed 2022-10-21 by b-a-9-0