tbnobody/OpenDTU #254: Home Assist Auto Discovery does not work

LuckyLuzz, 2022-10-21:
Hi,
first of all THANKS :) This is a great project!

My OpenDTU Divice is running since two months without issues. Last week I tried to set ip up with Home Assistant but it did not work out of the Box. I had to add line to the YAML configuration file withhin HA. Following this Post [https://community.home-assistant.io/t/hoymiles-dtu-microinverters-pv/253674/21](url)

After I got it working with the updated YAML script I had a look over the MQTT Broker via MQTT Explorer (screenshot attached). 

[image]

I don't really know why auto discovery does not work but if you comepare the OpenDTU Topic with the Tasmoto Topics you see that they are totaly different. The structure might be it.

Thanks again for this great project :)

tbnobody, 2022-10-21:
This is the raw structure of the mqtt topics... This has nothing todo with the Home Assistant Auto Discovery Feature... Did you enable Home Assistant Auto Discovery in the MQTT Topics?

tbnobody, 2022-10-21:
First of all, your firmware version is very old. 
I don't know anything about a Fritzbox VPN Function. But if you enable HASS Auto Discovery you should also see a topic called `homeassistant/` where several dtu sensor appear. 
Auto Discovery works in a way that each device just publishes it's sensors into the `homeassistant/#` topic and HASS reads this topic.

tbnobody, 2022-10-21:
Can you make a screenshot of your mqtt settings? Especially the Home Assistant section?

tbnobody, 2022-10-21:
And btw... HASS will not publish any inverter data to the `homeassistant`/ topic... It will just place some json config there which describe where HASS can find the data itself.

LuckyLuzz, 2022-10-21:
alright I got it. The base topic setting withhin OpenDTU was set to solar/ per default. I changed it to homeassistant/ and now HASS is recognising the device. 

Thanks!

tbnobody, 2022-10-22:
Ah I remember that the HASS Auto Discovery Topic was wrong in early releases. The default was `solar/` instead of `homeassistant/`. 
Your setup should be similar to this screenshot: https://github.com/tbnobody/OpenDTU/blob/master/docs/screenshots/04_MqttAdmin.png

closed 2022-10-22 by tbnobody