tbnobody/OpenDTU #255: Long connection failures between opendtu and hm-600

Jonas960, 2022-10-22:
Hello,

I have a problem with the connection between opendtu and my Hoymiles HM-600. 
On 1/3 of the day the DTU data send to openDTU without problems and everything works. But the remaining 2/3 simply no data arrives (Data Age 8000 seconds are not uncommon). I have already changed the location but I can't get any closer. It's about 10 meters with a window in between. 
It is also strange that it really works 1 hour at a time without problems and then 4-5 hours not at all.  If it were due to the distance, the connection should always be unstable and not run for a long time in between?

The following components I have used for opendtu.

[image]

[image]

Is there perhaps the possibility of implementing a signal strength function or is that not possible with this transmission technology? 
I look forward to helpful hints.
Of course I have installed the latest version and I use a high quality cable :) 

Thanks!

oskook, 2022-10-23:
I used the same nRF24L01+ modules as in your picture with ESP32 and had endless issues to establish a stable connection to the inverter. Tried different nRF-modules (same brand), different antennas, different esp8266 and esp32, checked and changed the cabling a hundred times, always bad link. Eventually I stumbled across an entry here in the github issues where somebody suggested to use a shielded nRF-module (EBYTE, https://www.amazon.de/gp/product/B07P8TTHZH/) - voilà all problems gone, stable connection even from the basement to the roof.

Jonas960, 2022-10-23:
@oskook Thank you for your reply. I have ordered the module and will test it. I give Tuesday or Wednesday an update if it worked. In the Amazon reviews someone had the same problem and could solve it with this module. I am curious if it works.

OlSteph, 2022-10-28:
Hello,
i got this combination too and it works very well.

Jonas960, 2022-10-29:
Hi everybody,
I tested this new nRF24L01+ module and it works well. The only problem is, the module loses the connection after I add the antenna to it but I think its a production error of the module. 
Thank you!

tbnobody, 2022-10-31:
Hm this sounds like a power supply issue

ooassel, 2022-11-02:
I also use the same nRF24L01+ module as in the initial post and can not connect to the inverter at all, no matter how small the distance is - no data is transmitted. 
But: strangely enough, the only function that works is the "turn on" "turn off" and "restart" button and without delay, the inverter can be operated reliably. 
Suspect.

tbnobody, 2022-11-02:
Could you please provide the output of the serial log for further investigation?

Leviathan09, 2022-11-05:
I had the same connection problems but i fixed it with adding a 10 µF capacitor as mentioned in the troubleshooting section of the readme

frahri, 2022-11-10:
> Hello,
>
> I have a problem with the connection between opendtu and my Hoymiles HM-600. On 1/3 of the day the DTU data send to openDTU without problems and everything works. But the remaining 2/3 [...]

Hi,
a little question.
I own the same "setup".
There is no chance to get a connection.
Could you give me a little info about your wiring - controller <-> rf module?
I`m not sure if I did it right, did you change anything in programming the "pin" layout of your controller?
For me I took a generic controller.

Thx

Jonas960, 2022-11-10:
@frahri Hi, I used the same wiring as the README.md shows. Are you sure you used the right pinout for the rf module? 
[image]
(source: https://lastminuteengineers.com/nrf24l01-arduino-wireless-communication)

Also keep a look on the pinout for the esp. Be sure, it is the same than mine (wroom-32):

[image]
(source: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/).

frahri, 2022-11-10:
@Jonas960 ,

ihm a little confused about the pin assignment of the controller.
I use the same controller you mentioned in your first post.
In your last reply the controller has a different setup of the pins?

Here is my layout:

RF:                                        Controller:
1 Gnd                                    Gnd  is mine / GPIO6 in the Picture?
2 VCC                                   3V3
3CE                                       D4 is mine / GPIO0 in the Picture?
4CSN                                    D5 is mine / GPIO4 in the Picture?
5SCK                                    D18 is mine / GPIO16 in the Picture?
6MOSI                                  D23 is mine /? in the Picture? 
Here im totally confused , the controller of mine has only 15 pins on each side like the controller in your first post??

Jonas960, 2022-11-10:
@frahri sorry, my bad! It's a bit different on the board in the last post. I try to explain how my pinout looks like:

RF: Controller
GND: GND
VCC: 3V3
CE: D5/GPIO5
CSN: D4/GPIO4
SCK: D23/GPIO23
MOSI: D18/GPIO18
MISO: RX2/GPIO16
IRQ: D19/GPIO19

Try this and tell me if it works :)

Jonas960, 2022-11-10:
I close this issue for now. The problem was solved with a shielded nRF-module.

closed 2022-11-10 by Jonas960

ooassel, 2022-11-20:
After the confusion about the correct connection of the 6 data lines to the radio module, I tried both variants and can say for clarification that the one posted last by frahri is the correct one for the 30-pin module to which is shown in the opening post: 
https://github.com/tbnobody/OpenDTU/issues/255#issuecomment-1310050812
https://www.makershop.de/plattformen/nodemcu/espressif-esp32-dev-kit-board/
Only with this wiring, I get "connected" and the (correct) chip type in the info menu -> radio information. 
Since frahri as well as OlSteph seem to have no problems with this combination of ESP32 and rf-module, but I get no connection even after adding the recommended 10µF capacitor (except control the switching on and off of the inverter) I keep searching for the issue with mine before I invest in another rf-module.

frahri, 2022-11-23:
@ooassel 
short info:

[https://amzn.eu/aYWeIbp](url)

I use this radio module.
Greetings ☺️